victronenergy/venus #293: vedirect devices not found / disconnected

jhofstee, 2018-07-13:
Robert already mentioned that in his setup vedirect services appear as disconnected on the GUI. The ccgx was replaced, but this one shows similar behavior.

The easy did not publish any service on the dbus for the solar charger. After restarting the process it appeared again. Running on v2.20~37.

jhofstee, 2018-07-20:
Robert his ccgx is in quite a weird state again:

'''
[255586.167040] Unable to handle kernel NULL pointer dereference at virtual address 00000000
[255586.167080] pgd = c8a04000
[255586.167093] [00000000] *pgd=88941831, *pte=00000000, *ppte=00000000
[255586.167133] Internal error: Oops: 80000007 [#1] ARM
[255586.167146] Modules linked in: asix usbnet mii ftdi_sio usbserial
[255586.167182] CPU: 0 PID: 27817 Comm: start-vedirect. Not tainted 4.9.82 #1
[255586.167193] Hardware name: Generic AM3517 (Flattened Device Tree)
[255586.167205] task: c88ae840 task.stack: cd75e000
[255586.167220] PC is at 0x0
[255586.167256] LR is at task_work_run+0x74/0x98
[255586.167269] pc : [<00000000>]    lr : [<c0140cc8>]    psr: a0030013
[255586.167269] sp : cd75ff88  ip : cd599c00  fp : be95763c
[255586.167284] r10: 00000000  r9 : cd75e000  r8 : c0107784
[255586.167295] r7 : 00000000  r6 : c0962524  r5 : c88ae840  r4 : c88aec54
[255586.167306] r3 : 00000000  r2 : 00000000  r1 : 00000000  r0 : cd599c00
[255586.167320] Flags: NzCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment none
[255586.167332] Control: 10c5387d  Table: 88a04019  DAC: 00000051
[255586.167344] Process start-vedirect. (pid: 27817, stack limit = 0xcd75e210)
[255586.167356] Stack: (0xcd75ff88 to 0xcd760000)
[255586.167375] ff80:                   ffffe000 c0107784 cd75ffb0 0000005b c0107784 c010aef8
[255586.167396] ffa0: b6f308a4 b6f30000 80000001 c0107620 00000000 00001c32 b6f2e850 b6f2afff
[255586.167415] ffc0: b6f308a4 b6f30000 80000001 0000005b b6e71100 b6e878b8 00000000 be95763c
[255586.167434] ffe0: be9574c0 be9575dc b6f157ec b6f191bc 80030010 b6f2b000 00000000 00000000
[255586.167487] [<c0140cc8>] (task_work_run) from [<c010aef8>] (do_work_pending+0xbc/0xc0)
[255586.167515] [<c010aef8>] (do_work_pending) from [<c0107620>] (slow_work_pending+0xc/0x20)
[255586.167536] Code: bad PC value
[255586.167553] ---[ end trace 24a3069ad749deb4 ]---
'''

/service/vedirect-interface.ttyUSB1 and /service/gps-dbus.ttyUSB1 are both down, /var/lock/serial-starter/ttyUSB1 still exists

mpvader, 2018-08-15:
Continued in the hang issue: #301.

closed 2018-08-15 by mpvader