victronenergy/venus #294: mqtt-rpc: no connected devices found wording

mpvader, 2018-07-13:
Replace `No connected devices found` with `No devices found that can be firmware updated remotely`.

[image]

Seen on https://vrm.victronenergy.com/installation/1039/settings/firmware-update, which has devices, but no VE.Can and VE.Direct devices.

mauritsvdvijgh, 2018-07-14:
Will be in master with the other changes, addressed here: https://gitlab.trimm.nl/fronteers/victron-gonzales-frontend/commit/6a3015cbe7d9ed8fbdcafd897f9b5c2a17fcb564

mauritsvdvijgh, 2018-07-16:
This change is live on production so this issue can be closed.

mpvader, 2018-07-17:
[image]

closed 2018-07-17 by mpvader