victronenergy/venus #296: mqtt-rpc: add error code that links to reason for not updatable

mpvader, 2018-07-16:
[image]

The reasons for not being updatable are:

A) Blacklisted (Ion Control, Lynxes, cannot be updated remotely; can be updated locally though)
B) vup / dup reports that it cannot update the device, typically because its not a victron device.
C) Not in library

Once implemented code are defined, @mpvader, will add the info to the [manual](https://www.victronenergy.com/live/vrm_portal:remote-firmware-update).

mauritsvdvijgh, 2018-07-23:
The following codes are returned by vrm-mqtt-rpc when updatable is false:

```
1341 Not updatable: No product id returned by vup/dup. Typically because it is not a victron product
1342 Not updatable: Updatable not true as returned by the device/vup/dup.
1343 Not updatable: Wrong mode. The device says it it updatable, but the device is not in an updatable mode.
1344 Not updatable: Blacklisted. The device says it is updatable, the device is in an updatable mode, but the device is blacklisted
```

mauritsvdvijgh, 2018-07-23:
https://acceptancevrm.victronenergy.com/installation/975/firmware-update

[image]

mauritsvdvijgh, 2018-07-25:
Not on acceptance, can put it on acceptance if/when you want to test it.

mpvader, 2018-07-26:
Lijkt me prima, push maar naar live.

mpvader, 2018-07-26:
One more question, what about 1343, wrong mode? What’s the condition for that one again?

During scanning, devices will never be in boot mode.

mauritsvdvijgh, 2018-07-26:
Devices can be in boot mode during scanning, right? 

But we allow updating when in boot mode now: https://github.com/victronenergy/venus/issues/246

1343 happens when the device is not in mode app, rc or boot. Should not happen ordinarily.

mauritsvdvijgh, 2018-07-26:
Deployed to production:
https://vrm.victronenergy.com/installation/975/firmware-update

mpvader, 2018-07-26:
> Devices can be in boot mode during scanning, right?
> But we allow updating when in boot mode now: #246
> 1343 happens when the device is not in mode app, rc or boot. Should not happen ordinarily.

I don't understand; I'll contact you on Slack.

mauritsvdvijgh, 2018-07-26:
The check that triggers a 1343 has been removed after it turned out that it is unnecessary.

error code 1344 will take the place of 1343 (so renamed 1344 to 1343).

List of related errors is now as follows:

```
1341 Not updatable: No product id returned by vup/dup. Typically because it is not a victron product
1342 Not updatable: Updatable not true as returned by the device/vup/dup.
1343 Not updatable: Blacklisted. The device says it is updatable, but the device is blacklisted
```

Added the error descriptions to the tech docs here: https://github.com/victronenergy/venus-private/wiki/mqtt-rpc-and-vrm