victronenergy/venus #299: VRM / mqtt-rpc

jhofstee, 2018-07-23:
After a successful update there are sometimes no products found. Rescanning makes them appear again.

mauritsvdvijgh, 2018-07-23:
If it happens again, please update this issue with some details (when it happened and on which site it happened), so I can investigate.

mpvader, 2018-07-24:
@mauritsvdvijgh try on 
https://vrm.victronenergy.com/installation/13535/overview

We released a new firmware version for the MPPTs just now: you can try 10 orso times without having to resort to manual file selection.

mauritsvdvijgh, 2018-07-25:
seems offline today.

mauritsvdvijgh, 2018-07-26:
Updated them all, did not notice any weird behavior/emtpy device lists.

Used macos and firefox.

Did you perhaps have multiple tabs open on the firmware-update page or something?

mpvader, 2018-08-23:
Closing; we'll re-open this when we see it again.

closed 2018-08-23 by mpvader