victronenergy/venus #301: mkx process stuck in an uninterruptible sleep

jhofstee, 2018-07-30:
On Robert's CCGX the mk2 process is stuck for severals days in an uninterruptible sleep state, running v2.20~43-debug which is v2.20~44 + linux patch to dump a stack trace if a NULL pointer is scheduled as work.

1284  1265 root     D     2924   1%   0% /opt/victronenergy/vecan-mk2/mk2-dbus --log-before 25 --log-after 25 --banner -w -s /dev/ttyO1 -i -t mk3 --settings /data/var/lib/mk2-dbus/mkxport.settings

siteid: 22268

jhofstee, 2018-07-30:
echo w >/proc/sysrq-trigger

[605111.399762]   task                PC stack   pid father
[605111.400055] mk2-dbus        D    0  1284   1265 0x00000000
[605111.400123] [<c05c2e20>] (__schedule) from [<c05c31e0>] (schedule+0x40/0xa0)
[605111.400153] [<c05c31e0>] (schedule) from [<c05c5548>] (rwsem_down_read_failed+0xa8/0xe0)
[605111.400182] [<c05c5548>] (rwsem_down_read_failed) from [<c03df4d4>] (n_tty_read+0xc0/0x880)
[605111.400204] [<c03df4d4>] (n_tty_read) from [<c03da07c>] (tty_read+0x90/0xd4)
[605111.400238] [<c03da07c>] (tty_read) from [<c021f8a4>] (__vfs_read+0x1c/0x10c)
[605111.400264] [<c021f8a4>] (__vfs_read) from [<c0220610>] (vfs_read+0x8c/0x110)
[605111.400286] [<c0220610>] (vfs_read) from [<c0221498>] (SyS_read+0x3c/0x90)
[605111.400318] [<c0221498>] (SyS_read) from [<c0107560>] (ret_fast_syscall+0x0/0x48)
[605111.400515] kworker/u2:0    D    0 19898      2 0x00000000
[605111.400554] Workqueue: events_unbound flush_to_ldisc
[605111.400580] [<c05c2e20>] (__schedule) from [<c05c31e0>] (schedule+0x40/0xa0)
[605111.400601] [<c05c31e0>] (schedule) from [<c05c5548>] (rwsem_down_read_failed+0xa8/0xe0)
[605111.400622] [<c05c5548>] (rwsem_down_read_failed) from [<c03dfd60>] (n_tty_receive_buf_common+0x30/0x9c0)
[605111.400643] [<c03dfd60>] (n_tty_receive_buf_common) from [<c03e0704>] (n_tty_receive_buf2+0x14/0x1c)
[605111.400662] [<c03e0704>] (n_tty_receive_buf2) from [<c03e2dec>] (flush_to_ldisc+0x98/0xc4)
[605111.400692] [<c03e2dec>] (flush_to_ldisc) from [<c013d274>] (process_one_work+0x1c4/0x45c)
[605111.400714] [<c013d274>] (process_one_work) from [<c013d56c>] (worker_thread+0x60/0x598)
[605111.400736] [<c013d56c>] (worker_thread) from [<c01425a8>] (kthread+0x100/0x108)
[605111.400757] [<c01425a8>] (kthread) from [<c0107630>] (ret_from_fork+0x14/0x24)

jhofstee, 2018-07-31:
Another getting stuck on ps and top:

https://vrm.victronenergy.com/installation/1465/diagnostics

v2.20~45
Victron Energy
20180725224151

root@ccgx:~# technexion-serial        
151244000000000339 -> Week 44, 2012

root@ccgx:~# uname -a
Linux ccgx 4.9.113 #1 Thu Jul 26 00:43:26 CEST 2018 armv7l GNU/Linux

[468649.867951] sysrq: SysRq : Show Blocked State
[468649.868008]   task                PC stack   pid father
[468649.868364] ls              D    0  9020   9019 0x00000000
[468649.868435] [<c05c2de0>] (__schedule) from [<c05c31a0>] (schedule+0x40/0xa0)
[468649.868464] [<c05c31a0>] (schedule) from [<c05c5508>] (rwsem_down_read_failed+0xa8/0xe0)
[468649.868501] [<c05c5508>] (rwsem_down_read_failed) from [<c012a460>] (do_exit+0x124/0x98c)
[468649.868528] [<c012a460>] (do_exit) from [<c012ad34>] (do_group_exit+0x40/0xa4)
[468649.868551] [<c012ad34>] (do_group_exit) from [<c012ada8>] (__wake_up_parent+0x0/0x18)
[468649.868570] ps              D    0  9092   9088 0x00000001
[468649.868597] [<c05c2de0>] (__schedule) from [<c05c31a0>] (schedule+0x40/0xa0)
[468649.868616] [<c05c31a0>] (schedule) from [<c05c5508>] (rwsem_down_read_failed+0xa8/0xe0)
[468649.868648] [<c05c5508>] (rwsem_down_read_failed) from [<c0275e08>] (proc_pid_cmdline_read+0xf0/0x5bc)
[468649.868687] [<c0275e08>] (proc_pid_cmdline_read) from [<c021f86c>] (__vfs_read+0x1c/0x10c)
[468649.868711] [<c021f86c>] (__vfs_read) from [<c02205d8>] (vfs_read+0x8c/0x110)
[468649.868733] [<c02205d8>] (vfs_read) from [<c0221460>] (SyS_read+0x3c/0x90)
[468649.868763] [<c0221460>] (SyS_read) from [<c0107560>] (ret_fast_syscall+0x0/0x48)
[468649.868781] top             D    0  9099   9096 0x00000000
[468649.868806] [<c05c2de0>] (__schedule) from [<c05c31a0>] (schedule+0x40/0xa0)
[468649.868826] [<c05c31a0>] (schedule) from [<c05c5508>] (rwsem_down_read_failed+0xa8/0xe0)
[468649.868849] [<c05c5508>] (rwsem_down_read_failed) from [<c0275e08>] (proc_pid_cmdline_read+0xf0/0x5bc)
[468649.868872] [<c0275e08>] (proc_pid_cmdline_read) from [<c021f86c>] (__vfs_read+0x1c/0x10c)
[468649.868893] [<c021f86c>] (__vfs_read) from [<c02205d8>] (vfs_read+0x8c/0x110)
[468649.868914] [<c02205d8>] (vfs_read) from [<c0221460>] (SyS_read+0x3c/0x90)
[468649.868935] [<c0221460>] (SyS_read) from [<c0107560>] (ret_fast_syscall+0x0/0x48)

jhofstee, 2018-08-02:
and another one,, hub1 this time:

[475255.166751] sysrq: SysRq : Show Blocked State
[475255.166806]   task                PC stack   pid father
[475255.167122] vedirect-dbus   D    0 19553  19551 0x00000000
[475255.167190] [<c05c2e20>] (__schedule) from [<c05c31e0>] (schedule+0x40/0xa0)
[475255.167218] [<c05c31e0>] (schedule) from [<c05c5548>] (rwsem_down_read_failed+0xa8/0xe0)
[475255.167246] [<c05c5548>] (rwsem_down_read_failed) from [<c03df4d4>] (n_tty_read+0xc0/0x880)
[475255.167268] [<c03df4d4>] (n_tty_read) from [<c03da07c>] (tty_read+0x90/0xd4)
[475255.167303] [<c03da07c>] (tty_read) from [<c021f8a4>] (__vfs_read+0x1c/0x10c)
[475255.167329] [<c021f8a4>] (__vfs_read) from [<c0220610>] (vfs_read+0x8c/0x110)
[475255.167351] [<c0220610>] (vfs_read) from [<c0221498>] (SyS_read+0x3c/0x90)
[475255.167383] [<c0221498>] (SyS_read) from [<c0107560>] (ret_fast_syscall+0x0/0x48)
[475255.167397] kworker/u2:2    D    0  1166      2 0x00000000
[475255.167430] Workqueue: events_unbound flush_to_ldisc
[475255.167453] [<c05c2e20>] (__schedule) from [<c05c31e0>] (schedule+0x40/0xa0)
[475255.167472] [<c05c31e0>] (schedule) from [<c05c5548>] (rwsem_down_read_failed+0xa8/0xe0)
[475255.167493] [<c05c5548>] (rwsem_down_read_failed) from [<c03dfd60>] (n_tty_receive_buf_common+0x30/0x9c0)
[475255.167513] [<c03dfd60>] (n_tty_receive_buf_common) from [<c03e0704>] (n_tty_receive_buf2+0x14/0x1c)
[475255.167533] [<c03e0704>] (n_tty_receive_buf2) from [<c03e2dec>] (flush_to_ldisc+0x98/0xc4)
[475255.167562] [<c03e2dec>] (flush_to_ldisc) from [<c013d274>] (process_one_work+0x1c4/0x45c)
[475255.167585] [<c013d274>] (process_one_work) from [<c013d56c>] (worker_thread+0x60/0x598)
[475255.167605] [<c013d56c>] (worker_thread) from [<c01425a8>] (kthread+0x100/0x108)
[475255.167627] [<c01425a8>] (kthread) from [<c0107630>] (ret_from_fork+0x14/0x24)

mpvader, 2018-08-26:
Found on on a 3.7 kernel (v2.20~54)

```
[   30.710662] turning CAN driver on
[   30.710723] ti_hecc ti_hecc can0: setting CANBTC=0x30049
[ 7080.248901] INFO: task serial-starter.:1597 blocked for more than 120 seconds.
[ 7080.248931] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[ 7080.248962] serial-starter. D c036c778     0  1597   1571 0x00000000
[ 7080.249023] [<c036c778>] (__schedule+0x208/0x44c) from [<c036bbec>] (__mutex_lock_slowpath+0x7c/0xb8)
[ 7080.249053] [<c036bbec>] (__mutex_lock_slowpath+0x7c/0xb8) from [<c0094308>] (anon_vma_fork+0x98/0x11c)
[ 7080.249084] [<c0094308>] (anon_vma_fork+0x98/0x11c) from [<c002bc08>] (dup_mm+0x160/0x3f0)
[ 7080.249114] [<c002bc08>] (dup_mm+0x160/0x3f0) from [<c002c90c>] (copy_process.part.7+0xa34/0xc88)
[ 7080.249145] [<c002c90c>] (copy_process.part.7+0xa34/0xc88) from [<c002cc88>] (do_fork+0x104/0x290)
[ 7080.249176] [<c002cc88>] (do_fork+0x104/0x290) from [<c000e1e0>] (ret_fast_syscall+0x0/0x30)
[ 7200.248901] INFO: task serial-starter.:1597 blocked for more than 120 seconds.
[ 7200.248931] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[ 7200.248931] serial-starter. D c036c778     0  1597   1571 0x00000000
[ 7200.249023] [<c036c778>] (__schedule+0x208/0x44c) from [<c036bbec>] (__mutex_lock_slowpath+0x7c/0xb8)
[ 7200.249053] [<c036bbec>] (__mutex_lock_slowpath+0x7c/0xb8) from [<c0094308>] (anon_vma_fork+0x98/0x11c)
[ 7200.249084] [<c0094308>] (anon_vma_fork+0x98/0x11c) from [<c002bc08>] (dup_mm+0x160/0x3f0)
[ 7200.249114] [<c002bc08>] (dup_mm+0x160/0x3f0) from [<c002c90c>] (copy_process.part.7+0xa34/0xc88)
[ 7200.249114] [<c002c90c>] (copy_process.part.7+0xa34/0xc88) from [<c002cc88>] (do_fork+0x104/0x290)
[ 7200.249145] [<c002cc88>] (do_fork+0x104/0x290) from [<c000e1e0>] (ret_fast_syscall+0x0/0x30)
[ 7320.256256] INFO: task serial-starter.:1597 blocked for more than 120 seconds.
[ 7320.256256] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[ 7320.256286] serial-starter. D c036c778     0  1597   1571 0x00000000
[ 7320.256347] [<c036c778>] (__schedule+0x208/0x44c) from [<c036bbec>] (__mutex_lock_slowpath+0x7c/0xb8)
[ 7320.256408] [<c036bbec>] (__mutex_lock_slowpath+0x7c/0xb8) from [<c0094308>] (anon_vma_fork+0x98/0x11c)
[ 7320.256439] [<c0094308>] (anon_vma_fork+0x98/0x11c) from [<c002bc08>] (dup_mm+0x160/0x3f0)
[ 7320.256469] [<c002bc08>] (dup_mm+0x160/0x3f0) from [<c002c90c>] (copy_process.part.7+0xa34/0xc88)
[ 7320.256469] [<c002c90c>] (copy_process.part.7+0xa34/0xc88) from [<c002cc88>] (do_fork+0x104/0x290)
[ 7320.256500] [<c002cc88>] (do_fork+0x104/0x290) from [<c000e1e0>] (ret_fast_syscall+0x0/0x30)
[ 7440.256256] INFO: task serial-starter.:1597 blocked for more than 120 seconds.
[ 7440.256256] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[ 7440.256286] serial-starter. D c036c778     0  1597   1571 0x00000000
[ 7440.256347] [<c036c778>] (__schedule+0x208/0x44c) from [<c036bbec>] (__mutex_lock_slowpath+0x7c/0xb8)
[ 7440.256378] [<c036bbec>] (__mutex_lock_slowpath+0x7c/0xb8) from [<c0094308>] (anon_vma_fork+0x98/0x11c)
[ 7440.256439] [<c0094308>] (anon_vma_fork+0x98/0x11c) from [<c002bc08>] (dup_mm+0x160/0x3f0)
[ 7440.256439] [<c002bc08>] (dup_mm+0x160/0x3f0) from [<c002c90c>] (copy_process.part.7+0xa34/0xc88)
[ 7440.256469] [<c002c90c>] (copy_process.part.7+0xa34/0xc88) from [<c002cc88>] (do_fork+0x104/0x290)
[ 7440.256500] [<c002cc88>] (do_fork+0x104/0x290) from [<c000e1e0>] (ret_fast_syscall+0x0/0x30)
[ 7560.256256] INFO: task serial-starter.:1597 blocked for more than 120 seconds.
[ 7560.256286] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[ 7560.256286] serial-starter. D c036c778     0  1597   1571 0x00000000
[ 7560.256347] [<c036c778>] (__schedule+0x208/0x44c) from [<c036bbec>] (__mutex_lock_slowpath+0x7c/0xb8)
[ 7560.256408] [<c036bbec>] (__mutex_lock_slowpath+0x7c/0xb8) from [<c0094308>] (anon_vma_fork+0x98/0x11c)
[ 7560.256439] [<c0094308>] (anon_vma_fork+0x98/0x11c) from [<c002bc08>] (dup_mm+0x160/0x3f0)
[ 7560.256469] [<c002bc08>] (dup_mm+0x160/0x3f0) from [<c002c90c>] (copy_process.part.7+0xa34/0xc88)
[ 7560.256469] [<c002c90c>] (copy_process.part.7+0xa34/0xc88) from [<c002cc88>] (do_fork+0x104/0x290)
[ 7560.256500] [<c002cc88>] (do_fork+0x104/0x290) from [<c000e1e0>] (ret_fast_syscall+0x0/0x30)
[ 7680.256256] INFO: task serial-starter.:1597 blocked for more than 120 seconds.
[ 7680.256286] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[ 7680.256286] serial-starter. D c036c778     0  1597   1571 0x00000000
[ 7680.256378] [<c036c778>] (__schedule+0x208/0x44c) from [<c036bbec>] (__mutex_lock_slowpath+0x7c/0xb8)
[ 7680.256408] [<c036bbec>] (__mutex_lock_slowpath+0x7c/0xb8) from [<c0094308>] (anon_vma_fork+0x98/0x11c)
[ 7680.256439] [<c0094308>] (anon_vma_fork+0x98/0x11c) from [<c002bc08>] (dup_mm+0x160/0x3f0)
[ 7680.256469] [<c002bc08>] (dup_mm+0x160/0x3f0) from [<c002c90c>] (copy_process.part.7+0xa34/0xc88)
[ 7680.256469] [<c002c90c>] (copy_process.part.7+0xa34/0xc88) from [<c002cc88>] (do_fork+0x104/0x290)
[ 7680.256500] [<c002cc88>] (do_fork+0x104/0x290) from [<c000e1e0>] (ret_fast_syscall+0x0/0x30)
[ 7800.256256] INFO: task serial-starter.:1597 blocked for more than 120 seconds.
[ 7800.256286] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[ 7800.256286] serial-starter. D c036c778     0  1597   1571 0x00000000
[... cut ...]
[ 8160.256317] [<c036bbec>] (__mutex_lock_slowpath+0x7c/0xb8) from [<c0094308>] (anon_vma_fork+0x98/0x11c)
[ 8160.256347] [<c0094308>] (anon_vma_fork+0x98/0x11c) from [<c002bc08>] (dup_mm+0x160/0x3f0)
[ 8160.256378] [<c002bc08>] (dup_mm+0x160/0x3f0) from [<c002c90c>] (copy_process.part.7+0xa34/0xc88)
[ 8160.256378] [<c002c90c>] (copy_process.part.7+0xa34/0xc88) from [<c002cc88>] (do_fork+0x104/0x290)
[ 8160.256408] [<c002cc88>] (do_fork+0x104/0x290) from [<c000e1e0>] (ret_fast_syscall+0x0/0x30)
```

mpvader, 2018-08-28:
kernel 4.14

https://vrm.victronenergy.com/installation/12032/diagnostics

```
[  194.633679] udevd[74]: worker [86] /devices/platform/68000000.ocp/48064000.usbhshost/48064800.ehci/usb1/1-1/1-1.2/1-1.2:1.0/host0/target0:0:0/0:0:0:0/block/sda timeout; kill it
[  194.634130] udevd[74]: seq 1221 '/devices/platform/68000000.ocp/48064000.usbhshost/48064800.ehci/usb1/1-1/1-1.2/1-1.2:1.0/host0/target0:0:0/0:0:0:0/block/sda' killed
[  243.715919] INFO: task scsi_eh_0:88 blocked for more than 120 seconds.
[  243.716031]       Not tainted 4.14.67 #1
[  243.716056] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[  243.716084] scsi_eh_0       D    0    88      2 0x00000000
[  243.716184] [<c069f0f0>] (__schedule) from [<c069f8d4>] (schedule+0x54/0xc0)
[  243.716226] [<c069f8d4>] (schedule) from [<c069fc04>] (schedule_preempt_disabled+0x28/0x40)
[  243.716266] [<c069fc04>] (schedule_preempt_disabled) from [<c06a0c94>] (__mutex_lock+0x1f8/0x66c)
[  243.716303] [<c06a0c94>] (__mutex_lock) from [<c06a1124>] (mutex_lock_nested+0x1c/0x24)
[  243.716410] [<c06a1124>] (mutex_lock_nested) from [<bf00099c>] (device_reset+0x1c/0x48 [usb_storage])
[  243.716489] [<bf00099c>] (device_reset [usb_storage]) from [<c04b3b0c>] (scsi_eh_ready_devs+0x250/0x80c)
[  243.716531] [<c04b3b0c>] (scsi_eh_ready_devs) from [<c04b4e00>] (scsi_error_handler+0x3a8/0x3b8)
[  243.716571] [<c04b4e00>] (scsi_error_handler) from [<c014a5a8>] (kthread+0x15c/0x164)
[  243.716619] [<c014a5a8>] (kthread) from [<c0107d94>] (ret_from_fork+0x14/0x20)
[  243.716656] INFO: task usb-storage:90 blocked for more than 120 seconds.
[  243.716682]       Not tainted 4.14.67 #1
[  243.716705] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[  243.716729] usb-storage     D    0    90      2 0x00000000
[  243.716784] [<c069f0f0>] (__schedule) from [<c069f8d4>] (schedule+0x54/0xc0)
[  243.716823] [<c069f8d4>] (schedule) from [<c06a3dd4>] (schedule_timeout+0x23c/0x480)
[  243.716859] [<c06a3dd4>] (schedule_timeout) from [<c06a0378>] (wait_for_common+0xb4/0x164)
[  243.716906] [<c06a0378>] (wait_for_common) from [<c0519cb8>] (usb_sg_wait+0xf0/0x13c)
[  243.716973] [<c0519cb8>] (usb_sg_wait) from [<bf0012f8>] (usb_stor_bulk_transfer_sglist.part.2+0x98/0xf4 [usb_storage])
[  243.717058] [<bf0012f8>] (usb_stor_bulk_transfer_sglist.part.2 [usb_storage]) from [<bf001388>] (usb_stor_bulk_srb+0x34/0x4c [usb_storage])
[  243.717140] [<bf001388>] (usb_stor_bulk_srb [usb_storage]) from [<bf0014a8>] (usb_stor_Bulk_transport+0x108/0x364 [usb_storage])
[  243.717222] [<bf0014a8>] (usb_stor_Bulk_transport [usb_storage]) from [<bf001d18>] (usb_stor_invoke_transport+0x20/0x4b0 [usb_storage])
[  243.717303] [<bf001d18>] (usb_stor_invoke_transport [usb_storage]) from [<bf00316c>] (usb_stor_control_thread+0x164/0x2a0 [usb_storage])
[  243.717366] [<bf00316c>] (usb_stor_control_thread [usb_storage]) from [<c014a5a8>] (kthread+0x15c/0x164)
[  243.717404] [<c014a5a8>] (kthread) from [<c0107d94>] (ret_from_fork+0x14/0x20)
[  243.717655] 
[  243.717655] Showing all locks held in the system:
[  243.717703] 2 locks held by kworker/0:0/3:
[  243.717762]  #0:  ("events_freezable_power_efficient"){+.+.}, at: [<c01434c8>] process_one_work+0x1d4/0x74c
[  243.717868]  #1:  ((&(&ev->dwork)->work)){+.+.}, at: [<c01434c8>] process_one_work+0x1d4/0x74c
[  243.717964] 1 lock held by khungtaskd/11:
[  243.717987]  #0:  (tasklist_lock){.+.+}, at: [<c01653c8>] debug_show_all_locks+0x38/0x1ac
[  243.718121] 1 lock held by udevd/86:
[  243.718144]  #0:  (&bdev->bd_mutex){+.+.}, at: [<c02a82e8>] __blkdev_put+0x3c/0x1fc
[  243.718231] 1 lock held by scsi_eh_0/88:
[  243.718253]  #0:  (&us_interface_key[i]){+.+.}, at: [<bf00099c>] device_reset+0x1c/0x48 [usb_storage]
[  243.718376] 1 lock held by usb-storage/90:
[  243.718400]  #0:  (&us_interface_key[i]){+.+.}, at: [<bf003040>] usb_stor_control_thread+0x38/0x2a0 [usb_storage]
[  243.718776] 
[  243.718801] =============================================
[  243.718801] 
[  366.595981] INFO: task kworker/0:0:3 blocked for more than 120 seconds.
[  366.596043]       Not tainted 4.14.67 #1
[  366.596069] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[  366.596097] kworker/0:0     D    0     3      2 0x00000000
[  366.596176] Workqueue: events_freezable_power_ disk_events_workfn
[  366.596251] [<c069f0f0>] (__schedule) from [<c069f8d4>] (schedule+0x54/0xc0)
[  366.596293] [<c069f8d4>] (schedule) from [<c04b2268>] (scsi_block_when_processing_errors+0x54/0xb4)
[  366.596338] [<c04b2268>] (scsi_block_when_processing_errors) from [<c04c2128>] (sd_check_events+0x9c/0x124)
[  366.596375] [<c04c2128>] (sd_check_events) from [<c03cdb64>] (disk_check_events+0x40/0x11c)
[  366.596424] [<c03cdb64>] (disk_check_events) from [<c0143594>] (process_one_work+0x2a0/0x74c)
[  366.596465] [<c0143594>] (process_one_work) from [<c0143a90>] (worker_thread+0x50/0x5a8)
[  366.596503] [<c0143a90>] (worker_thread) from [<c014a5a8>] (kthread+0x15c/0x164)
[  366.596623] [<c014a5a8>] (kthread) from [<c0107d94>] (ret_from_fork+0x14/0x20)
[  366.596717] INFO: task udevd:86 blocked for more than 120 seconds.
[  366.596745]       Not tainted 4.14.67 #1
[  366.596768] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[  366.596792] udevd           D    0    86     74 0x00000005
[  366.596852] [<c069f0f0>] (__schedule) from [<c069f8d4>] (schedule+0x54/0xc0)
[... cut ...]
[  489.479224] 1 lock held by udevd/86:
[  489.479248]  #0:  (&bdev->bd_mutex){+.+.}, at: [<c02a82e8>] __blkdev_put+0x3c/0x1fc
[  489.479333] 1 lock held by scsi_eh_0/88:
[  489.479355]  #0:  (&us_interface_key[i]){+.+.}, at: [<bf00099c>] device_reset+0x1c/0x48 [usb_storage]
[  489.479474] 1 lock held by usb-storage/90:
[  489.479496]  #0:  (&us_interface_key[i]){+.+.}, at: [<bf003040>] usb_stor_control_thread+0x38/0x2a0 [usb_storage]
[  489.479773] 
[  489.479798] =============================================
[  489.479798] 
```

jhofstee, 2018-08-28:
Linux version 3.7.1 (oe-user@oe-host) (gcc version 7.3.0 (GCC) ) #1 Fri Aug 24 00:06:32 CEST 2018
```
Aug 25 02:33:24 ccgx user.info kernel: [    5.593139] usb 1-1.2: reset high-speed USB device number 3 using ehci-omap
Aug 25 02:33:24 ccgx user.info kernel: [   36.155609] usb 1-1.2: reset high-speed USB device number 3 using ehci-omap
Aug 25 02:33:24 ccgx user.info kernel: [   46.358764] usb 1-1.2: reset high-speed USB device number 3 using ehci-omap
Aug 25 02:33:24 ccgx user.info kernel: [   62.561828] usb 1-1.2: reset high-speed USB device number 3 using ehci-omap
Aug 25 02:33:24 ccgx user.info kernel: [   62.772705] usb 1-1.2: reset high-speed USB device number 3 using ehci-omap
Aug 25 02:33:24 ccgx user.info kernel: [   72.975921] usb 1-1.2: reset high-speed USB device number 3 using ehci-omap
Aug 25 02:33:24 ccgx user.info kernel: [   73.093536] sd 0:0:0:0: Device offlined - not ready after error recovery
Aug 25 02:33:24 ccgx user.info kernel: [   73.093627] sd 0:0:0:0: [sda] Unhandled error code
Aug 25 02:33:24 ccgx user.info kernel: [   73.093658] sd 0:0:0:0: [sda]  
Aug 25 02:33:24 ccgx user.warn kernel: [   73.093658] Result: hostbyte=0x05 driverbyte=0x00
Aug 25 02:33:24 ccgx user.info kernel: [   73.093688] sd 0:0:0:0: [sda] CDB: 
Aug 25 02:33:24 ccgx user.warn kernel: [   73.093688] cdb[0]=0x28: 28 00 00 78 04 78 00 00 a8 00
Aug 25 02:33:24 ccgx user.err kernel: [   73.093749] end_request: I/O error, dev sda, sector 7865464
Aug 25 02:33:24 ccgx user.err kernel: [   73.093780] Buffer I/O error on device sda, logical block 983183
Aug 25 02:33:24 ccgx user.err kernel: [   73.093811] Buffer I/O error on device sda, logical block 983184
Aug 25 02:33:24 ccgx user.err kernel: [   73.093841] Buffer I/O error on device sda, logical block 983185
Aug 25 02:33:24 ccgx user.err kernel: [   73.093872] Buffer I/O error on device sda, logical block 983186
Aug 25 02:33:24 ccgx user.err kernel: [   73.093902] Buffer I/O error on device sda, logical block 983187
Aug 25 02:33:24 ccgx user.err kernel: [   73.093902] Buffer I/O error on device sda, logical block 983188
Aug 25 02:33:24 ccgx user.err kernel: [   73.093933] Buffer I/O error on device sda, logical block 983189
Aug 25 02:33:24 ccgx user.err kernel: [   73.093963] Buffer I/O error on device sda, logical block 983190
Aug 25 02:33:24 ccgx user.err kernel: [   73.093994] Buffer I/O error on device sda, logical block 983191
Aug 25 02:33:24 ccgx user.err kernel: [   73.093994] Buffer I/O error on device sda, logical block 983192
Aug 25 02:33:24 ccgx user.err kernel: [   73.094116] sd 0:0:0:0: rejecting I/O to offline device
Aug 25 02:33:24 ccgx user.info kernel: [   73.094146] sd 0:0:0:0: [sda] killing request
Aug 25 02:33:24 ccgx user.err kernel: [   73.094177] sd 0:0:0:0: rejecting I/O to offline device
Aug 25 02:33:25 ccgx user.err kernel: [   73.094268] sd 0:0:0:0: rejecting I/O to offline device
Aug 25 02:33:25 ccgx user.info kernel: [   73.094360] sd 0:0:0:0: [sda] Unhandled error code
Aug 25 02:33:25 ccgx user.info kernel: [   73.094390] sd 0:0:0:0: [sda]  
Aug 25 02:33:25 ccgx user.warn kernel: [   73.094390] Result: hostbyte=0x01 driverbyte=0x00
Aug 25 02:33:25 ccgx user.info kernel: [   73.094421] sd 0:0:0:0: [sda] CDB: 
Aug 25 02:33:25 ccgx user.warn kernel: [   73.094421] cdb[0]=0x28: 28 00 00 78 05 28 00 00 30 00
Aug 25 02:33:25 ccgx user.err kernel: [   73.094482] end_request: I/O error, dev sda, sector 7865640
Aug 25 02:33:25 ccgx user.err kernel: [   73.094635] sd 0:0:0:0: rejecting I/O to offline device
Aug 25 02:33:25 ccgx user.err kernel: [   73.095062] sd 0:0:0:0: rejecting I/O to offline device
Aug 25 02:33:25 ccgx user.err kernel: [   73.095153] sd 0:0:0:0: rejecting I/O to offline device
```

jhofstee, 2018-09-14:
continues here, https://github.com/victronenergy/venus/issues/350 since there are still hung processes in 4.14.