victronenergy/venus #303: Missing features from the generator auto-start functionality

Wholistic, 2018-08-01:
The following features are currently missing from the generator auto-start functionality. They are all very similar so I thought best to keep them grouped.

**Warm up time** - delay from when the signal to start the generator is sent to when the internal transfer switch is closed to allow the generator to warm up. Programmable 0-10 minutes

**Cool down time** - delay from when the ‘target’ is reached (eg 80%), where internal transfer switch is opened for programmable amount of time before the signal to turn the generator off is sent. This allows the generator to cool down, so it doesn’t go from full load to off, which can lead to overheating of the engine.

**Maximum number of retries and delay between retries** - This feature needs to work in conjunction with AC-Input detection AND the warm up timer relay delay. If the generator is out of fuel, the autostart command can flatten the starter battery OR destroy the starter motor through attempting to autostart indefinitely.

This setting applies some limit to number of retries before VENUS gives up hope of starting it, and the time between tries to let parts cool down and starter battery recover voltage.

- Guy Stewart

mpvader, 2018-08-13:
Hi @Wholistic, its not straightforward to control the transfer relay in the inverter/chargers from within Venus.

Requests for a warm-up period come up now and then, but I've discussed this with the sales team and nobody thinks its high enough a priority to shift other things aside.

Another reason to not do this is that its quite easy to fix without software by using a contactor and a timer, or timed relay/contactor, all readily available in a good electrical stop.

(welcome to re-open, if you have a really compelling argument!)

jepefe, 2018-08-28:
I think that setting 0A to force inverter mode doesn't work when power assist is enabled. Instad we can set /Mode to inverter only. I've been testing this with a modified generator start/stop script for more than one year with a Multiplus and works perfectly.

mpvader, 2018-08-28:
You mean that you have warm-up and cool-down already implemented?

jepefe, 2018-08-28:
Sorry, not the warm-up & cool-down. I've been using /Mode to force the Multi to inverter only. This modified version handles the inverter mode instead of the relay.

So should be also fine to use it for warm-up & cool-down.

mpvader, 2018-08-29:
Ok then I’ll open this issue & we’ll wait for this issuebto float to the top of the todo list; nice!

radven, 2018-10-02:
Chiming in to support this issue.

Experimenting with a new Quattro setup - the most significant issue I am finding is no easy way to allow the generator to warm up / cool down. The secondary output has a two minute timer before it is engaged - I'd love to have an option to enable a similar timer on the primary generator input. 

And for shutting down....   I would love to be able to manually force a shut off the input relay via the Venus/CCGX so that I can let the generator run at idle for a few minutes before shutting it off. 

It is definitely not ideal to shut off a generator while the Quattro is charging the batteries at full speed.

These are what is needed for manually starting / stopping the generator cleanly. This all should be configurable via the Auto-Start UI as well. Thanks!

anilRghatikar, 2018-10-04:
It may be also possible ( easier) to assign the second not used/spare contact at least in VENUS GX to operate a power contactor. The power contactor will on through the contact after a time delay following generator start command - the dealy can be enabled in GUI in generator start-stop.

During stop mode, the aux contact will open  and after a settable delay generator stop command is sent 

If the user does not want to use the power contactor He can wire the spare contact to  us ignore AC  assitrant input on Quattro so Generator will get loaded after a predefined time and will get unloaded before a stop is actually sent

dewitpj, 2018-12-17:
Hi Guys,

Reviewing what is said above and the current source code this problem appears simple to solve at first sight but it does have some complicated ends. For example - starting the generator based on low battery, if you have a warm up time this has to be taken into consideration. Looking at the README file, can I get some comments on the following:

Available conditions:
- Manual - This one is "easy" - run the genset for X seconds and then bring up the load
- State of charge (SOC) - Calculate the run time remaining (or use it) and factor in the warm up time
- AC load - This one is a bit harder since loads are not easy predictable (unless this is loss of AC)
- Battery current - same as SoC
- Battery voltage - same as SoC
- Maintenance - Same as manual ?
- Inverter high temperature warning - same as AC loads ?
- Inverter overload warning - same as AC loads ?

So, that leaves us with pretty much only one option - let's tick off the easiest one, IMHO - loss of AC. This one is also fairly easy to extend into battery SoC.

I think at the start a few assumptions has to be made:

* AC Input 1 is the grid/default power source
* AC Input 2 is the generator
* If AC Input 1 is dead, we will run on DC until the generator is ready
* AC Input 2 will have a 0 current rating until the generator is ready
* When AC Input 1 is returned, it will be allowed to settle and the load will be gracefully transferred
* After the transfer, the generator will be ran for X seconds and then the remote start will be lifted

How does that sound to everyone ?

anilRghatikar, 2018-12-18:
With the upcoming Node Red implantation I think a user can himself write the necessary  routines to force inverter only mode during warm up and shutdown 

Will the inverter only /Mode be exposed to NODE red

Wholistic, 2019-12-17:
An similar feature request was made here on community - the ability to ramp up, and ramp down charging - https://community.victronenergy.com/questions/9190/overload-l1-as-generator-stops.html

Wholistic, 2020-02-14:
still in demand :) 

https://community.victronenergy.com/questions/41930/venus-disconnect-ac-in-1-when-generator-signal-is.html

gokuro, 2020-11-18:
it would seem simple enough to at least be able to disable the charger during a cool down period.

KidA001, 2021-01-29:
Commenting here to put my support on this feature. I've currently gotten around it by using NodeRed to control my auto start/stop and putting the Multi in Inverter Only mode over MQTT, delaying x-min, then shutting down and restoring the inverter to inverter/charger mode.

None the less this seems like a important feature to add to the Venus devices under the auto start ui

anilRghatikar, 2021-01-29:
Interesting . Do you run node red on Venus on in separate Rasp Pi 
 I have used a simple timer and contactor but I have to keep GEN not available on AC input alarm off . Did you find a way around it ?

Wholistic, 2021-02-15:
https://community.victronenergy.com/questions/77767/generator-warm-up-charger-walk-in.html

KW4NP, 2022-09-26:
> Hi Guys,
>
> Reviewing what is said above and the current source code this problem appears simple to solve at first sight but it does have some complicated ends. For example - starting the generator [...]

This makes sense to me, although the manual recommends wiring generator to AC in 1. It's now 2022 and Victron users are still asking about this. https://community.victronenergy.com/idea/155673/generator-cool-down-incorporated-in-auto-gen-start.html
 I think a start delay is less critical than a cool down delay, asking users to write their own timer is not good customer service.

Wholistic, 2022-11-27:
I have posted several other links to ongoing community feature request and work arounds. Here is the complete relevant list to date:

https://community.victronenergy.com/questions/3023/generator-cool-down-timer.html

https://community.victronenergy.com/questions/7975/delay-generator-synchronisation-after-genetator-is.html

https://community.victronenergy.com/questions/55483/3-wire-onan-generator-start-stop-based-on-soc-with.html

https://community.victronenergy.com/questions/42589/warm-up-and-cool-down-with-ags.html

https://community.victronenergy.com/questions/41930/venus-disconnect-ac-in-1-when-generator-signal-is.html

https://community.victronenergy.com/questions/158896/assistant-for-generator-cooldown.html

https://community.victronenergy.com/questions/109223/is-it-possible-to-have-a-warm-up-and-cool-down-tim.html

KW4NP, 2022-11-28:
> I have posted several other links to ongoing community feature request and work arounds. Here is the complete relevant list to date: [...]

Thanks for these workarounds. I have at least two installations where this is a feature that the customer wants. Workarounds are helpful, but I'm still hoping that Victron can integrate this into their system so that no additional programming or wiring is required. It seems like it must be possible to implement as others have written custom solutions. What needs to happen for this request to move forward?

mpvader, 2023-03-25:
This is all or mostly taken care of in #1031

closed 2023-03-25 by mpvader