victronenergy/venus #304: digital inputs: Make the multiplier more flexible for pulse counters

izak, 2018-08-08:
In the GUI, the multiplier works in multiples of 0.1. Some pulse counting hardware are not quite decimal. In addition a more flexible multiplier would allow imperial and US devices to be used for metric measurements, and the other round too.

Plan: Replace the widget in the GUI with a text entry widget.

gsosna, 2018-08-08:
Please consider allowing at least 5 or 6 decimal digits - I'm aware of readily available pulse flow meters with 700-1000 pulses per litre.

closed 2018-10-20 by jhofstee

gsosna, 2018-10-20:
@jhofstee can I test it for you guys?

jhofstee, 2018-10-21:
yes, see https://github.com/victronenergy/venus/wiki/release-cycle.

The fact that I closed the issue, means it is included in a next Venus build. There might be some delay between this happening and the develop images being available though, but typically several hours. This is a completely automated step, please don't auto install develop images!

After some basic test it is moved to testing as well, see https://updates.victronenergy.com/feeds/venus/testing, but that is a manual action. I will likely do that tomorrow.

You can install the testing version, once it is available, to check if it is working properly or reopen the issue.

mpvader, 2018-10-27:
The comment with the change made was:

—- 
improve flexibility of multiplier setting of pulse counter
This replaces the SpinBox with an editable text element.
Up to 6 decimal places are allowed
—- 

Now I understand why you are on the testing feed 🙂; see community comments

gsosna, 2018-10-28:
Thanks @jhofstee and @mpvader. I am a bit of a noob with this! I was hoping to have it tested today, and the software seems to be working as expected - I could enter 0.001429 successfully, but now my flow meter is not working! I'll be replacing it in the next few days, and will report back.

gsosna, 2018-11-10:
I can confirm that the counter and multiplier is working correctly for me. The counter changes to 1L after 500mL, but that's because of rounding - and I'm OK with that, even it's not quite what a user would expect. Thank you for your work on this.