victronenergy/venus #308: characterdisplay: complete the project

mpvader, 2018-08-16:
https://github.com/victronenergy/dbus-characterdisplay

Overall goals of the display:
- show what the system is doing 
- show details in case of an error
- show network status / ipaddresses
- show system as the system of a whole, especially for the EasySolar. User doesn't care that for Victron people its actually two products.

Existing pages:
- Grid
- Solar
- Battery
- LAN
- WiFi

Todo:
- Show status (inverting / feeding back / power assist etc)
- Add showing errors (both of MPPT & of Multi)
- Add DVCC reason codes
- Replace hardcoded `grid` with AC input type
- Add GSM page
- Make the button work: it should advance to the next page, switch backlight on, & delay the automatic moving to the next page for a while (10 seconds orso)
- Make the light sensor work.

Note: there is a known bug in the H3, one of the two possible PWM pins doesn't work. The one that does work is the same pin as the debug console. Before starting on this, first await the outcome of lower-level & hardware design.

The intented design for the backlight is to have it automatically switch between low (when it detects dark ambient light), and high (when it detects high ambient light). Possible with a software setting to just disable backlight automatically after x minutes, like on the CCGX.

izak, 2018-10-24:
Some screenshots.

Status page, showing two reason codes

[image]

Error page, skipped if there are no errors in the system:

[image]

Battery page:
[image]

AC page, showing the source of the AC as well:
[image]
[image]

Solar page showing production values and state of charger:
[image]

Network page. There's one for ethernet and one for Wifi. Not shown as part of the cyclical display, only reachable by button press. Skipped if not connected to the network.

[image]

rzijlstra-VictronEnergy, 2018-11-15:
Comments of the display text.

1. In a 3 phase system, it only provides the power of 1 unit, better is to provide the total power and maybe on other pages the power of L1, L2 and L3.

2. The display is also used in a Multiplus-II without MPPT, so now the "error"  "MPPT not found check connections should" should be removed the user is not able to check the connections anyway.

3.  sometimes the display does not respond to the button. After a reset of the Nanopi it is resolved again. The other functions (VRM website) are still running.

4. when I attach a MPPT (via USB) the LCD does not report the solar power. On VRM website it is vissible

izak, 2019-01-28:
IPV6 display is still broken.

rzijlstra-VictronEnergy, 2019-01-31:
we als have a Multiplus-II with a nanopi device. At this moment unit calls itself "Easysolar" this is not always correct. I would suggest to change it to something general. Or make it smart. If there is a MPPT connected to the internal VE-Direct device (next to the Powersupply) call is easysolar and else call in Multiplus-II.

Next in the latest firmware I see below screen. I Think Battery is missing: 
[image]

rzijlstra-VictronEnergy, 2019-01-31:
Make the display multiluange is something for the "TODO" list?

vejpasop, 2019-03-22:
I have some remarks about the way the Multi status is displayed on the MultiPlus-II with NanoPi.
On this device the Multi status leds are not available so there is only the display to rely on.  There are some situations where the user will have a hard time finding out what is going on.
1. The status display shows status "Off" even when the status is unknown.  If the Multi stays in bootloader mode due to a failed firmware update, usually the Multi will flash its leds quickly. On the character display it just says "Off" for ever. Better would be to indicate state "Unknown.
2. During (remote) VeConfigure the MK3 in the NanoPi is in blocked mode. This is shown in the "Device List" of the Venus GUI. The character display just shows "Off" in this case. 

[image]

The current status messages seem to represent the system status as shown in the Multi symbol of the "overview" page on the Venus GUI.  Here it is ok.  If something is not working you go to the menu and will get more info about the status of the Multi. From the device list it will shows state "Unknown" or not show the Multi at all or show state "Blocked"
Since the character display is integrated in the Multi I would expect to be able see the status of the Multi it selves.

vejpasop, 2019-03-22:
Can we we add the Multi state "Unknown" and "Blocked" to the system state? (The displayed status is derived from com.victronenergy.system/SystemState/State)

In addition to that, it would be nice if we can flash the backlight of the display in case of an alarm or VE.Bus error. Currently, for instance when there is a low battery alarm, the user has no idea when he/she is near the unit. 

Even nicer would be to have a backlight which turns red when there is something wrong.

mpvader, 2019-05-09:
Remove the decimal point for SOC

[image]