victronenergy/venus #319: Updating from 2.16 to 2.20 on my RPi fails

xychix, 2018-08-23:
Updating from 2.16 to 2.20 on my RPi fails while pressing the buttons in the Gui.
Attached the log as per discussed on the maillinglist.

`
2018-08-23 13:51:03.767702500 *** Checking for updates ***
2018-08-23 13:51:03.775938500 arguments: -check
2018-08-23 13:51:03.809067500 Retrieving latest version (feed=release)...
2018-08-23 13:51:04.738296500 installed: 20171003130541 v2.11
2018-08-23 13:51:04.746027500 available: 20180625141209 v2.16
2018-08-23 13:51:12.407555500 *** Checking for updates ***
2018-08-23 13:51:12.418670500 arguments: -update
2018-08-23 13:51:12.448998500 Retrieving latest version (feed=release)...
2018-08-23 13:51:13.462061500 installed: 20171003130541 v2.11
2018-08-23 13:51:13.469923500 available: 20180625141209 v2.16
2018-08-23 13:51:13.489272500 Starting swupdate to install version 20180625141209 v2.16 ...
2018-08-23 13:51:13.525906500 software set: stable mode: copy2
2018-08-23 13:51:14.527207500 Swupdate v2016.07.0
2018-08-23 13:51:14.537623500 
2018-08-23 13:51:14.537630500 Licensed under GPLv2. See source distribution for detailed copyright notices.
2018-08-23 13:51:14.537636500 
2018-08-23 13:51:14.537638500 Registered handlers:
2018-08-23 13:51:14.537640500 	archive
2018-08-23 13:51:14.537642500 	tar
2018-08-23 13:51:14.537644500 	raw
2018-08-23 13:51:14.537645500 	rawfile
2018-08-23 13:51:14.537647500 	uboot
2018-08-23 13:51:14.537649500 Main loop Daemon
2018-08-23 13:51:14.537652500 [NOTIFY] : SWUPDATE running :  [network_thread] : Incoming network request: processing...
2018-08-23 13:51:14.544355500 [NOTIFY] : SWUPDATE started :  Software Update started !
2018-08-23 13:51:14.544361500 [NOTIFY] : SWUPDATE downloading :  
2018-08-23 13:51:14.544364500 [NOTIFY] : SWUPDATE running :  [download_from_url] : Image download started : https://updates.victronenergy.com/feeds/venus/release/images/raspberrypi2/venus-swu-raspberrypi2-20180625141209-v2.16.swu
2018-08-23 13:51:15.433958500 [NOTIFY] : SWUPDATE running :  [extract_file_to_tmp] : Found file:
2018-08-23 13:51:15.441612500 	filename sw-description
2018-08-23 13:51:15.441619500 	size 786
2018-08-23 13:51:15.441621500 Version 0.1.0
2018-08-23 13:51:15.441623500 [NOTIFY] : SWUPDATE running :  [parse_images] : Found compressed Image  : venus-image-raspberrypi2.ext4.gz in device : /dev/mmcblk0p3 for handler raw (installed from stream) 
2018-08-23 13:51:15.441634500 
2018-08-23 13:51:15.441636500 [NOTIFY] : SWUPDATE running :  [parse_uboot] : U-Boot var: version = 2
2018-08-23 13:51:15.441641500 
2018-08-23 13:51:15.441642500 [NOTIFY] : SWUPDATE running :  [extract_files] : Found file:
2018-08-23 13:51:15.448479500 	filename venus-image-raspberrypi2.ext4.gz
2018-08-23 13:51:15.448485500 	size 70872748 required
2018-08-23 13:51:15.448488500 [NOTIFY] : SWUPDATE running :  [extract_files] : Installing STREAM venus-image-raspberrypi2.ext4.gz, 70872748 bytes
2018-08-23 13:51:15.448495500 
2018-08-23 13:51:15.448497500 [NOTIFY] : SWUPDATE running :  [install_single_image] : Found installer for stream venus-image-raspberrypi2.ext4.gz raw
2018-08-23 13:51:15.535922500 [NOTIFY] : SWUPDATE running :  [download_info] : Received : 48882 / 77886976
2018-08-23 13:51:16.538826500 [NOTIFY] : SWUPDATE running :  [download_info] : Received : 638706 / 77886976
2018-08-23 13:51:17.539660500 [NOTIFY] : SWUPDATE running :  [download_info] : Received : 1359602 / 77886976
2018-08-23 13:51:18.541124500 [NOTIFY] : SWUPDATE running :  [download_info] : Received : 2129650 / 77886976
2018-08-23 13:51:19.542883500 [NOTIFY] : SWUPDATE running :  [download_info] : Received : 2916082 / 77886976
2018-08-23 13:51:20.543653500 [NOTIFY] : SWUPDATE running :  [download_info] : Received : 3686130 / 77886976
2018-08-23 13:51:21.562954500 [NOTIFY] : SWUPDATE running :  [download_info] : Received : 4243186 / 77886976
2018-08-23 13:51:22.566178500 [NOTIFY] : SWUPDATE running :  [download_info] : Received : 5258994 / 77886976
2018-08-23 13:51:23.568554500 [NOTIFY] : SWUPDATE running :  [download_info] : Received : 6045426 / 77886976
2018-08-23 13:51:24.703363500 [NOTIFY] : SWUPDATE running :  [download_info] : Received : 6668018 / 77886976
2018-08-23 13:51:25.704635500 [NOTIFY] : SWUPDATE running :  [download_info] : Received : 7700210 / 77886976
2018-08-23 13:51:26.705393500 [NOTIFY] : SWUPDATE running :  [download_info] : Received : 8486642 / 77886976
2018-08-23 13:51:27.706377500 [NOTIFY] : SWUPDATE running :  [download_info] : Received : 9273074 / 77886976
2018-08-23 13:51:28.707237500 [NOTIFY] : SWUPDATE running :  [download_info] : Received : 10059506 / 77886976
2018-08-23 13:51:29.712014500 [NOTIFY] : SWUPDATE running :  [download_info] : Received : 10845938 / 77886976
2018-08-23 13:51:30.712242500 [NOTIFY] : SWUPDATE running :  [download_info] : Received : 11615986 / 77886976
2018-08-23 13:51:31.713751500 [NOTIFY] : SWUPDATE running :  [download_info] : Received : 12402418 / 77886976
2018-08-23 13:51:33.847052500 [NOTIFY] : SWUPDATE running :  [download_info] : Received : 12795634 / 77886976
2018-08-23 13:51:48.176964500 [NOTIFY] : SWUPDATE running :  [download_info] : Received : 14909170 / 77886976
2018-08-23 13:51:48.176975500 [NOTIFY] : SWUPDATE running :  [download_info] : Received : 15679218 / 77886976
2018-08-23 13:51:48.176981500 [NOTIFY] : SWUPDATE running :  [download_info] : Received : 16252658 / 77886976
2018-08-23 13:51:48.176987500 [NOTIFY] : SWUPDATE running :  [download_info] : Received : 17448690 / 77886976
2018-08-23 13:51:48.176992500 [NOTIFY] : SWUPDATE running :  [download_info] : Received : 18235122 / 77886976
2018-08-23 13:51:48.197259500 [NOTIFY] : SWUPDATE running :  [download_info] : Received : 19005170 / 77886976
2018-08-23 13:51:48.197297500 [NOTIFY] : SWUPDATE running :  [download_info] : Received : 19791602 / 77886976
[... 80 lines cut ...]
2018-08-23 13:52:52.297254500 [NOTIFY] : SWUPDATE running :  [download_info] : Received : 64536306 / 77886976
2018-08-23 13:52:53.298487500 [NOTIFY] : SWUPDATE running :  [download_info] : Received : 64536306 / 77886976
2018-08-23 13:52:54.299719500 [NOTIFY] : SWUPDATE running :  [download_info] : Received : 64536306 / 77886976
2018-08-23 13:52:55.300997500 [NOTIFY] : SWUPDATE running :  [download_info] : Received : 64536306 / 77886976
2018-08-23 13:52:56.302298500 [NOTIFY] : SWUPDATE running :  [download_info] : Received : 64536306 / 77886976
2018-08-23 13:52:57.303518500 [NOTIFY] : SWUPDATE running :  [download_info] : Received : 64536306 / 77886976
2018-08-23 13:52:58.304729500 [NOTIFY] : SWUPDATE running :  [download_info] : Received : 64536306 / 77886976
2018-08-23 13:52:59.306558500 [NOTIFY] : SWUPDATE running :  [download_info] : Received : 64536306 / 77886976
2018-08-23 13:53:00.307793500 [NOTIFY] : SWUPDATE running :  [download_info] : Received : 64536306 / 77886976
2018-08-23 13:53:01.309019500 [NOTIFY] : SWUPDATE running :  [download_info] : Received : 64536306 / 77886976
2018-08-23 13:53:02.310246500 [NOTIFY] : SWUPDATE running :  [download_info] : Received : 64536306 / 77886976
2018-08-23 13:53:03.311459500 [NOTIFY] : SWUPDATE running :  [download_info] : Received : 64536306 / 77886976
2018-08-23 13:53:04.312768500 [NOTIFY] : SWUPDATE running :  [download_info] : Received : 64536306 / 77886976
2018-08-23 13:53:05.313917500 [NOTIFY] : SWUPDATE running :  [download_info] : Received : 64536306 / 77886976
2018-08-23 13:53:06.315132500 [NOTIFY] : SWUPDATE running :  [download_info] : Received : 64536306 / 77886976
2018-08-23 13:53:07.316356500 [NOTIFY] : SWUPDATE running :  [download_info] : Received : 64536306 / 77886976
2018-08-23 13:53:08.317583500 [NOTIFY] : SWUPDATE running :  [download_info] : Received : 64536306 / 77886976
2018-08-23 13:53:09.318828500 [NOTIFY] : SWUPDATE running :  [download_info] : Received : 64536306 / 77886976
2018-08-23 13:53:10.320040500 [NOTIFY] : SWUPDATE running :  [download_info] : Received : 64536306 / 77886976
2018-08-23 13:53:11.321264500 [NOTIFY] : SWUPDATE running :  [download_info] : Received : 64536306 / 77886976
2018-08-23 13:53:12.322479500 [NOTIFY] : SWUPDATE running :  [download_info] : Received : 64536306 / 77886976
2018-08-23 13:53:13.323735500 [NOTIFY] : SWUPDATE running :  [download_info] : Received : 64536306 / 77886976
2018-08-23 13:53:13.563771500 [NOTIFY] : SWUPDATE running :  [download_from_url] : Connection with server interrupted, try RESUME after 64536306
2018-08-23 13:53:34.212383500 Error, do_swupdate stopped with exitcode 141, unlock and exit.
`

mpvader, 2018-08-23:
Hi @xychix, what I see here is some sort of internet interruption; which can happen.

Did a retry (from the gui) make it work OK for you?

xychix, 2018-08-23:
tried this several times. Yes my 4g is somewhat rotten.

Tried this several rounds to stable and to release candidate.

xychix, 2018-08-23:
I still appear to have some challenge there.

```
root@venus:/data# md5sum venus-swu-raspberrypi2-20180625141209-v2.16.swu 
3182cc9053d16d23ad534d59ec0e576a  venus-swu-raspberrypi2-20180625141209-v2.16.swu
root@venus:/data# /opt/victronenergy/swupdate-scripts/check-updates.sh -swu /data/venus-swu-raspberrypi2-20180625141209-v2.16.swu 
*** Checking for updates ***
arguments: -swu /data/venus-swu-raspberrypi2-20180625141209-v2.16.swu
Updating to /data/venus-swu-raspberrypi2-20180625141209-v2.16.swu
Starting swupdate to install version /data/venus-swu-raspberrypi2-20180625141209-v2.16.swu ...
software set: stable mode: copy2
Swupdate v2016.07.0

Licensed under GPLv2. See source distribution for detailed copyright notices.

Registered handlers:
	archive
	tar
	raw
	rawfile
	uboot
[NOTIFY] : SWUPDATE running :  [extract_sw_description] : Found file:
	filename sw-description
	size 786
	checksum 0xcc31 VERIFIED

Main loop Daemon
Version 0.1.0
[NOTIFY] : SWUPDATE running :  [parse_images] : Found compressed Image  : venus-image-raspberrypi2.ext4.gz in device : /dev/mmcblk0p3 for handler raw (installed from stream) 

[NOTIFY] : SWUPDATE running :  [parse_uboot] : U-Boot var: version = 2

[NOTIFY] : SWUPDATE running :  [cpio_scan] : Found file:
	filename venus-image-raspberrypi2.ext4.gz
	size 70872748
	REQUIRED

ERROR core/cpio_utils.c : cpio_scan : 382 : Checksum verification failed for venus-image-raspberrypi2.ext4.gz: 334ca555 != e4d180dd

[NOTIFY] : SWUPDATE failed [0] ERROR core/cpio_utils.c : cpio_scan : 382 : Checksum verification failed for venus-image-raspberrypi2.ext4.gz: 334ca555 != e4d180dd

Error, do_swupdate stopped with exitcode 1, unlock and exit.
root@venus:/data# 
```

mpvader, 2018-08-23:
It works fine here on a rpi. Also the md5sum is different than yours, while filename is the same.

md5sum:
```
# wget https://updates.victronenergy.com/feeds/venus/release/images/raspberrypi2/venus-swu-raspberrypi2-20180625141209-v2.16.swu

...

# md5sum ./venus-swu-raspberrypi2-20180625141209-v2.16.swu 
6998183fcf0fe333a11b9bdc99b167d0  ./venus-swu-raspberrypi2-20180625141209-v2.16.swu
```

update log:
```
# /opt/victronenergy/swupdate-scripts/check-updates.sh -update -force
*** Checking for updates ***
arguments: -update -force
Retrieving latest version (feed=release)...
installed: 20180312111209 v2.20~12
available: 20180625141209 v2.16
Starting swupdate to install version 20180625141209 v2.16 ...
software set: stable mode: copy2
Swupdate v2016.10.0

Licensed under GPLv2. See source distribution for detailed copyright notices.

Registered handlers:
	archive
	tar
	raw
	rawfile
	uboot
Main loop Daemon
[NOTIFY] : SWUPDATE running :  [network_thread] : Incoming network request: processing...
[NOTIFY] : SWUPDATE downloading :  
[NOTIFY] : SWUPDATE started :  Software Update started !
[NOTIFY] : SWUPDATE running :  [download_from_url] : Image download started : https://updates.victronenergy.com/feeds/venus/release/images/raspberrypi2/venus-swu-raspberrypi2-20180625141209-v2.16.swu
[NOTIFY] : SWUPDATE running :  [extract_file_to_tmp] : Found file:
	filename sw-description
	size 786
Version 0.1.0
[NOTIFY] : SWUPDATE running :  [parse_images] : Found compressed Image  : venus-image-raspberrypi2.ext4.gz in device : /dev/mmcblk0p3 for handler raw (installed from stream) 

[NOTIFY] : SWUPDATE running :  [parse_uboot] : U-Boot var: version = 2

[NOTIFY] : SWUPDATE running :  [extract_files] : Found file:
	filename venus-image-raspberrypi2.ext4.gz
	size 70872748 required
[NOTIFY] : SWUPDATE running :  [extract_files] : Installing STREAM venus-image-raspberrypi2.ext4.gz, 70872748 bytes

[NOTIFY] : SWUPDATE running :  [install_single_image] : Found installer for stream venus-image-raspberrypi2.ext4.gz raw
[NOTIFY] : SWUPDATE running :  [download_info] : Received : 2572018 / 77886976
[NOTIFY] : SWUPDATE running :  [download_info] : Received : 6913778 / 77886976
[NOTIFY] : SWUPDATE running :  [download_info] : Received : 11239154 / 77886976
[NOTIFY] : SWUPDATE running :  [download_info] : Received : 12730098 / 77886976
[NOTIFY] : SWUPDATE running :  [download_info] : Received : 12861170 / 77886976
[NOTIFY] : SWUPDATE running :  [download_info] : Received : 17268466 / 77886976
[NOTIFY] : SWUPDATE running :  [download_info] : Received : 21642994 / 77886976
[NOTIFY] : SWUPDATE running :  [download_info] : Received : 26017522 / 77886976
[NOTIFY] : SWUPDATE running :  [download_info] : Received : 30310130 / 77886976
[NOTIFY] : SWUPDATE running :  [download_info] : Received : 34225906 / 77886976
[NOTIFY] : SWUPDATE running :  [download_info] : Received : 38338290 / 77886976
[NOTIFY] : SWUPDATE running :  [download_info] : Received : 41959154 / 77886976
[NOTIFY] : SWUPDATE running :  [download_info] : Received : 45596402 / 77886976
[NOTIFY] : SWUPDATE running :  [download_info] : Received : 49905394 / 77886976
[NOTIFY] : SWUPDATE running :  [download_info] : Received : 54198002 / 77886976
[NOTIFY] : SWUPDATE running :  [download_info] : Received : 58261234 / 77886976
[NOTIFY] : SWUPDATE running :  [download_info] : Received : 62553842 / 77886976
[NOTIFY] : SWUPDATE running :  [download_info] : Received : 66846450 / 77886976
[NOTIFY] : SWUPDATE running :  [download_info] : Received : 70827762 / 77886976
[NOTIFY] : SWUPDATE running :  [extract_files] : END INSTALLING STREAMING
[NOTIFY] : SWUPDATE running :  [extract_files] : Found file:
	filename venus-boot-image-raspberrypi2.vfat.gz
	size 7012773 Not required: skipping
[NOTIFY] : SWUPDATE running :  [download_info] : Received : 70958834 / 77886976
[NOTIFY] : SWUPDATE running :  [download_info] : Received : 75251442 / 77886976
[NOTIFY] : SWUPDATE running :  [network_initializer] : Valid image found: copying to FLASH
[NOTIFY] : SWUPDATE running :  Installation in progress
[NOTIFY] : SWUPDATE running :  [update_uboot_env] : Updating U-boot environment
[NOTIFY] : SWUPDATE running :  [download_from_url] : Success : 77886976 bytes
[NOTIFY] : SWUPDATE successful ! SWUPDATE successful !
[NOTIFY] : SWUPDATE running :  [network_initializer] : Main thread sleep again !
[NOTIFY] : No SWUPDATE running :  Waiting for requests...
Main loop Daemon
do_swupdate completed OK. Rebooting
```

xychix, 2018-08-23:
I'm afraid it's not the 2.16 image that is the issue but the 2.11 (or my specific install).

```
root@venus:~# md5sum /tmp/venus-swu-raspberrypi2-20180625141209-v2.16.swu 
6998183fcf0fe333a11b9bdc99b167d0  /tmp/venus-swu-raspberrypi2-20180625141209-v2.16.swu
```
```
root@venus:~# /opt/victronenergy/swupdate-scripts/check-updates.sh -force -swu /tmp/venus-swu-raspberrypi2-20180625141209-v2.16.swu 
*** Checking for updates ***
arguments: -force -swu /tmp/venus-swu-raspberrypi2-20180625141209-v2.16.swu
Updating to /tmp/venus-swu-raspberrypi2-20180625141209-v2.16.swu
Starting swupdate to install version /tmp/venus-swu-raspberrypi2-20180625141209-v2.16.swu ...
software set: stable mode: copy2
Swupdate v2016.07.0

Licensed under GPLv2. See source distribution for detailed copyright notices.

Registered handlers:
	archive
	tar
	raw
	rawfile
	uboot
Main loop Daemon
[NOTIFY] : SWUPDATE running :  [extract_sw_description] : Found file:
	filename sw-description
	size 786
	checksum 0xcc31 VERIFIED

Version 0.1.0
[NOTIFY] : SWUPDATE running :  [parse_images] : Found compressed Image  : venus-image-raspberrypi2.ext4.gz in device : /dev/mmcblk0p3 for handler raw (installed from stream) 

[NOTIFY] : SWUPDATE running :  [parse_uboot] : U-Boot var: version = 2

[NOTIFY] : SWUPDATE running :  [cpio_scan] : Found file:
	filename venus-image-raspberrypi2.ext4.gz
	size 70872748
	REQUIRED

[NOTIFY] : SWUPDATE running :  [cpio_scan] : Found file:
	filename venus-boot-image-raspberrypi2.vfat.gz
	size 7012773
	not required

[NOTIFY] : SWUPDATE running :  [install_single_image] : Found installer for stream venus-image-raspberrypi2.ext4.gz raw
ERROR core/cpio_utils.c : copyfile : 147 : gunzip failure -5 (errno 28) -- aborting

[NOTIFY] : SWUPDATE failed [0] ERROR core/cpio_utils.c : copyfile : 147 : gunzip failure -5 (errno 28) -- aborting

[NOTIFY] : SWUPDATE running :  [install_single_image] : Installer for raw not successful !
Software updated failed
Error, do_swupdate stopped with exitcode 1, unlock and exit.
root@venus:~# 
```

mpvader, 2018-08-23:
Hi again, you mean that you've redownloaded it? The text shows a different md5sum than before.

And yes, its not possible to install v2.16 on top of v2.11 for a raspberry pi. Izak explained on the mailing list:

> Is this on a Raspberry Pi, or one of the others? The partition sizes were changed and the kernel was upgraded some times after 2.11, so the initial releases unfortunately need to be updated be reflashing and restoring the settings.

> For any of the other platforms this should just work.

xychix, 2018-08-23:
Ah, sorry missed that post. Will search for another sdcard and do a clean install. Assume it will pick the MAC address for unique ID so VRM should keep working after the sdcard swap?

mpvader, 2018-08-23:
Yes, though you'll need to reset a token. VRM Portal will show you a message about that once the new install has been sending data for the first time. success!

closed 2018-08-23 by mpvader

xychix, 2018-08-23:
thanks, I'll reinstall and place back the stuff in /data/    that should possibly do the trick.