victronenergy/venus #322: QA v2.20~57 for CCGX

mpvader, 2018-08-27:
Another extensive QA, this time because switch to kernel 4.14. There are no other changes to test.

### Other machines

- [ ] Other machines (briefly login to Venus GX, check that it boots / updates.

#### general check on various test systems:

- [ ] svstat / vrm / gui / remote console

#### WiFi:

- [x] Zyxel NWD2105 part number BPP900100100 - - USB 0586:341e - RT chipset 3070, rev 0201, RF chipset 0005
- [ ] Startech USB300WN2X2D part number BPP900200100 - - USB 148f:3572 - RT chipset 3572, rev 0223, RF chipset 0009
- [ ] Nano USB part number BPP900100200 - - USB 148f:5370 - RT chipset 5390, rev 0502, RF chipset 5370
- [ ] Gembird WNP-UA-002 part number BPP900200200 - - USB 148f:3072 - RT chipset 3071, rev 021c, RF chipset 0008: NOT TESTED. BUT USES SAME DRIVER: OK.

#### Other USB:

- [ ] Helmsjlund RS485 isolated USB:
- [ ] GX GSM

USB/SD card log function & automounting

#### USB & sdcard Storage

1. Insert media 
2. disconnect network (ethernet/wifi)
3. check in Settings -> VRM Logger that it recognizes the stick and writes to it

- [ ] USB stick
- [ ] SD card

#### Updating
- [ ] Update online
- [ ] Update offline (sdcard)
- [ ] Boot into the backup image

#### RTC
- [ ] reading time from rtc (check boot log)
- [ ] writing time to rtc

#### Other
- [ ] Crond: automatic update
- [ ] Diskspace
```

copy result here
```

#### Special images
- [ ] Installer image
- [ ] Upgrade image

mpvader, 2018-08-28:
Seen om Matthijs CCGX (not reproducable, and WiFi did still work):

A quick google search yielded this link, might be related, or not:
https://lkml.org/lkml/2018/8/21/396

```
[   25.812164] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[   30.079192] 
[   30.079285] ======================================================
[   30.079327] WARNING: possible circular locking dependency detected
[   30.079374] 4.14.67 #1 Not tainted
[   30.079404] ------------------------------------------------------
[   30.079443] wpa_supplicant/782 is trying to acquire lock:
[   30.079480]  ("%s"wiphy_name(local->hw.wiphy)){+.+.}, at: [<c0142268>] flush_work+0x180/0x280
[   30.079582] 
[   30.079582] but task is already holding lock:
[   30.079623]  (&local->mtx){+.+.}, at: [<bf0d5124>] ieee80211_request_scan+0x20/0x44 [mac80211]
[   30.080645] 
[   30.080645] which lock already depends on the new lock.
[   30.080645] 
[   30.080701] 
[   30.080701] the existing dependency chain (in reverse order) is:
[   30.080750] 
[   30.080750] -> #2 (&local->mtx){+.+.}:
[   30.080835]        mutex_lock_nested+0x1c/0x24
[   30.081355]        ieee80211_scan_work+0x20/0x744 [mac80211]
[   30.081424]        process_one_work+0x2a0/0x74c
[   30.081464]        worker_thread+0x50/0x5a8
[   30.081504]        kthread+0x15c/0x164
[   30.081548]        ret_from_fork+0x14/0x20
[   30.081580] 
[   30.081580] -> #1 ((&(&local->scan_work)->work)){+.+.}:
[   30.081654]        worker_thread+0x50/0x5a8
[   30.081690]        kthread+0x15c/0x164
[   30.081724]        ret_from_fork+0x14/0x20
[   30.081753] 
[   30.081753] -> #0 ("%s"wiphy_name(local->hw.wiphy)){+.+.}:
[   30.081825]        flush_work+0x1a8/0x280
[   30.081860]        __cancel_work_timer+0x100/0x1d4
[   30.082258]        __ieee80211_start_scan+0x4e0/0xc54 [mac80211]
[   30.082612]        ieee80211_request_scan+0x2c/0x44 [mac80211]
[   30.083500]        nl80211_trigger_scan+0x6cc/0x960 [cfg80211]
[   30.083587]        genl_rcv_msg+0x2d8/0x3a4
[   30.083628]        netlink_rcv_skb+0x9c/0xe8
[   30.083666]        genl_rcv+0x24/0x34
[   30.083702]        netlink_unicast+0x16c/0x1f0
[   30.083739]        netlink_sendmsg+0x358/0x39c
[   30.083787]        sock_sendmsg+0x14/0x24
[   30.083824]        ___sys_sendmsg+0x204/0x214
[   30.083864]        __sys_sendmsg+0x40/0x6c
[   30.083906]        ret_fast_syscall+0x0/0x28
[   30.083938] 
[   30.083938] other info that might help us debug this:
[   30.083938] 
[   30.083993] Chain exists of:
[   30.083993]   "%s"wiphy_name(local->hw.wiphy) --> (&(&local->scan_work)->work) --> &local->mtx
[   30.083993] 
[   30.084101]  Possible unsafe locking scenario:
[   30.084101] 
[   30.084143]        CPU0                    CPU1
[   30.084173]        ----                    ----
[   30.084202]   lock(&local->mtx);
[   30.084238]                                lock((&(&local->scan_work)->work));
[   30.084291]                                lock(&local->mtx);
[   30.084335]   lock("%s"wiphy_name(local->hw.wiphy));
[   30.084378] 
[   30.084378]  *** DEADLOCK ***
[   30.084378] 
[   30.084428] 4 locks held by wpa_supplicant/782:
[   30.084458]  #0:  (cb_lock){++++}, at: [<c05cdcc8>] genl_rcv+0x14/0x34
[   30.084528]  #1:  (genl_mutex){+.+.}, at: [<c05cde34>] genl_rcv_msg+0x14c/0x3a4
[   30.084601]  #2:  (rtnl_mutex){+.+.}, at: [<bf030d9c>] nl80211_pre_doit+0xec/0x180 [cfg80211]
[   30.085126]  #3:  (&local->mtx){+.+.}, at: [<bf0d5124>] ieee80211_request_scan+0x20/0x44 [mac80211]
[   30.085904] 
[   30.085904] stack backtrace:
[   30.085956] CPU: 0 PID: 782 Comm: wpa_supplicant Not tainted 4.14.67 #1
[   30.086000] Hardware name: Generic AM3517 (Flattened Device Tree)
[   30.086071] [<c010e13c>] (unwind_backtrace) from [<c010c6a8>] (show_stack+0x10/0x14)
[   30.086140] [<c010c6a8>] (show_stack) from [<c01699f4>] (print_circular_bug.constprop.20+0x2e0/0x2e8)
[   30.086206] [<c01699f4>] (print_circular_bug.constprop.20) from [<c0167ce4>] (__lock_acquire+0x15d8/0x1a08)
[   30.086270] [<c0167ce4>] (__lock_acquire) from [<c0168a28>] (lock_acquire+0xe0/0x214)
[   30.086336] [<c0168a28>] (lock_acquire) from [<c0142290>] (flush_work+0x1a8/0x280)
[   30.086397] [<c0142290>] (flush_work) from [<c0144300>] (__cancel_work_timer+0x100/0x1d4)
[   30.086887] [<c0144300>] (__cancel_work_timer) from [<bf0d3944>] (__ieee80211_start_scan+0x4e0/0xc54 [mac80211])
[   30.087589] [<bf0d3944>] (__ieee80211_start_scan [mac80211]) from [<bf0d5130>] (ieee80211_request_scan+0x2c/0x44 [mac80211])
[   30.088581] [<bf0d5130>] (ieee80211_request_scan [mac80211]) from [<bf032a74>] (nl80211_trigger_scan+0x6cc/0x960 [cfg80211])
[   30.089051] [<bf032a74>] (nl80211_trigger_scan [cfg80211]) from [<c05cdfc0>] (genl_rcv_msg+0x2d8/0x3a4)
[   30.089141] [<c05cdfc0>] (genl_rcv_msg) from [<c05cd0c4>] (netlink_rcv_skb+0x9c/0xe8)
[   30.089206] [<c05cd0c4>] (netlink_rcv_skb) from [<c05cdcd8>] (genl_rcv+0x24/0x34)
[   30.089266] [<c05cdcd8>] (genl_rcv) from [<c05cc890>] (netlink_unicast+0x16c/0x1f0)
[   30.089326] [<c05cc890>] (netlink_unicast) from [<c05ccd38>] (netlink_sendmsg+0x358/0x39c)
[   30.089399] [<c05ccd38>] (netlink_sendmsg) from [<c057dba8>] (sock_sendmsg+0x14/0x24)
[   30.089463] [<c057dba8>] (sock_sendmsg) from [<c057e638>] (___sys_sendmsg+0x204/0x214)
[   30.089527] [<c057e638>] (___sys_sendmsg) from [<c057f3e8>] (__sys_sendmsg+0x40/0x6c)
[   30.089595] [<c057f3e8>] (__sys_sendmsg) from [<c0107ce0>] (ret_fast_syscall+0x0/0x28)
```

mpvader, 2018-08-28:
previous comment can be ignored: its a known issue and has been there forever.

jhofstee, 2018-08-28:
https://lkml.org/lkml/2018/8/22/205 suppressed this, but it is harmless

vejpasop, 2018-09-03:
VE.Bus Standby line tested OK (tested on CCGX and Venus GX).
VE.Bus Panel detect is connected to GND or controlled by the MK3. No Linux kernel dependency here.

mpvader, 2018-09-06:
GX GSM: Various users are testing the GX GSM including gps. I have heard no more complaints since pushing. v2.20~60

mpvader, 2019-01-07:
Superseded by #395

closed 2019-01-07 by mpvader