victronenergy/venus #337: Generator start-stop: improve running-hour counter

mpvader, 2018-09-04:
- If the generator is started manually (on the generator itself), the counter will not run; would be better if it would
- If the generator is requested to start, but does not start for whatever reason, the counter does run. Would be better if it would not.

Instead of improving this in generator start/stop; the alternative is to do this by looking at Quattro AC-in or even System ac-source.

mpvader, 2023-04-28:
Hi both - is this still relevant?

warwickchapman, 2024-02-01:
I believe this is very important to the hybrid generator people and I believe it is being worked on, so perhaps this issue is a duplicate.

warwickchapman, 2024-02-01:
Per @ReinvdZee :

The generator runtime is accumulated when it operates in either of the following states: `States.RUNNING`, `States.WARMUP`, `States.COOLDOWN`, `States.STOPPING` (see `startstop.py`). Both the cooldown time and the new generator stop-time are within the `STOPPING` state.

My comment:

1. If the start is triggered but no AC input detected and ERROR state results, then no runtime should be logged.
2. If the start is triggered and an AC input is detected but is rejected (voltage or frequency out of range), the running time should be logged.

ReinvdZee, 2024-08-08:
**Connected gensets**
For connected gensets, we can probably use `/Engine/OperatingHours` or `/Ac/L1/Voltage`. Since the paths provided by the genset service clearly reflect the genset's state, I'll use those for counting local runtime of connected gensets.
@philipptrenz do you know if `/Ac/L1/Voltage` or `/Engine/OperatingHours` or maybe `/Engine/Speed` are always there?

**Relay controlled gensets**
For relay-controlled gensets, the `/NoGeneratorAtAcIn` alarm is triggered when the genset is supposed to run and the AC input is configured as generator (`com.victronenergy.system/Ac/ActiveIn/Source` == 2), but `/Ac/ActiveIn/Connected` is not true.  

I don't expect we can use `/Ac/ActiveIn/Connected` to start/stop counting runtime, as this will probably not be equal to 1 when the input source is rejected (due to voltage/frequency for example). I'll see if we can use `/Ac/ActiveIn/L1/V` being nonzero for this.

In all cases we need to take warm-up and cool-down periods into account.

Of course, this still has some limitations as runtime may not be properly counted when the wiring is incorrect, breakers are open or the system is not configured correctly. 

**User interface**
And do we need an additional state for when the generator is running when the GX did not request it to? In case of a genset, this currently looks like this: 
[image]

For the relay-controlled genset, there is no feedback at all at the moment.
If we can count runtime when the genset is running when it is not supposed to (from GX perspective), we can also display this.

kwindrem, 2024-08-08:
FYI, one of the generator modifications I've made and included now in GuiMods is to factor in the generator running state in counting run time. I use the generator digital input if defined or AC frequency if the input type is generator as the trigger. If I can't determine the running state via either mechanism I accumulate time like the stock code does (if the GX device it telling the generator to run). Note my current code requires that the GX device expects the generator to be running to accumulate time. So ...

In addition I sync the GX manual start with the external generator state so if the generator is started externally, the GX device goes into manual run. If the generator is stopped externally, I turn off manual run if it's active. If auto run is active I display an "override" message in the generator status.

philipptrenz, 2024-08-19:
My thoughts on this: 

- Yes, manual runs should definitely also be counted
- For connected gensets: From my perspective, the best way to figure out if it is running, is to check if its `/StatusCode` is between 1 and 9. The status code is already implicitly mandatory (as used by the GUI's) and the producers take care that it is available. For some DSE units, it is determined based on engine speed. But as availability of certain readings differ from model to model, I think we should not put the logic into `dbus-generator-starter`, but keep it to the producers to decide how to provide that.
- For relay-controlled gensets, we have this digital input config, as Kevin mentioned. I would expect this to improve the  counters and we can advise customers to use this solution to improve the runtime counter. Figuring it out via Multi/Quattro frequency or voltage would be a nice to have, but I would still advise users to actually use the digital input, as it is the most accurate – even if the genset is used in some other weird use cases not being connected to a inverter/charger. E.g. custom DC generator configs, which I have already seen in the wild.

ReinvdZee, 2024-08-19:
> From my perspective, the best way to figure out if it is running, is to check if its /StatusCode is between 1 and 9. 

Sound great! let's use that.

> For relay-controlled gensets, we have this digital input config, as Kevin mentioned.

Sure that is the most reliable way of tracking run time. Still, we should improve run time tracking without the digital input.

kwindrem, 2024-08-19:
As mentioned, the generator digital input provides an easy and reliable way to determine generator running status.

I added the AC input detection to the logic on request from an installer that did not want to add a relay to the generator output to provide drive the digital input. It works only if the generator is actually connected to the AC input of the Multi/Quattro so it would not be accurate if there's a transfer switch ahead of the Multi AC input.

DC generators are more problematic since they will simply tie in to the DC bus and not provide separate voltage sensing.

I'd suggest making the generator digital input a requirement for relay start generators. As I did in GuiMods, if not provided, simply accumulate time when the GX device tells the generator to run regardless of the actual running state of the genset.

ReinvdZee, 2024-08-21:
I discussed this with @philipptrenz, we decided the best implementation for now would be:

- Count running time for connected gensets when `/StatusCode` is between 1 and 9. 
- Count running time for relay controlled gensets when the digital input is configured as "Generator". 
- For now, leave the running time count mechanism for relay controlled gensets without the digital input as is. If the user needs better runtime counting, use the digital input.

ReinvdZee, 2024-09-18:
I implemented it as discussed. For the relay-started generator, we should restrict the user to be able to only use 1 digital input at a time. `dbus_generator` will simply take the first correctly configured digital input to count run time with, but to prevent confusion, there should only be 1 input configured as "Generator" at a time. The easiest to do this is to only allow this function on one digital input.

@mpvader Agreed if we only let digital input 1 be configurable as Generator?

ReinvdZee, 2024-09-18:
And there's also some other GUI-related changes: 

- Rename the digital input function "Generator" to "Relay controlled genset"
- Rename the relay function "Generator start/stop" to "Relay controlled genset" (already done [here](https://github.com/victronenergy/gui/commit/5e2956fed5205dc4d0eb4a2df4dc470cfc3089c4))
- Move the generator start/stop menu out of the settings menu and merge it with the "Generator" device (Will then be called "Relay controlled genset"). The device will be there as soon as the relay is configured as "Relay controlled genset" and contains the digital input as well. If only the digital input is configured (and not the relay), then no device will show up, as the digital input cannot be used without the relay generator (correct?).

Any objections? 
@mpvader @philipptrenz

philipptrenz, 2024-09-18:
> The device will be there as soon as the relay is configured as "Relay controlled genset" and contains the digital input as well. If only the digital input is configured (and not the relay), then no device will show up, as the digital input cannot be used without the relay generator (correct?).

@ReinvdZee Yes, that's how I would expect it.

philipptrenz, 2024-09-18:
@kwindrem In the long run, it would indeed be valuable having a more flexible approach how functions are assigned to I/O. But I think we need an intermediate solution for now. 

To not break existing installations, I would suggest to just remove the option from the dropdown of Digital Input 2-4 (for example), so that existing settings keep working while for new installations, the "Relay controlled genset" input is only available on Digital Input 1.

philipptrenz, 2024-09-18:
Discussed from above is also what was proposed in #1335

mpvader, 2024-09-30:
hey @ReinvdZee , @kwindrem , @philipptrenz ,

wrt:

> I implemented it as discussed. For the relay-started generator, we should restrict the user to be able to only use 1 digital input at a time. dbus_generator will simply take the first correctly [...]

I would do this in the way that it costs Rein the least amount of time. Which I assume is allowing it on all inputs, and taking the first correctly configured digital input.

kwindrem, 2024-09-30:
> Which I assume is allowing it on all inputs, and taking the first correctly configured digital input.
This is what I've done for detecting the generator digital input and the External transfer switch digital input in GuiMods (previously GenneratorConnector). If more than one is configured, they are ignored once the first is found.

closed 2024-10-02 by jhofstee