victronenergy/venus #339: Make adding a VE.Bus Inverter to a Multi/Quattro system work well

mpvader, 2018-09-05:
- [ ] Use the main VE.Bus system in systemcalc and on the overview (select on lowest instance)
- [ ] Use all VE.Bus systems in kWh calculations
- [ ] Use the main VE.Bus system in Generator start/stop logic & genset runtime calculation
- [ ] Verify that what Systemcalc does [wrt. totalizing battery current](https://github.com/victronenergy/dbus-systemcalc-py/blob/master/dbus_systemcalc.py#L364) make sense
- [ ] Add unit tests for the various calculations
------

Possible (typical?) configurations: three-phase main system, and a small extra one, mostly an Inverter only, rather than a Multi or Quattro.

The reason to focus on an extra Inverter only, rather than a Multi or Quattro, is also AC Input (type) handling.

mpvader, 2018-09-05:
kwhdeltas already [uses all VE.Bus systems](https://git.victronenergy.com/ccgx/dbus_vrm/blob/master/kwhdeltas.py#L225) in its calculations. Also verified with an actual system (17160): it has a second VE.Bus Inverter, that powers a constant load over the night; the main three-phase system has all loads switched off during the night.

On September 4th, the VE.Bus system was connected to the Venus GX with an MK3-USB: in the kWh excel download from VRM you can see it showing up.

First box ticked.

mpvader, 2018-09-06:
Because systemcalc only looks at the VE.Bus system with lowest instance, the power under AC Consumption as shown on VRM Portal will *not* include the output power of the second system.

[image]

mpvader, 2018-10-03:
> Use the main VE.Bus system in systemcalc and on the overview (select on lowest instance)

that was added in v2.17

mpvader, 2018-10-03:
@izak how do systemcalc / hub4control select which VE.Bus service to use for ESS setpoints?

mpvader, 2018-10-03:
@jepefe how does generator start/stop select which VE.Bus service to use for VE.Bus specific data that it needs? And what data does it consume then directly from the VE.Bus service, and what (AC) data does it get from systemcalc?

jepefe, 2018-10-05:
VE.Bus service that is used by generator start/stop is picked from systemcalc `/VebusService`

Data consumed from systemcalc:
```
/Ac/Consumption/L1/Power
/Ac/Consumption/L2/Power
/Ac/Consumption/L3/Power
/Ac/ActiveIn/Source #To know if the active input is configured as Grid, Generator, Shore...
/VebusService
```

izak, 2018-10-17:
> @izak how do systemcalc / hub4control select which VE.Bus service to use for ESS setpoints?

Present situation: Only one is used, and whichever one showed up most recently (ie last) replaces the previous one. Most likely not what we want.

izak, 2019-02-14:
hub4control now gets the vebus service from systemcalc, as was initially suggested. It turned out to not be as difficult as I thought. It's in version v1.2.9.

vejpasop, 2019-02-14:
Added VE,Bus Inverter product name support to mk2-dbus v2.80

When an inverter is connected the MultiPlus product name is converted as follows.
- Replace "MultiPlus" with "Phoenix Inverter"
- Remove the current transfer capability form the product name by deleting everything from the second occurrence of a forward slash in the product name.

 Example:
 nameIn = "MultiPlus 12/3000/120-16"
 nameOut = "Phoenix Inverter 12/3000"

Loop through test output of all VE.Bus products in products.c
```
Converted 1900, MultiPlus 12/3000/120-50 -> Phoenix Inverter 12/3000
Converted 1901, MultiPlus 12/3000/120-30 -> Phoenix Inverter 12/3000
Converted 1906, MultiPlus Compact 12/1600/70-16 -> Phoenix Inverter Compact 12/1600
Converted 1907, MultiPlus Compact 12/1200/50-16 -> Phoenix Inverter Compact 12/1200
Converted 1908, MultiPlus Compact 12/2000/80-30 -> Phoenix Inverter Compact 12/2000
Converted 1909, MultiPlus 12/3000/120-16 -> Phoenix Inverter 12/3000 
Converted 1910, MultiPlus 24/3000/70-50 -> Phoenix Inverter 24/3000 
Converted 1911, MultiPlus 24/3000/70-30 -> Phoenix Inverter 24/3000 
Converted 1912, MultiPlus 24/3000/70-16 -> Phoenix Inverter 24/3000 
Converted 1916, MultiPlus Compact 24/1600/40-16 -> Phoenix Inverter Compact 24/1600
Converted 1917, MultiPlus Compact 24/1200/25-16 -> Phoenix Inverter Compact 24/1200
Converted 1918, MultiPlus Compact 24/2000/50-30 -> Phoenix Inverter Compact 24/2000
Converted 1920, MultiPlus 48/3000/35-50 -> Phoenix Inverter 48/3000 
Converted 1921, MultiPlus 48/3000/35-30 -> Phoenix Inverter 48/3000 
Converted 1922, MultiPlus 48/3000/35-16 -> Phoenix Inverter 48/3000 
Converted 1949, MultiPlus 24/5000/120-50 -> Phoenix Inverter 24/5000 
Converted 1959, MultiPlus 48/5000/70-50 -> Phoenix Inverter 48/5000 
Converted 2002, MultiPlus 12/3000/120-50 120V -> Phoenix Inverter 12/3000 
Converted 2008, MultiPlus Compact 12/2000/80-50 120V -> Phoenix Inverter Compact 12/2000
Converted 2012, MultiPlus 24/3000/70-50 120V -> Phoenix Inverter 24/3000 
Converted 2018, MultiPlus Compact 24/2000/50-50 120V -> Phoenix Inverter Compact 24/2000
Converted 2062, MultiPlus Compact 12/2000/80-50(30)-120/240V -> Phoenix Inverter Compact 12/2000
Converted 2072, MultiPlus Compact 24/2000/50-50(30)-120/240V -> Phoenix Inverter Compact 24/2000
Converted 2600, MultiPlus 12/3000/120-50 -> Phoenix Inverter 12/3000 
Converted 2606, MultiPlus Compact 12/1600/70-16 -> Phoenix Inverter Compact 12/1600
Converted 2607, MultiPlus Compact 12/1200/50-16 -> Phoenix Inverter Compact 12/1200
Converted 2608, MultiPlus Compact 12/2000/80-30 -> Phoenix Inverter Compact 12/2000
Converted 2609, MultiPlus 12/3000/120-16 -> Phoenix Inverter 12/3000 
Converted 2610, MultiPlus 24/3000/70-50 -> Phoenix Inverter 24/3000 
Converted 2612, MultiPlus 24/3000/70-16 -> Phoenix Inverter 24/3000 
Converted 2614, MultiPlus 24/5000/120-100 -> Phoenix Inverter 24/5000 
Converted 2616, MultiPlus Compact 24/1600/40-16 -> Phoenix Inverter Compact 24/1600
Converted 2617, MultiPlus Compact 24/1200/25-16 -> Phoenix Inverter Compact 24/1200
Converted 2618, MultiPlus Compact 24/2000/50-30 -> Phoenix Inverter Compact 24/2000
Converted 2620, MultiPlus 48/3000/35-50 -> Phoenix Inverter 48/3000 
Converted 2622, MultiPlus 48/3000/35-16 -> Phoenix Inverter 48/3000 
Converted 2624, MultiPlus 48/5000/70-100 -> Phoenix Inverter 48/5000 
Converted 2625, MultiPlus-II 48/3000/35-32 -> Phoenix Inverter-II 48/3000/
Converted 2626, MultiPlus-II 48/5000/70-48 -> Phoenix Inverter-II 48/5000/
Converted 2628, MultiPlus-II 48/3000/35-32 -> Phoenix Inverter-II 48/3000/
Converted 2629, MultiPlus-II 48/3000/35-32 -> Phoenix Inverter-II 48/3000/
Converted 2649, MultiPlus 24/5000/120-50 -> Phoenix Inverter 24/50000
Converted 2659, MultiPlus 48/5000/70-50 -> Phoenix Inverter 48/50000
Converted 2660, MultiPlus 12/500/20-16 -> Phoenix Inverter 12/5000
Converted 2661, MultiPlus 12/800/35-16 -> Phoenix Inverter 12/8000
Converted 2662, MultiPlus 12/1200/50-16 -> Phoenix Inverter 12/1200
Converted 2663, MultiPlus 12/1600/70-16 -> Phoenix Inverter 12/1600
Converted 2665, MultiPlus 24/500/10-16 -> Phoenix Inverter 24/5000
Converted 2666, MultiPlus 24/800/16-16 -> Phoenix Inverter 24/8000
Converted 2667, MultiPlus 24/1200/25-16 -> Phoenix Inverter 24/1200
Converted 2668, MultiPlus 24/1600/40-16 -> Phoenix Inverter 24/1600
Converted 2670, MultiPlus 48/500/6-16 -> Phoenix Inverter 48/5000
Converted 2671, MultiPlus 48/800/8-16 -> Phoenix Inverter 48/8000
Converted 2672, MultiPlus 48/1200/12-16 -> Phoenix Inverter 48/1200
Converted 2673, MultiPlus 48/1600/20-16 -> Phoenix Inverter 48/1600
Converted 2702, MultiPlus 12/3000/120-50 120V -> Phoenix Inverter 12/3000
Converted 2708, MultiPlus Compact 12/2000/80-50 120V -> Phoenix Inverter Compact 12/2000
Converted 2712, MultiPlus 24/3000/70-50 120V -> Phoenix Inverter 24/3000 
Converted 2718, MultiPlus Compact 24/2000/50-50 120V -> Phoenix Inverter Compact 24/2000
```

mpvader, 2019-02-14:
Nice @vejpasop !

izak, 2019-03-05:
> Verify that the alarms work for both, and the configuration of the alarms.

This works partially.
Good:
* Alarms for both show up on the GUI
* Alarms for both are logged to VRM and show up on the alarm log page

Bad:
* The `ALARM` tag only shows up on VRM for the first vebus inverter.
* Alarm configuration is shared, changing notification settings for one also changes it for the other.

mpvader, 2021-08-17:
Above "bad" things are accepted.

and this feature is documented for customers as well as used by them: closing.

closed 2021-08-17 by mpvader