victronenergy/venus #345: mqtt-rpc: fix veconf file upload to browser retry issue

mpvader, 2018-09-10:
mqtt-rpc transmits that file in multiple chunks, but does not have a retry mechanism, and also does not use QoS.

So, getting that the file as a user is only possible if both links maintain solid during the upload process:
- the link between Venus-device and the cloud broker.
- the link between the cloud broker.

- [ ] The chunking is not necessary: mqtt can handle one large chunk.
- [ ] The send a chunk & wait for ack mechanism makes the process unnecessary long (think of a high latency link); and therefore extra vulnerable for often resetting internet links.

Note that the file-transfer from Celery to Venus-device does have retries implemented. So its better, but when we start sending files in one message we can remove that code: even better.

mpvader, 2018-09-10:
And then mqtt-rpc should use QoS 1 when publishing the file. Doing this, TCP level retries will be used as long as the connection is up, and mqtt retries will take over in case of a connection loss during the upload.

Both from Venus-Device to broker as broker to browser.

I don’t see a downside in using QoS 1 here.

On top of that, Celery and/or Browser can start a clean, though persistent, session when starting the task. Closing it once done.

Correct me if I’m wrong!

elnino-ict, 2019-01-09:
This issue is on hold for now until MTU has been changed.

wiebeytec, 2019-01-09:
Even though it's on-hold, I would like to comment on QoS 1 for the future: I do see a few issues. One was already mentioned by Maurits, and that is that you can't cancel it. Even if the installation of offline for a long time, the message will be received when it connects.

Another issue is that you can't specify a different setting for `clean session` for _connects_ vs _reconnects_, and therefore you need to use `cleansession: false`. This will have side effects.

I don't recall the full specifics, but I think caution is indicated.

Perhaps MQTTv5 will help though.

mauritsvdvijgh, 2019-05-07:
Proposal: Make device-update, vebusremoteconfig read/vebusremoteconfig write take a base64 encoded file as input parameter

Meaning no more seperate file upload and then call device-update or vebusremoteconfig read and no more seperate file download and then call vebusremoteconfig write. This makes these actions atomic (as they are now 1 command instead of 2 commands in quick succession). This also does away with the chunked file upload.

why?
- Atomic commands, either it completely happens or it does not
- Reduced code complexity in 2 code bases, mainly in victronconnect where the commands are working but the code that uses these commands (file upload + firmware update) is not made yet.
- Not sure if the chunked file upload/download adds something. TCP already creates <=MTU size packets and implements acks/retries.

mauritsvdvijgh, 2019-05-27:
https://github.com/victronenergy/mqtt-rpc/commit/90f85b2f0118e8564167e3db3c4614b297e470cb
This has been implemented. Closed it, then I remembered that you close issues when the relevant changes are included in Venus OS.

mpvader, 2019-11-04:
@thlassche lets also put out an egg on this one this week.

mpvader, 2019-11-04:
Izak: please make sure that its in; and then close it. thanks

izak, 2019-11-11:
The one-chunk file upload was in mqtt-rpc v1.28, which was released in Venus 2.30 already. Closing this.

closed 2019-11-11 by izak