victronenergy/venus #349: gui: notify the user of misconfigured settings etc

jhofstee, 2018-09-12:
It would save quite some time and be far more user friendly if it is pointed out that there is a problem with the current configuration / setup etc. So lets first make a list of problems which should be pointed out to the user, the user should be notified when:

- system calc or the relay control is set to a battery which does not exists for a certain amount of time

mpvader, 2019-02-22:
- Shared Voltage Sense enabled for batteries where that is not allowed; (such as byd)

vejpasop, 2019-03-11:
* Detect if all Multi's contain an ESS assistant. Currently (mk2-dbus <= v2.81) if L1 does not contain an ESS assistant and L2 and or L3 does, mk2-dbus even crashes when trying to access a non existing L1 ESS item.

mpvader, 2019-04-03:
For parallel inverter/chargers; report dc voltage per unit; or delivered power per unit when underload. Mice to be part of commissioning a system too.

mpvader, 2019-04-03:
Also dc ripple (for each unit of a large system)

mpvader, 2020-05-12:
- Notify the user when Remote Console on VRM is unable to send out the security token (because port 443 is blocked). It can a bit obscure now why Remote Console on VRM doesn't work in a network that blocks outgoing connections on port 443. Even though VRM logging does work (when in that case configured to use port 80), Remote Console on VRM still will not work. More info [here](https://www.victronenergy.com/live/ccgx:ccgx_faq#q15what_type_of_networking_is_used_by_the_color_control_gx_tcp_and_udp_ports).