victronenergy/venus #350: hung task on Linux ccgx 4.14.67

jhofstee, 2018-09-13:
```
Sep 13 14:25:59 ccgx user.err kernel: [43008.664533] INFO: task kworker/u2:0:5 blocked for more than 120 seconds.
Sep 13 14:25:59 ccgx user.err kernel: [43008.664573]       Not tainted 4.14.67 #1
Sep 13 14:25:59 ccgx user.err kernel: [43008.664584] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
Sep 13 14:25:59 ccgx user.info kernel: [43008.664600] kworker/u2:0    D    0     5      2 0x00000000
Sep 13 14:25:59 ccgx user.info kernel: [43008.664650] Workqueue: events_unbound flush_to_ldisc
Sep 13 14:25:59 ccgx user.warn kernel: [43008.664694] [<c0607cfc>] (__schedule) from [<c060811c>] (schedule+0x40/0xa0)
Sep 13 14:25:59 ccgx user.warn kernel: [43008.664801] [<c060811c>] (schedule) from [<c06083d0>] (schedule_preempt_disabled+0xc/0x10)
Sep 13 14:25:59 ccgx user.warn kernel: [43008.664827] [<c06083d0>] (schedule_preempt_disabled) from [<c0609540>] (__mutex_lock.constprop.4+0xf8/0x324)
Sep 13 14:25:59 ccgx user.warn kernel: [43008.664847] [<c0609540>] (__mutex_lock.constprop.4) from [<c03fd510>] (tty_throttle_safe+0x14/0x7c)
Sep 13 14:25:59 ccgx user.warn kernel: [43008.664885] [<c03fd510>] (tty_throttle_safe) from [<c03fb7c0>] (n_tty_receive_buf_common+0xd8/0x9bc)
Sep 13 14:25:59 ccgx user.warn kernel: [43008.664910] [<c03fb7c0>] (n_tty_receive_buf_common) from [<c03fc0b8>] (n_tty_receive_buf2+0x14/0x1c)
Sep 13 14:25:59 ccgx user.warn kernel: [43008.664933] [<c03fc0b8>] (n_tty_receive_buf2) from [<c03fea38>] (tty_port_default_receive_buf+0x38/0x58)
Sep 13 14:25:59 ccgx user.warn kernel: [43008.664953] [<c03fea38>] (tty_port_default_receive_buf) from [<c03fe364>] (flush_to_ldisc+0x7c/0xa8)
Sep 13 14:25:59 ccgx user.warn kernel: [43008.664984] [<c03fe364>] (flush_to_ldisc) from [<c013c2f4>] (process_one_work+0x1c4/0x45c)
Sep 13 14:25:59 ccgx user.warn kernel: [43008.665007] [<c013c2f4>] (process_one_work) from [<c013c5ec>] (worker_thread+0x60/0x590)
Sep 13 14:25:59 ccgx user.warn kernel: [43008.665029] [<c013c5ec>] (worker_thread) from [<c0141af8>] (kthread+0x148/0x150)
Sep 13 14:25:59 ccgx user.warn kernel: [43008.665060] [<c0141af8>] (kthread) from [<c0107928>] (ret_from_fork+0x14/0x2c)
Sep 13 14:25:59 ccgx user.err kernel: [43008.665315] INFO: task vedirect-dbus:1339 blocked for more than 120 seconds.
Sep 13 14:25:59 ccgx user.err kernel: [43008.665327]       Not tainted 4.14.67 #1
Sep 13 14:25:59 ccgx user.err kernel: [43008.665337] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
Sep 13 14:25:59 ccgx user.info kernel: [43008.665348] vedirect-dbus   D    0  1339   1336 0x00000000
Sep 13 14:25:59 ccgx user.warn kernel: [43008.665378] [<c0607cfc>] (__schedule) from [<c060811c>] (schedule+0x40/0xa0)
Sep 13 14:25:59 ccgx user.warn kernel: [43008.665398] [<c060811c>] (schedule) from [<c06083d0>] (schedule_preempt_disabled+0xc/0x10)
Sep 13 14:25:59 ccgx user.warn kernel: [43008.665418] [<c06083d0>] (schedule_preempt_disabled) from [<c0609540>] (__mutex_lock.constprop.4+0xf8/0x324)
Sep 13 14:25:59 ccgx user.warn kernel: [43008.665438] [<c0609540>] (__mutex_lock.constprop.4) from [<c03fd58c>] (tty_unthrottle_safe+0x14/0x7c)
Sep 13 14:25:59 ccgx user.warn kernel: [43008.665462] [<c03fd58c>] (tty_unthrottle_safe) from [<c03fafc8>] (n_tty_read+0x1fc/0x880)
Sep 13 14:25:59 ccgx user.warn kernel: [43008.665484] [<c03fafc8>] (n_tty_read) from [<c03f60fc>] (tty_read+0x90/0xd4)
Sep 13 14:25:59 ccgx user.warn kernel: [43008.665511] [<c03f60fc>] (tty_read) from [<c023036c>] (__vfs_read+0x1c/0x11c)
Sep 13 14:25:59 ccgx user.warn kernel: [43008.665533] [<c023036c>] (__vfs_read) from [<c02304f8>] (vfs_read+0x8c/0x110)
Sep 13 14:25:59 ccgx user.warn kernel: [43008.665552] [<c02304f8>] (vfs_read) from [<c0230978>] (SyS_read+0x3c/0x90)
Sep 13 14:25:59 ccgx user.warn kernel: [43008.665573] [<c0230978>] (SyS_read) from [<c0107840>] (ret_fast_syscall+0x0/0x54)
Sep 13 14:28:02 ccgx user.err kernel: [43131.541907] INFO: task kworker/u2:0:5 blocked for more than 120 seconds.
Sep 13 14:28:02 ccgx user.err kernel: [43131.541945]       Not tainted 4.14.67 #1
Sep 13 14:28:02 ccgx user.err kernel: [43131.541956] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
Sep 13 14:28:02 ccgx user.info kernel: [43131.541972] kworker/u2:0    D    0     5      2 0x00000000
Sep 13 14:28:02 ccgx user.info kernel: [43131.542022] Workqueue: events_unbound flush_to_ldisc
Sep 13 14:28:02 ccgx user.warn kernel: [43131.542067] [<c0607cfc>] (__schedule) from [<c060811c>] (schedule+0x40/0xa0)
Sep 13 14:28:02 ccgx user.warn kernel: [43131.542091] [<c060811c>] (schedule) from [<c06083d0>] (schedule_preempt_disabled+0xc/0x10)
Sep 13 14:28:02 ccgx user.warn kernel: [43131.542116] [<c06083d0>] (schedule_preempt_disabled) from [<c0609540>] (__mutex_lock.constprop.4+0xf8/0x324)
Sep 13 14:28:02 ccgx user.warn kernel: [43131.542136] [<c0609540>] (__mutex_lock.constprop.4) from [<c03fd510>] (tty_throttle_safe+0x14/0x7c)
Sep 13 14:28:02 ccgx user.warn kernel: [43131.542175] [<c03fd510>] (tty_throttle_safe) from [<c03fb7c0>] (n_tty_receive_buf_common+0xd8/0x9bc)
Sep 13 14:28:02 ccgx user.warn kernel: [43131.542200] [<c03fb7c0>] (n_tty_receive_buf_common) from [<c03fc0b8>] (n_tty_receive_buf2+0x14/0x1c)
Sep 13 14:28:02 ccgx user.warn kernel: [43131.542224] [<c03fc0b8>] (n_tty_receive_buf2) from [<c03fea38>] (tty_port_default_receive_buf+0x38/0x58)
Sep 13 14:28:02 ccgx user.warn kernel: [43131.542245] [<c03fea38>] (tty_port_default_receive_buf) from [<c03fe364>] (flush_to_ldisc+0x7c/0xa8)
Sep 13 14:28:02 ccgx user.warn kernel: [43131.542275] [<c03fe364>] (flush_to_ldisc) from [<c013c2f4>] (process_one_work+0x1c4/0x45c)
Sep 13 14:28:02 ccgx user.warn kernel: [43131.542300] [<c013c2f4>] (process_one_work) from [<c013c5ec>] (worker_thread+0x60/0x590)
Sep 13 14:28:02 ccgx user.warn kernel: [43131.542322] [<c013c5ec>] (worker_thread) from [<c0141af8>] (kthread+0x148/0x150)
Sep 13 14:28:02 ccgx user.warn kernel: [43131.542353] [<c0141af8>] (kthread) from [<c0107928>] (ret_from_fork+0x14/0x2c)
Sep 13 14:28:02 ccgx user.err kernel: [43131.542608] INFO: task vedirect-dbus:1339 blocked for more than 120 seconds.
Sep 13 14:28:02 ccgx user.err kernel: [43131.542620]       Not tainted 4.14.67 #1
Sep 13 14:28:02 ccgx user.err kernel: [43131.542630] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
Sep 13 14:28:02 ccgx user.info kernel: [43131.542640] vedirect-dbus   D    0  1339   1336 0x00000000
Sep 13 14:28:02 ccgx user.warn kernel: [43131.542670] [<c0607cfc>] (__schedule) from [<c060811c>] (schedule+0x40/0xa0)
Sep 13 14:28:02 ccgx user.warn kernel: [43131.542690] [<c060811c>] (schedule) from [<c06083d0>] (schedule_preempt_disabled+0xc/0x10)
Sep 13 14:28:02 ccgx user.warn kernel: [43131.542807] [<c06083d0>] (schedule_preempt_disabled) from [<c0609540>] (__mutex_lock.constprop.4+0xf8/0x324)
Sep 13 14:28:02 ccgx user.warn kernel: [43131.542828] [<c0609540>] (__mutex_lock.constprop.4) from [<c03fd58c>] (tty_unthrottle_safe+0x14/0x7c)
Sep 13 14:28:02 ccgx user.warn kernel: [43131.542853] [<c03fd58c>] (tty_unthrottle_safe) from [<c03fafc8>] (n_tty_read+0x1fc/0x880)
Sep 13 14:28:02 ccgx user.warn kernel: [43131.542875] [<c03fafc8>] (n_tty_read) from [<c03f60fc>] (tty_read+0x90/0xd4)
Sep 13 14:28:02 ccgx user.warn kernel: [43131.542901] [<c03f60fc>] (tty_read) from [<c023036c>] (__vfs_read+0x1c/0x11c)
Sep 13 14:28:02 ccgx user.warn kernel: [43131.542924] [<c023036c>] (__vfs_read) from [<c02304f8>] (vfs_read+0x8c/0x110)
Sep 13 14:28:02 ccgx user.warn kernel: [43131.542943] [<c02304f8>] (vfs_read) from [<c0230978>] (SyS_read+0x3c/0x90)
Sep 13 14:28:02 ccgx user.warn kernel: [43131.542965] [<c0230978>] (SyS_read) from [<c0107840>] (ret_fast_syscall+0x0/0x54)
Sep 13 14:29:25 ccgx daemon.info connmand[790]: ntp: time slew +0.002858 s
Sep 13 14:30:05 ccgx user.err kernel: [43254.418929] INFO: task kworker/u2:0:5 blocked for more than 120 seconds.
Sep 13 14:30:05 ccgx user.err kernel: [43254.418967]       Not tainted 4.14.67 #1
Sep 13 14:30:05 ccgx user.err kernel: [43254.418979] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
Sep 13 14:30:05 ccgx user.info kernel: [43254.418994] kworker/u2:0    D    0     5      2 0x00000000
Sep 13 14:30:05 ccgx user.info kernel: [43254.419043] Workqueue: events_unbound flush_to_ldisc
Sep 13 14:30:05 ccgx user.warn kernel: [43254.419089] [<c0607cfc>] (__schedule) from [<c060811c>] (schedule+0x40/0xa0)
[... 66 lines cut ...]
[... cut ...]
Sep 13 14:34:10 ccgx user.warn kernel: [43500.179503] [<c03fafc8>] (n_tty_read) from [<c03f60fc>] (tty_read+0x90/0xd4)
Sep 13 14:34:10 ccgx user.warn kernel: [43500.179531] [<c03f60fc>] (tty_read) from [<c023036c>] (__vfs_read+0x1c/0x11c)
Sep 13 14:34:10 ccgx user.warn kernel: [43500.179553] [<c023036c>] (__vfs_read) from [<c02304f8>] (vfs_read+0x8c/0x110)
Sep 13 14:34:10 ccgx user.warn kernel: [43500.179572] [<c02304f8>] (vfs_read) from [<c0230978>] (SyS_read+0x3c/0x90)
Sep 13 14:34:10 ccgx user.warn kernel: [43500.179594] [<c0230978>] (SyS_read) from [<c0107840>] (ret_fast_syscall+0x0/0x54)
```

idsite 958

jhofstee, 2018-09-14:
another one:
```
[67338.344533]       Not tainted 4.14.67 #1
[67338.344544] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[67338.344560] sleep           D    0  3190    932 0x00000000
[67338.344624] [<c0607cfc>] (__schedule) from [<c060811c>] (schedule+0x40/0xa0)
[67338.344681] [<c060811c>] (schedule) from [<c060a340>] (rwsem_down_read_failed+0xb0/0x104)
[67338.344720] [<c060a340>] (rwsem_down_read_failed) from [<c0129668>] (do_exit+0x13c/0xa0c)
[67338.344746] [<c0129668>] (do_exit) from [<c0129fa4>] (do_group_exit+0x40/0xa4)
[67338.344769] [<c0129fa4>] (do_group_exit) from [<c012a018>] (__wake_up_parent+0x0/0x18)
[67461.224439] INFO: task sleep:3190 blocked for more than 120 seconds.
[67461.224476]       Not tainted 4.14.67 #1
[67461.224488] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[67461.224504] sleep           D    0  3190    932 0x00000000
[67461.224567] [<c0607cfc>] (__schedule) from [<c060811c>] (schedule+0x40/0xa0)
[67461.224597] [<c060811c>] (schedule) from [<c060a340>] (rwsem_down_read_failed+0xb0/0x104)
[67461.224635] [<c060a340>] (rwsem_down_read_failed) from [<c0129668>] (do_exit+0x13c/0xa0c)
[67461.224661] [<c0129668>] (do_exit) from [<c0129fa4>] (do_group_exit+0x40/0xa4)
[67461.224684] [<c0129fa4>] (do_group_exit) from [<c012a018>] (__wake_up_parent+0x0/0x18)
[67584.104411] INFO: task sleep:3190 blocked for more than 120 seconds.
[67584.104448]       Not tainted 4.14.67 #1
[67584.104459] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
```
idSite 13976

jhofstee, 2018-09-14:
```
[67707.512621] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[67707.512637] vedirect-dbus   D    0  1267   1257 0x00000000
[67707.512700] [<c0607cfc>] (__schedule) from [<c060811c>] (schedule+0x40/0xa0)
[67707.512724] [<c060811c>] (schedule) from [<c06083d0>] (schedule_preempt_disabled+0xc/0x10)
[67707.512750] [<c06083d0>] (schedule_preempt_disabled) from [<c0609540>] (__mutex_lock.constprop.4+0xf8/0x324)
[67707.512774] [<c0609540>] (__mutex_lock.constprop.4) from [<c03fd58c>] (tty_unthrottle_safe+0x14/0x7c)
[67707.512812] [<c03fd58c>] (tty_unthrottle_safe) from [<c03fafc8>] (n_tty_read+0x1fc/0x880)
[67707.512835] [<c03fafc8>] (n_tty_read) from [<c03f60fc>] (tty_read+0x90/0xd4)
[67707.512863] [<c03f60fc>] (tty_read) from [<c023036c>] (__vfs_read+0x1c/0x11c)
[67707.512885] [<c023036c>] (__vfs_read) from [<c02304f8>] (vfs_read+0x8c/0x110)
[67707.512904] [<c02304f8>] (vfs_read) from [<c0230978>] (SyS_read+0x3c/0x90)
[67707.512934] [<c0230978>] (SyS_read) from [<c0107840>] (ret_fast_syscall+0x0/0x54)
[67707.512990] INFO: task kworker/u2:3:32662 blocked for more than 120 seconds.
[67707.513002]       Not tainted 4.14.67 #1
[67707.513011] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[67707.513021] kworker/u2:3    D    0 32662      2 0x00000000
[67707.513050] Workqueue: events_unbound flush_to_ldisc
[67707.513074] [<c0607cfc>] (__schedule) from [<c060811c>] (schedule+0x40/0xa0)
[67707.513092] [<c060811c>] (schedule) from [<c06083d0>] (schedule_preempt_disabled+0xc/0x10)
[67707.513112] [<c06083d0>] (schedule_preempt_disabled) from [<c0609540>] (__mutex_lock.constprop.4+0xf8/0x324)
[67707.513131] [<c0609540>] (__mutex_lock.constprop.4) from [<c03fd510>] (tty_throttle_safe+0x14/0x7c)
[67707.513154] [<c03fd510>] (tty_throttle_safe) from [<c03fb7c0>] (n_tty_receive_buf_common+0xd8/0x9bc)
[67707.513177] [<c03fb7c0>] (n_tty_receive_buf_common) from [<c03fc0b8>] (n_tty_receive_buf2+0x14/0x1c)
[67707.513201] [<c03fc0b8>] (n_tty_receive_buf2) from [<c03fea38>] (tty_port_default_receive_buf+0x38/0x58)
[67707.513222] [<c03fea38>] (tty_port_default_receive_buf) from [<c03fe364>] (flush_to_ldisc+0x7c/0xa8)
[67707.513251] [<c03fe364>] (flush_to_ldisc) from [<c013c2f4>] (process_one_work+0x1c4/0x45c)
[67707.513274] [<c013c2f4>] (process_one_work) from [<c013c5ec>] (worker_thread+0x60/0x590)
[67707.513384] [<c013c5ec>] (worker_thread) from [<c0141af8>] (kthread+0x148/0x150)
[67707.513408] [<c0141af8>] (kthread) from [<c0107928>] (ret_from_fork+0x14/0x2c)
[67830.392585] INFO: task vedirect-dbus:1267 blocked for more than 120 seconds.
[67830.392625]       Not tainted 4.14.67 #1
```
idSite 18659, it is an onboard ve.direct tty

jhofstee, 2018-09-15:
another one 958
```[43008.664533] INFO: task kworker/u2:0:5 blocked for more than 120 seconds.
[43008.664573]       Not tainted 4.14.67 #1
[43008.664584] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[43008.664600] kworker/u2:0    D    0     5      2 0x00000000
[43008.664650] Workqueue: events_unbound flush_to_ldisc
[43008.664694] [<c0607cfc>] (__schedule) from [<c060811c>] (schedule+0x40/0xa0)
[43008.664801] [<c060811c>] (schedule) from [<c06083d0>] (schedule_preempt_disabled+0xc/0x10)
[43008.664827] [<c06083d0>] (schedule_preempt_disabled) from [<c0609540>] (__mutex_lock.constprop.4+0xf8/0x324)
[43008.664847] [<c0609540>] (__mutex_lock.constprop.4) from [<c03fd510>] (tty_throttle_safe+0x14/0x7c)
[43008.664885] [<c03fd510>] (tty_throttle_safe) from [<c03fb7c0>] (n_tty_receive_buf_common+0xd8/0x9bc)
[43008.664910] [<c03fb7c0>] (n_tty_receive_buf_common) from [<c03fc0b8>] (n_tty_receive_buf2+0x14/0x1c)
[43008.664933] [<c03fc0b8>] (n_tty_receive_buf2) from [<c03fea38>] (tty_port_default_receive_buf+0x38/0x58)
[43008.664953] [<c03fea38>] (tty_port_default_receive_buf) from [<c03fe364>] (flush_to_ldisc+0x7c/0xa8)
[43008.664984] [<c03fe364>] (flush_to_ldisc) from [<c013c2f4>] (process_one_work+0x1c4/0x45c)
[43008.665007] [<c013c2f4>] (process_one_work) from [<c013c5ec>] (worker_thread+0x60/0x590)
[43008.665029] [<c013c5ec>] (worker_thread) from [<c0141af8>] (kthread+0x148/0x150)
[43008.665060] [<c0141af8>] (kthread) from [<c0107928>] (ret_from_fork+0x14/0x2c)
[43008.665315] INFO: task vedirect-dbus:1339 blocked for more than 120 seconds.
[43008.665327]       Not tainted 4.14.67 #1
[43008.665337] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[43008.665348] vedirect-dbus   D    0  1339   1336 0x00000000
[43008.665378] [<c0607cfc>] (__schedule) from [<c060811c>] (schedule+0x40/0xa0)
[43008.665398] [<c060811c>] (schedule) from [<c06083d0>] (schedule_preempt_disabled+0xc/0x10)
[43008.665418] [<c06083d0>] (schedule_preempt_disabled) from [<c0609540>] (__mutex_lock.constprop.4+0xf8/0x324)
[43008.665438] [<c0609540>] (__mutex_lock.constprop.4) from [<c03fd58c>] (tty_unthrottle_safe+0x14/0x7c)
[43008.665462] [<c03fd58c>] (tty_unthrottle_safe) from [<c03fafc8>] (n_tty_read+0x1fc/0x880)
[43008.665484] [<c03fafc8>] (n_tty_read) from [<c03f60fc>] (tty_read+0x90/0xd4)
[43008.665511] [<c03f60fc>] (tty_read) from [<c023036c>] (__vfs_read+0x1c/0x11c)
[43008.665533] [<c023036c>] (__vfs_read) from [<c02304f8>] (vfs_read+0x8c/0x110)
[43008.665552] [<c02304f8>] (vfs_read) from [<c0230978>] (SyS_read+0x3c/0x90)
[43008.665573] [<c0230978>] (SyS_read) from [<c0107840>] (ret_fast_syscall+0x0/0x54)
[43131.541907] INFO: task kworker/u2:0:5 blocked for more than 120 seconds.
[43131.541945]       Not tainted 4.14.67 #1
[43131.541956] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[43131.541972] kworker/u2:0    D    0     5      2 0x00000000
[43131.542022] Workqueue: events_unbound flush_to_ldisc
[43131.542067] [<c0607cfc>] (__schedule) from [<c060811c>] (schedule+0x40/0xa0)
[43131.542091] [<c060811c>] (schedule) from [<c06083d0>] (schedule_preempt_disabled+0xc/0x10)
[43131.542116] [<c06083d0>] (schedule_preempt_disabled) from [<c0609540>] (__mutex_lock.constprop.4+0xf8/0x324)
[43131.542136] [<c0609540>] (__mutex_lock.constprop.4) from [<c03fd510>] (tty_throttle_safe+0x14/0x7c)
[43131.542175] [<c03fd510>] (tty_throttle_safe) from [<c03fb7c0>] (n_tty_receive_buf_common+0xd8/0x9bc)
[43131.542200] [<c03fb7c0>] (n_tty_receive_buf_common) from [<c03fc0b8>] (n_tty_receive_buf2+0x14/0x1c)
[43131.542224] [<c03fc0b8>] (n_tty_receive_buf2) from [<c03fea38>] (tty_port_default_receive_buf+0x38/0x58)
[43131.542245] [<c03fea38>] (tty_port_default_receive_buf) from [<c03fe364>] (flush_to_ldisc+0x7c/0xa8)
[43131.542275] [<c03fe364>] (flush_to_ldisc) from [<c013c2f4>] (process_one_work+0x1c4/0x45c)
[43131.542300] [<c013c2f4>] (process_one_work) from [<c013c5ec>] (worker_thread+0x60/0x590)
[43131.542322] [<c013c5ec>] (worker_thread) from [<c0141af8>] (kthread+0x148/0x150)
[43131.542353] [<c0141af8>] (kthread) from [<c0107928>] (ret_from_fork+0x14/0x2c)
[43131.542608] INFO: task vedirect-dbus:1339 blocked for more than 120 seconds.
[43131.542620]       Not tainted 4.14.67 #1
[43131.542630] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[43131.542640] vedirect-dbus   D    0  1339   1336 0x00000000
[43131.542670] [<c0607cfc>] (__schedule) from [<c060811c>] (schedule+0x40/0xa0)
[43131.542690] [<c060811c>] (schedule) from [<c06083d0>] (schedule_preempt_disabled+0xc/0x10)
[43131.542807] [<c06083d0>] (schedule_preempt_disabled) from [<c0609540>] (__mutex_lock.constprop.4+0xf8/0x324)
[43131.542828] [<c0609540>] (__mutex_lock.constprop.4) from [<c03fd58c>] (tty_unthrottle_safe+0x14/0x7c)
[43131.542853] [<c03fd58c>] (tty_unthrottle_safe) from [<c03fafc8>] (n_tty_read+0x1fc/0x880)
[43131.542875] [<c03fafc8>] (n_tty_read) from [<c03f60fc>] (tty_read+0x90/0xd4)
[43131.542901] [<c03f60fc>] (tty_read) from [<c023036c>] (__vfs_read+0x1c/0x11c)
[43131.542924] [<c023036c>] (__vfs_read) from [<c02304f8>] (vfs_read+0x8c/0x110)
[43131.542943] [<c02304f8>] (vfs_read) from [<c0230978>] (SyS_read+0x3c/0x90)
[43131.542965] [<c0230978>] (SyS_read) from [<c0107840>] (ret_fast_syscall+0x0/0x54)
[43254.418929] INFO: task kworker/u2:0:5 blocked for more than 120 seconds.
[43254.418967]       Not tainted 4.14.67 #1
[... cut ...]
[43500.179391] [<c03fd58c>] (tty_unthrottle_safe) from [<c03fafc8>] (n_tty_read+0x1fc/0x880)
[43500.179503] [<c03fafc8>] (n_tty_read) from [<c03f60fc>] (tty_read+0x90/0xd4)
[43500.179531] [<c03f60fc>] (tty_read) from [<c023036c>] (__vfs_read+0x1c/0x11c)
[43500.179553] [<c023036c>] (__vfs_read) from [<c02304f8>] (vfs_read+0x8c/0x110)
[43500.179572] [<c02304f8>] (vfs_read) from [<c0230978>] (SyS_read+0x3c/0x90)
[43500.179594] [<c0230978>] (SyS_read) from [<c0107840>] (ret_fast_syscall+0x0/0x54)
```

mpvader, 2019-01-07:
[This commit](https://github.com/victronenergy/meta-victronenergy/commit/419281ad97720b326c9bcd73dff6ac62cfb48c63) fixed most; but not all. In the end there were other types of locks still happening; fixed by changing kernel config back to how it was before updating the kernel.

It also happened on the 3.7 kernel if we disabled the option there.

closed 2019-01-07 by mpvader