victronenergy/venus #352: vrmlogger crashing

jhofstee, 2018-09-15:
there are several sites reporting 'tmp file already read' since vrmlogger asserts on:
```
2018-08-27 14:34:18.369569500 AssertionError: Buffer /data/db/vrmlogger-backlog.sqlite3, expected to delete ids [94466, 94467, 94468, 94469, 94470, 94471, 94472, 94473, 94474, 94475, 94476, 94477, 94478, 94479, 94480, 94481, 94482, 94483, 94484, 94485, 94486, 94487, 94488, 94489, 94490, 94491, 94492, 94493, 94494, 94495, 94496, 94497, 94498, 94499, 94500, 94501, 94502, 94503, 94504, 94505, 94506, 94507, 94508, 94509, 94510, 94511, 94512, 94513, 94514, 94515, 94516, 94517, 94518, 94519, 94520, 94521, 94522, 94523, 94524, 94525, 94526, 94527, 94528, 94529, 94530, 94531, 94532, 94533, 94534, 94535, 94536, 94537, 94538, 94539, 94540, 94541, 94542, 94543, 94544, 94545, 94546, 94547, 94548, 94549, 94550, 94551, 94552, 94553, 94554, 94555, 94556, 94557, 94558, 94559, 94560, 94561, 94562, 94563, 94564, 94565], but delete rowcount (99) was inconsistent!
```
also on v2.17
site 16989 e.g.

wiebeytec, 2018-12-05:
We had another occurrance today. The sqlite file in question is corrupt, which may be the cause of the assertion error, that it tried deleting rows but couldn't. The data was then resent over and over to the server, piling up the queue.

wiebeytec, 2019-01-29:
As discussed in real life: changing the assert into a warning.

closed 2019-08-23 by mpvader