victronenergy/venus #353: onboard usb hub not found on Linux ccgx 4.14.67

jhofstee, 2018-09-17:
While running a reflash test https://github.com/victronenergy/meta-victronenergy/blob/master/meta-venus/recipes-test/test-swupdate_1.3.bb with the swu on an USB pendrive, the test stops because the onboard root hub is not found (and hence the stick doesn't get mounted).

```
root@ccgx:~# lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

This is seen on two different ccgx's runing v2.20~66. It is unknown yet if 2.18 has the same issue.

jhofstee, 2018-09-18:
2.18 doesn't seem to have this issue, it has been installing / rebooting from usb for at least 12 hours now

jhofstee, 2018-09-20:
good boot on the same device:
```
Sep 20 12:42:58 ccgx syslog.info syslogd started: BusyBox v1.24.1
Sep 20 12:42:58 ccgx user.notice kernel: klogd started: BusyBox v1.24.1 (2018-07-25 16:03:36 CEST)
Sep 20 12:42:58 ccgx user.info kernel: [    0.000000] Booting Linux on physical CPU 0x0
Sep 20 12:42:58 ccgx user.notice kernel: [    0.000000] Linux version 4.14.67 (oe-user@oe-host) (gcc version 7.3.0 (GCC)) #2 Fri Sep 14 22:12:14 CEST 2018
Sep 20 12:42:58 ccgx user.info kernel: [    0.000000] CPU: ARMv7 Processor [411fc087] revision 7 (ARMv7), cr=10c5387d
Sep 20 12:42:58 ccgx user.info kernel: [    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
Sep 20 12:42:58 ccgx user.info kernel: [    0.000000] OF: fdt: Machine model: Color Control GX
Sep 20 12:42:58 ccgx user.info kernel: [    0.000000] Memory policy: Data cache writeback
Sep 20 12:42:58 ccgx user.info kernel: [    0.000000] cma: Reserved 16 MiB at 0x8e800000
Sep 20 12:42:58 ccgx user.debug kernel: [    0.000000] On node 0 totalpages: 64768
Sep 20 12:42:58 ccgx user.debug kernel: [    0.000000] free_area_init_node: node 0, pgdat c0a5f1bc, node_mem_map cfab1000
Sep 20 12:42:58 ccgx user.debug kernel: [    0.000000]   Normal zone: 570 pages used for memmap
Sep 20 12:42:58 ccgx user.debug kernel: [    0.000000]   Normal zone: 0 pages reserved
Sep 20 12:42:58 ccgx user.debug kernel: [    0.000000]   Normal zone: 64768 pages, LIFO batch:15
Sep 20 12:42:58 ccgx user.info kernel: [    0.000000] CPU: All CPU(s) started in SVC mode.
Sep 20 12:42:58 ccgx user.info kernel: [    0.000000] AM3517 ES1.1 (l2cache sgx neon)
Sep 20 12:42:58 ccgx user.debug kernel: [    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
Sep 20 12:42:58 ccgx user.debug kernel: [    0.000000] pcpu-alloc: [0] 0 
Sep 20 12:42:58 ccgx user.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 64198
Sep 20 12:42:58 ccgx user.notice kernel: [    0.000000] Kernel command line: mem=254M root=ubi0:rootfs1 rootfstype=ubifs ubi.mtd=12 omapdss.def_disp=lcd omapfb.vram=0:2M@0x8fe00000 fbskip fbcon=vc:64 mtdparts=omap2-nand.0:128k(spl1),128k(spl2),128k(spl3),128k(spl4),1m
Sep 20 12:42:58 ccgx user.info kernel: [    0.000000] PID hash table entries: 1024 (order: 0, 4096 bytes)
Sep 20 12:42:58 ccgx user.info kernel: [    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes)
Sep 20 12:42:58 ccgx user.info kernel: [    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes)
Sep 20 12:42:58 ccgx user.info kernel: [    0.000000] Memory: 229972K/259072K available (6144K kernel code, 399K rwdata, 1808K rodata, 1024K init, 187K bss, 12716K reserved, 16384K cma-reserved, 0K highmem)
Sep 20 12:42:58 ccgx user.notice kernel: [    0.000000] Virtual kernel memory layout:
Sep 20 12:42:58 ccgx user.notice kernel: [    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
Sep 20 12:42:58 ccgx user.notice kernel: [    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
Sep 20 12:42:58 ccgx user.notice kernel: [    0.000000]     vmalloc : 0xd0000000 - 0xff800000   ( 760 MB)
Sep 20 12:42:58 ccgx user.notice kernel: [    0.000000]     lowmem  : 0xc0000000 - 0xcfd00000   ( 253 MB)
Sep 20 12:42:58 ccgx user.notice kernel: [    0.000000]     pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
Sep 20 12:42:58 ccgx user.notice kernel: [    0.000000]     modules : 0xbf000000 - 0xbfe00000   (  14 MB)
Sep 20 12:42:58 ccgx user.notice kernel: [    0.000000]       .text : 0xc0008000 - 0xc0700000   (7136 kB)
Sep 20 12:42:58 ccgx user.notice kernel: [    0.000000]       .init : 0xc0900000 - 0xc0a00000   (1024 kB)
Sep 20 12:42:58 ccgx user.notice kernel: [    0.000000]       .data : 0xc0a00000 - 0xc0a63fa8   ( 400 kB)
Sep 20 12:42:58 ccgx user.notice kernel: [    0.000000]        .bss : 0xc0a6a658 - 0xc0a993fc   ( 188 kB)
Sep 20 12:42:58 ccgx user.info kernel: [    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
Sep 20 12:42:58 ccgx user.info kernel: [    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
Sep 20 12:42:58 ccgx user.info kernel: [    0.000000] IRQ: Found an INTC at 0xfa200000 (revision 4.0) with 96 interrupts
Sep 20 12:42:58 ccgx user.info kernel: [    0.000000] Clocking rate (Crystal/Core/MPU): 26.0/332/600 MHz
Sep 20 12:42:58 ccgx user.info kernel: [    0.000000] OMAP clockevent source: timer2 at 13000000 Hz
Sep 20 12:42:58 ccgx user.info kernel: [    0.000025] sched_clock: 32 bits at 13MHz, resolution 76ns, wraps every 165191050201ns
Sep 20 12:42:58 ccgx user.info kernel: [    0.000060] clocksource: timer1: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 147020034397 ns
Sep 20 12:42:58 ccgx user.info kernel: [    0.000075] OMAP clocksource: timer1 at 13000000 Hz
Sep 20 12:42:58 ccgx user.info kernel: [    0.000816] clocksource: 32k_counter: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 58327039986419 ns
Sep 20 12:42:58 ccgx user.info kernel: [    0.000830] OMAP clocksource: 32k_counter at 32768 Hz
Sep 20 12:42:58 ccgx user.info kernel: [    0.001464] Console: colour dummy device 80x30
Sep 20 12:42:58 ccgx user.info kernel: [    0.002040] console [tty0] enabled
Sep 20 12:42:58 ccgx user.info kernel: [    0.002085] Calibrating delay loop... 593.92 BogoMIPS (lpj=296960)
Sep 20 12:42:58 ccgx user.info kernel: [    0.010989] pid_max: default: 32768 minimum: 301
[... cut ...]
Sep 20 12:43:00 ccgx daemon.notice dbus[508]: [system] Successfully activated service 'fi.w1.wpa_supplicant1'
Sep 20 12:43:00 ccgx daemon.notice wpa_supplicant[579]: Successfully initialized wpa_supplicant
Sep 20 12:43:00 ccgx user.notice kernel: [   12.375361] UBIFS (ubi0:1): un-mount UBI device 0
Sep 20 12:43:11 ccgx daemon.info dnsmasq[515]: read /etc/hosts - 6 addresses
Sep 20 12:43:11 ccgx daemon.info dnsmasq[515]: using nameserver 192.168.4.11#53
Sep 20 12:43:16 ccgx daemon.info init: no more processes left in this runlevel
```

jhofstee, 2018-09-21:
a different pendrive caused the same issue after 5-6 hours flashing and rebooting:

```
Sep 20 18:48:22 ccgx user.info kernel: [    2.879391] usb 1-1.1: new high-speed USB device number 3 using ehci-omap
Sep 20 18:48:22 ccgx user.info kernel: [    2.979319] usb 1-1.1: New USB device found, idVendor=058f, idProduct=6387
Sep 20 18:48:22 ccgx user.info kernel: [    2.979397] usb 1-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Sep 20 18:48:22 ccgx user.info kernel: [    2.979430] usb 1-1.1: SerialNumber: 13042251045121
Sep 20 18:48:22 ccgx user.info kernel: [    6.828911] usb-storage 1-1.1:1.0: USB Mass Storage device detected
Sep 20 18:48:22 ccgx user.info kernel: [    6.835677] scsi host0: usb-storage 1-1.1:1.0
Sep 20 18:48:22 ccgx user.info kernel: [    6.836398] usbcore: registered new interface driver usb-storage
Sep 20 18:48:22 ccgx user.notice kernel: [    6.926809] UBIFS (ubi0:1): background thread "ubifs_bgt0_1" started, PID 297
Sep 20 18:48:22 ccgx user.notice kernel: [    7.843060] scsi 0:0:0:0: Direct-Access                               8.07 PQ: 0 ANSI: 4
Sep 20 18:48:22 ccgx user.notice kernel: [    7.849718] sd 0:0:0:0: [sda] 7680000 512-byte logical blocks: (3.93 GB/3.66 GiB)
Sep 20 18:48:22 ccgx user.notice kernel: [    7.851084] sd 0:0:0:0: [sda] Write Protect is off
Sep 20 18:48:22 ccgx user.debug kernel: [    7.851145] sd 0:0:0:0: [sda] Mode Sense: 23 00 00 00
Sep 20 18:48:22 ccgx user.notice kernel: [    7.853757] sd 0:0:0:0: [sda] Write cache: disabled, read cache: enabled, doesn't support DPO or FUA
Sep 20 18:48:22 ccgx user.info kernel: [    7.862911]  sda: sda1

```

jhofstee, 2018-09-21:
linux v3.10 seems to have this issue while linux v3.9 does not

jhofstee, 2018-12-30:
workaround is at https://github.com/victronenergy/meta-victronenergy/commit/ef84962dc8a5bedf75bb5fca8b79c2a6fbf1828a

mpvader, 2019-01-07:
Closing; we are not going to spend time fixing this for real.

closed 2019-01-07 by mpvader