victronenergy/venus #356: Feature request : Limit battery discharge rate when Grid is available

Paulf007, 2018-09-29:
Currently the option is a available to limit the inverter power if Grid is available but will it be possible to limit the discharge rate of the batteries as well ? This will ensure that the solar panels will still be able to perform at optimal and if the load is higher than what the panels can supply the grid will pick up the difference. I would assume that you will need a minimum discharge rate as the end user might want to attempt to set discharge to 0 w and that that would not be possible.

mpvader, 2019-11-04:
Though making the above is possible from a software poiint of view; using small batteries often creates all sorts of issues; so for now I prefer to spend our limited R&D capacitiy on other improvements. thanks.

Paulf007, 2019-11-04:
I agree that a small bank will be a problem but the goal is rather to ensure that the bank gets discharged more evenly rather than having to contend with high load spikes. This should ensure a longer battery life span.