victronenergy/venus #357: Lynx BMS: almost discharged flag

mpvader, 2018-10-01:
do we raise any alarms on that already? Is it on dbus? Looks like its not:

```
root@beaglebone:~# dbus -y com.victronenergy.battery.ttyO2 / GetText 
value = {
 'Alarms/Alarm': '',
 'Alarms/HighStarterVoltage': '0',
 'Alarms/HighTemperature': '0',
 'Alarms/HighVoltage': '0',
 'Alarms/LowSoc': '0',
 'Alarms/LowStarterVoltage': '0',
 'Alarms/LowTemperature': '0',
 'Alarms/LowVoltage': '0',
 'Alarms/MidVoltage': '0',
 'Connected': '1',
 'ConsumedAmphours': '-189.8Ah',
 'CustomName': '',
 'Dc/0/Current': '-46.9A',
 'Dc/0/MidVoltage': '',
 'Dc/0/MidVoltageDeviation': '',
 'Dc/0/Power': '-2371W',
 'Dc/0/Temperature': '',
 'Dc/0/Voltage': '50.56V',
 'Dc/1/Voltage': '',
 'DeviceInstance': '258',
 'FirmwareVersion': 'v3.08',
 'HardwareVersion': '',
 'History/AutomaticSyncs': '40',
 'History/AverageDischarge': '-367.1Ah',
 'History/ChargeCycles': '36',
 'History/ChargedEnergy': '2648.16kWh',
 'History/DeepestDischarge': '-574.0Ah',
 'History/DischargedEnergy': '2369.70kWh',
 'History/FullDischarges': '0',
 'History/HighStarterVoltageAlarms': '',
 'History/HighVoltageAlarms': '0',
 'History/LastDischarge': '-501.0Ah',
 'History/LowStarterVoltageAlarms': '',
 'History/LowVoltageAlarms': '0',
 'History/MaximumStarterVoltage': '0.00V',
 'History/MaximumVoltage': '58.32V',
 'History/MinimumStarterVoltage': '0.00V',
 'History/MinimumVoltage': '0.00V',
 'History/TimeSinceLastFullCharge': '158528s',
 'History/TotalAhDrawn': '-48103.4Ah',
 'Mgmt/Connection': 'VE.Direct',
 'Mgmt/ProcessName': 'vedirect-dbus',
 'Mgmt/ProcessVersion': '3.23',
 'ProductId': '0x204',
 'ProductName': 'BMV-702',
 'Relay/0/State': '1',
 'Serial': 'HQ1723D3PP2',
 'Settings/HasMidVoltage': '0',
 'Settings/HasStarterVoltage': '0',
 'Settings/HasTemperature': '1',
 'Soc': '79.9%',
 'TimeToGo': '56100s'}
```

mpvader, 2018-10-01:
Its a flag in the same VREG that has the AllowToCharge & Discharge flag, says Wilco

martinbosma, 2018-10-01:
Right now only `/ErrorCode` triggers an alarm for all Lynx BMSes. I can add an alarm on "almost discharged" quite easily. This requires modification of `gui` and `debus-vecan`.

I propose to add `/Alarms/AlmostDischarged` which reflects bit 3 of `VE_REG_BMS_FLAGS`,

@mpvader What is the priority of this?

mpvader, 2018-10-01:
its not urgent, but also just a small task. So pls do in the coming few weeks when you want a break of a larger project.

martinbosma, 2018-10-02:
@mpvader Should the `Alarms/AlmostDischarged` have the value `1` (warning) or `2` (alarm) on the dbus? Or is vrm-logger unable to cope with warnings and does it always expect the value `2`? Either way is fine for me.

martinbosma, 2018-10-03:
As discussed with @mpvader:
Instead of `Alarms/AlmostDischarged` we will make `Alarms/LowCellVoltage`, which can have the following values:
- `0` - No alarm
- `1` - Warning: almost discharged, based on bit 3 of `VE_REG_BMS_FLAGS`
- `2` - Alarm: fully discharged, based on bit 2 of `VE_REG_BMS_FLAGS`

This will work in case of the (new) Lynx Ion BMS, because bit 2 & 3 are determined on cell voltage.

For the old BMSes, we have a different VREG (`VE_REG_LYNX_ION_FLAGS`), which is almost the same as `VE_REG_BMS_FLAGS`. However, the definition of bit 2 (fully discharged), actually reflect the inverted "allow to discharge" signal, which is not only based on cell voltage, **but also on temperature.**

We can do 2 things:
1. Only support the warning (almost discharged) for these older BMSes. The new BMS will have both warning and alarm.
2. Only support the warning (almost discharged) for all BMSes. Perhaps even go back to the `Alarms/AlmostDischarged` proposal.

mpvader, 2018-10-03:
Only support the path in new BMS-es. We don’t sell the old one anymore.

martinbosma, 2018-10-05:
Does the path `Alarms/LowCellVoltage` also need to be added to others services? E.g. vrm-logger or modbus-tcp?

izak, 2019-01-31:
Included in Venus v2.30~28.

closed 2019-01-31 by izak