victronenergy/venus #361: Connman and link local

mpvader, 2018-10-03:
Does Connman keep retrying dhcp requests after falling back to link local?

After what timeout?

And with what interval does it keep retrying?

This is for the MFD use case; some manufacturer rely on link local: installer simply plugs multiple boxes together, and it then works. Without requiring dhcp or anything like that.

Two options (?):
- assign and keep a link local address; even when assigned a dhcp address or even a manual one.
- shorten or even eliminate the first timeout; so that a link local network works quickly after power up. And in case there is a dhcp server, it removes the link local again. Hopefully only requiring a change of timeout in Connman.

Gussoh, 2018-10-04:
For communicating with Navico MFDs I recommend assign and keep a IPv4 link local address. IPv6 is currently not supported.

mpvader, 2018-12-11:
```
root@beaglebone:~# route
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
default         192.168.178.1   0.0.0.0         UG    0      0        0 eth0
89.101.251.228  192.168.178.1   255.255.255.255 UGH   0      0        0 eth0
89.101.251.229  192.168.178.1   255.255.255.255 UGH   0      0        0 eth0
169.254.0.0     *               255.255.0.0     U     0      0        0 ll-eth0
172.24.24.0     *               255.255.255.0   U     0      0        0 ap0
192.168.178.0   *               255.255.255.0   U     0      0        0 eth0
192.168.178.1   *               255.255.255.255 UH    0      0        0 eth0
```

martinbosma, 2018-12-18:
Tested with v2.30~17 on various MFDs
- Navico: works much better; icon is displayed within ~10s
- Garmin: still works fine
- Raymarine: still works fine

Also the following:
- Remote Console on LAN: ok
- Remote Console on VRM: ok
- Remote Support: ok
- vrmlogging: ok

mpvader, 2018-12-21:
In Venus v2.30~16 or 17

closed 2018-12-21 by mpvader