victronenergy/venus #362: When accessing an address that maps to a non-existent dbus path, a zero is returned instead of an error

izak, 2018-10-04:
Confirmed by fetching register 259 from a BMV, and then changing the dbus path in `attributes.csv` to something that doesn't exist.

There is presently no known user-visible bug related to this, but it is incorrect behaviour and masks other bugs. What is more, the user is often unaware of it, because a zero result is often an acceptable result (for example for flags).

mpvader, 2018-10-04:
Back in the days 🙃; when this project started, care was taken to not create gaps in the mappings.

Do we have services with paths not the sometimes while being there are others?

How does it handle dbus-invalid?

izak, 2018-10-05:
A gap sometimes exists because something is not supported in a particular configuration. For example, on a BMV there might be a hole at register 260 (/Dc/1/Voltage) if the BMV is configured to use the auxiliary for something other than a starter battery, and similarly 262 (/Dc/0/Temperature) if configured as anything other than a temperature sensor. If you fetch the entire range from 259, as you might, then there is a difference in behaviour:

 * Presently it will return zero for the unsupported registers
 * Under the implementation suggested above, it will return an error.

Dbus-invalid is handled the same way as a non-existent path.

 * Presently it returns 0
 * Under the new implementation it will return an error.

My suggestion is to leave this one out of 2.20, as it is at least not a regression.

izak, 2019-02-26:
I'm closing this as a won't-fix, because we've had no actual complaints (I found the bug while working on victronenergy/venus#360). We're leaving the existing behaviour so that ranges can be fetched.

closed 2019-02-26 by izak