victronenergy/venus #369: CanBus errors during Venus GX hardwire reboot

wilcop85, 2018-10-19:
When the Venus GX is booting from power reset it produces CAN-bus errors and a bus heavy message.
In the attachment there are CAN-bus dumps from different booting procedures.

As you can see in the software reboot there are not problems, but in the hardwire reboot with or without any other communication devices connected there are error's.

Venus GX 
PN: BPP900400100
SN: 1721GUG8E
Firmware version: 2.20 20181016112144

[VenusGXHardwireReBootWithLynxIonBMS.txt](https://github.com/victronenergy/venus/files/2495566/VenusGXHardwireReBootWithLynxIonBMS.txt)
[VenusGXHardwireReBootWithoutDevices.txt](https://github.com/victronenergy/venus/files/2495567/VenusGXHardwireReBootWithoutDevices.txt)
[VenusGXSoftwareReBootWithLynxIonBMS.txt](https://github.com/victronenergy/venus/files/2495568/VenusGXSoftwareReBootWithLynxIonBMS.txt)

mpvader, 2018-10-19:
You also tested with a ccgx, right? What happened there?

mpvader, 2018-10-19:
And thanks for the report of course 👍👍

wilcop85, 2018-10-19:
Yes also tested with the CCGX and there are no problems.

[CCGXHardwareReBootWithoutDevices.txt](https://github.com/victronenergy/venus/files/2495676/CCGXHardwareReBootWithoutDevices.txt)

jhofstee, 2019-08-19:
closing this, since this can't be fixed software...

closed 2019-08-19 by jhofstee