victronenergy/venus #374: vebus-interface: add reading battery temperature

mpvader, 2018-11-04:
- is available in VE.Bus firmware since version 512.
- RAM ID 21.
- when no sensor is connected, 0xFF is returned in the high byte
- no difference visible between a remotely received temperature and a self-measured temperature
- temperature reading has been designed (a long time ago) for temperature compensation, which doesn't require accuracy up to the single degree celcius. Which this measurement also is not. How precise it is doesn't matter.

vejpasop, 2018-11-27:
Added to mk2-dbus v2.73

izak, 2019-01-31:
# Shared Temperature Sense

## Present implementation
* Only the Active Battery Service is considered as a source of both voltage or temperature
* If the Active Battery service is a Multi, temperature is assumed to be unavailable
* If temperature is available from the selected battery service, it is copied to the solar chargers and to the Multi

## Modified implementation
* It should be possible to use temperature and voltage from different devices
  * Example: BMV-700 is used as battery service, but temperature comes from vebus
* If the temperature comes from the Multi, don't write it back to the same Multi

mpvader, 2019-03-18:
@vejpasop you wrote: `is available in VE.Bus firmware since version 512.`.  That should be 412? Or?

And can the issue be closed now? Please discuss with Izak if there are any open last things.

vejpasop, 2019-03-19:
That should be 412 indeed.
The issue can be closed. (discussed with Izak)

mpvader, 2019-03-19:
thanks; closed. And I see now I wrote that 512; not you ;-)

closed 2019-03-19 by mpvader