victronenergy/venus #378: add node-red

mpvader, 2018-11-27:
### Todo

- [ ] Add nodejs & node-red into Venus OS
- [ ] Add switch to gui, including popup warning: 'This functionality is not supported by Victron; instead use https://community.victronenergy.com'.
- [ ] Implement Victron nodes
- [ ] Disable the node-red palette: it allows users to (npm-)install add-ons
- [ ] Create examples
- [ ] Document for the users

### What problem does this solve?

The value in this is that it enables capable installers that want to automatically control available outputs such as relays, or software set-points such as the `ESS Grid set-point`, or `Max Charge Current`.

The risk is that it creates endless possibilities, and its impossible to support by Victron: our support staff does not contain software developers; nor PLC exports, and also we don't want to start employing them for the support team. This risk is mitigated by declaring that there is no official support for this; only community support on [https://community.victronenergy.com](https://community.victronenergy.com).

### Details

- Node-red, including Victron nodes, will be installed on the rootfs
- Node-red will be off by default, enabled with a switch in the gui
- Node-red communicates to the D-Bus via the mqtt bridge, [dbus-mqtt](https://github.com/victronenergy/dbus-mqtt)
- Flows will be stored on the data partition; so they survive a Venus firmware update
- Venus OS version X will contain Node-red version Y, with Victron-node-set version Z.
- Users will not be able to install their own nodes; also not via the `Node Palette`. That `Node Palette`.

### Only runs on Venus GX and future devices

- Node-red will not run on the CCGX: too small 'disk' size.
- For Venus GX it only works when partitioned with the 2nd revision partitions sizes; introduced in production in  2018 Q3.

Future Venus-devices under development do have enough space for this. More information about available disk space is [here](https://github.com/victronenergy/venus/wiki/machines).

### Concept

[image]

mpvader, 2018-12-11:
First version of specification is available [here](https://github.com/victronenergy/node-red-contrib-victron/blob/master/documentation/specification.md).

Next step taken by @pkronstrom is to implement two simple nodes; the battery one and the relay one. Once they are OK; lets do the most complicated one: inverter/charger; to validate that the idea also works for the most complex one.

mpvader, 2018-12-13:
Summary of decisions made today:

-  when making flows; the UI needs to know what services are available and what not. So; node-red needs to enumerate them. After some back and forth; we realized that its not possible to do that with dbus-mqtt, since it doesn't support wildcards on read requests. Decided was to instead interface to D-Bus; using this code as a start: https://github.com/sbender9/signalk-venus-plugin
- there will be one relay-output-node. Which will allow user to select any of the relays available on the system. So; the victron node-red code needs to have hardcoded the various possible service  and path combinations, for example com.victronenergy.battery + /Relay, or com.victronenergy.system + /Relay/0/State); check which ones are available; and list those.
- For the html5 app; we are abstracting away the fact that a BMV has a second input measuring the voltage of a starter battery; and another example: a three output charger; of which the second and third output charge some stand-alone generator starter batteries. All of it will be under com.victronenergy.system /Batteries, in a dict. For node-red however; we don't want that abstraction; it would be confusing to, more technical, type of person that uses Node-red.

[image]

mpvader, 2019-01-06:
Some screenshots of functionality as in the branch on the repo now:

There are two nodes:
[image]

These are the settings of the battery node:
[image]

And these are the settings of the relay node:
[image]

mpvader, 2019-01-07:
### Adding nodejs & node-red to Venus - update

I've worked on this; the result is an image for the Venus GX; with nodejs v8.4.0 and node-red v0.19.15 installed under daemontools; starting up by default. Change in OE are in the [meta-victronenergy/matthijs/wip/node-red branch](https://github.com/victronenergy/meta-victronenergy/tree/matthijs/wip/node-red).

Node-red web UI is available on https://[ip-address]:1880/.

Image can be downloaded [here](https://www.dropbox.com/s/m27plfswx9bpw0y/venus-swu-beaglebone-20190107083549-v2.30~20-node-red.swu?dl=0). Install using a USB stick or sd-card; docs [here](https://www.victronenergy.com/live/ccgx:firmware_updating#microsd-card_or_usb-stick); or copy it with scp onto the device, and then use one of the commandline options of check-update.sh (or do it straight via stdin)

Note that it can only be installed on Venus GX-es that have been formatted with a large partition. How-to-do that is [here](https://github.com/SignalK/signalk-server-node/issues/517#issue-315239812) -> chapter IV, repartition the flash.

To add our node-red work, run:
```
npm install --production --no-optional -g https://github.com/victronenergy/node-red-contrib-victron/archive/feature/multiplebatteries.tar.gz
```

The --no-optional is there to make it skip installing abstract-socket; which requires compilation. I don't know yet if abstract-socket is required. Its a dependency of dbus-native; Which as far as I remember had no problem in connecting to local dbus in the work done [here](https://github.com/SignalK/signalk-server-node/issues/517); so that would imply it works without.

After installing that module, restart the node-red server by running:
```
svc -t /service/node-red
```

Looking at the node-red logs can be done by running:
```
tail -F /log/node-red/current | tai64nlocal
```

### Whats next?

The D-Bus connection doesn't seem to work yet; which is the next thing to do. It should connect through the system d-bus; just like it does in the Venus-signalk-plugin. Not via TCP. I briefly tried making some modifications; but failed and then gave up / had to do something else.

@pkronstrom please take it from here and let me know if I need to change anything.

pkronstrom, 2019-01-15:
I did a fresh install on a new Venus GX. I got the node-red working and victron nodes connecting to dbus following Matthijs' instructions from above:

1. run the installer image and expanded the filesystem (resize2fs.sh)
2. scp the node-red swu to Venus GX and Installed from the command line
`scp venus-swu-beaglebone-20190107083549-v2.30_20-node-red.swu root@192.168.2.8:~/venus.swu`
`./opt/victronenergy/swupdate-scripts/check-updates.sh -swu ~/venus.swu`
3. install the master branch npm install --production --no-optional -g https://github.com/victronenergy/node-red-contrib-victron/archive/master.tar.gz

At this point, opening the browser at the Venus GX's address `http://<venus-gx-ip>:1880 should show you the node-red ui. However, checking from the logs, it seems that the plugin crashes (on a missing environment variable -- this is irrelevant in this case, though)

I managed to get the connection working simply by replacing the tcp-connection string in [victronclient.js](https://github.com/victronenergy/node-red-contrib-victron/blob/master/src/services/victronclient.js#L60) with a null, which causes the dbus-listener.js to default to using dbus.systemBus().

**TL:DR; I need to make a small fix to victronclient.js**

pkronstrom, 2019-01-16:
> **TL:DR; I need to make a small fix to victronclient.js**
I changed the VictronClient to default to SystemBus if the environment variable NODE_RED_DBUS_ADDRESS is not set. See [commit](https://github.com/victronenergy/node-red-contrib-victron/commit/8b438e9c344cbb8a1bcf02868ddeae29c88171c6) for the changes.

mpvader, 2019-01-16:
The comments in the commit and code are a bit misleading

Where you say systembus; you mean the socket type (not tcp). And then depending on another env variable it chooses between system and session dbus.

pkronstrom, 2019-01-29:
Switch on commit [a4f440b](https://github.com/victronenergy/gui/commit/a4f440b20920301ffa3d29526365f8d31ed877ef)

Short toast for immediate feedback:
[image]

Static text shown below the switch, if the service is enabled:
[image]

mpvader, 2019-03-10:
Test images are available!

- https://www.dropbox.com/s/k6n6p2i45gdqnxw/venus-swu-raspberrypi2-20190310172138-v2.30~32-node-red.swu?dl=0
- https://www.dropbox.com/s/oqhv32ozalddkdr/venus-image-raspberrypi2-20190310172138-v2.30~32-node-red.rootfs.rpi-sdimg.zip?dl=0
- https://www.dropbox.com/s/llmeaav17ntte8f/venus-swu-beaglebone-20190310172552-v2.30~32-node-red.swu?dl=0

Node-red UI is at http://(ip address):1880

thomas-pfeifer, 2019-03-22:
Hi there. I am a new GitHub user, not knowing exactly whether this is the right way to report an issue related to mpvader's message above. However, I have experienced the following after repartitioning the flash and installing venus-swu-beaglebone-20190310172552-v2.30~32-node-red.swu:
1. the Node-Red switch does not appear in the GUI
2. the Node-Red UI can be accessed at IP:1880, but shows two "syntax-error-messages" on startup
[image]

Do I have to install venus-nodes manually? Or is there any other advice to get it running?
Thx, Thomas

mpvader, 2019-03-22:
Hi @thomas-pfeifer, thanks for reporting!

> 1. the Node-Red switch does not appear in the GUI

Thats correct; its always on for now.

> 2. the Node-Red UI can be accessed at IP:1880, but shows two "syntax-error-messages" on startup
> Do I have to install venus-nodes manually? Or is there any other advice to get it running?

No, its not necessary to install nodes manually. And actually we have a new version built recently; could you try them? See next comment.

mpvader, 2019-03-22:
New test images are available.

- https://www.dropbox.com/s/wl7m4pi8wzviefw/venus-image-raspberrypi2-20190317215210-v2.30_35-node-red.rootfs.rpi-sdimg.zip?dl=0
- https://www.dropbox.com/s/8lk1n4xrns0i5ka/venus-swu-beaglebone-20190317215624-v2.30_35-node-red.swu?dl=0
- https://www.dropbox.com/s/2u20aru076egdwp/venus-swu-raspberrypi2-20190317215210-v2.30_35-node-red.swu?dl=0

Node-red UI is at http://(ip address):1880

And its default on: there is no GUI switch.

thomas-pfeifer, 2019-03-22:
Hi, I have installed the Beaglebone-SWU 2.30-35 from the link above. Unfortunately, I get the same syntax-error messages in Node-Red UI, as before. Victron nodes are not visible in the UI. When disabling / enabling the "node-red-contrib-victron, 1.1.1" nodes under "User Settings" => "Palette", same syntax errors are thrown.
Any ideas?

jhofstee, 2019-03-22:
@thomas-pfeifer can you check with Firefox or Chrome? Perhaps it is an IE issue..

thomas-pfeifer, 2019-03-22:
Yes, the issue was caused by IE. Tried now with Chrome and Opera - both working fine.
Thanks a lot!

pkronstrom, 2019-03-27:
Indeed the frontend code used some features not supported by IE. This should be fixed in the next version (IE 10+).

SEspe, 2019-03-30:
Hi thanks. Is there a "noob - howto" to install this on 2.30~39 in an RPi ?

mpvader, 2019-03-30:
> Hi thanks. Is there a "noob - howto" to install this on 2.30~39 in an RPi ?

No. But search for venus raspberrypi image install in Google, and for questions see the Modifications space on community.victronenergy.com.

If you have inprovements for the documentation while going through the process, please do improve the docs yourself. All this Victron on rpi is a community effort.

Enjoy!

thomas-pfeifer, 2019-07-10:
Hi, I have a nice Node-Red "Energy Manager" running on a Venus GX, using the test image provided by @mpvader on March 22, based on Venus OS v2.30_35. However, I recently experienced a [Remote Console via VRM connection issue](https://community.victronenergy.com/questions/15183/remote-console-via-vrm-not-working.html), which hopefully could be solved by updating the Venus Firmware, as indicated in the [description of v2.31 bug fixes](https://www.victronenergy.com/blog/2019/06/06/venus-os-v2-31-various-fixes/).
Do you have a Beaglebone-SWU with current Venus OS and Node-Red available?
Otherwise, is there a simple walkthrough to install Node-Red on Venus GX after Firmware-Update?
Will there be a solution (somwhen in future) to have automatic FW updates on the Venus, with Node-Red and programmed flows on the device remaining unchanged?
Thx a lot in advance,
Thomas

mpvader, 2019-07-10:
Hi @thomas-pfeifer; yes, actually: this IP address verification issue was solved in v2.30.

I’ll see if I can make a new build for you (v2.32-nodered) tonight.

Wrt the updatability: once we make node-red support official, then you can do updates without (always) losing flows. If it will be 100% watertight remains to be seen.

And, how we’ll manage that for the Venus GX remains to be seen as well; due to the fact that we’ve produced many of them with the small version partition layout. There are solutions; but they’ll take (more) time. And thats whats lacking at the moment: time.

I’ll post here when I have updated images.

thomas-pfeifer, 2019-07-11:
Hi @mpvader, thank you so much for quick support. It worked great! Remote Console via VRM is now working again, and Node-Red seems to be functioning as before. Surprisingly (for me), my deployed Flows and additional Palette entries completely "survived" the FW update (via SD card), while Victron nodes were updated to v1.2.0. Of course, I have too less knowledge to understand all this stuff - just wanted to let you know that it worked perfect.

mpvader, 2019-07-11:
Hi @thomas-pfeifer: thats great to hear. Thanks for the testing report; useful - even though you might not understand everything; its the info I was looking for :-).

mpvader, 2020-01-11:
An update; I've just now built, and (very briefly) tested v2.42-node-red-1. Meaning that I took the v2.42 version; and added my node-red changes on top.

Details:
* node-red version v0.19.5; which is unchanged. There are newer version (1.0.something iirc); which to be looked into later.
* node-red-contrib-victron v1.2.0; also unchanged, and latest version.

Branches:
* https://github.com/victronenergy/openembedded-core/commits/matthijs/node-red
* https://github.com/victronenergy/meta-victronenergy/tree/matthijs/node-red

Images:
* Beaglebone (Venus GX & Octo GX): 
will follow later today
* Raspberrypi: will follow later today
* Cerbo GX: v2.42 is not available for Cerbo

See above threads for how to install on a Venus GX; depending on its serial number you might need to repartition it; which takes some extra effort. Not so for Octo GX and Raspberrypi.

mpvader, 2020-01-11:
images are available for download now, see previous comment. if you install & use it; please leave a note here with your experience (all went well is also an experience :) ), whishes or comments. Thanks!

osaether, 2020-01-11:
I tried it right now on my Venus GX and it works! I have no devices connected at the moment so there is not much to do.

mpvader, 2020-03-07:
An update, I've just now built v2.51-node-red-1. The change is that this adds an image for the new Cerbo GX (which is dual core & a lot faster, so better suited than the Venux GX).

No changes were made to node-red itself; its all the same as previous images, but then built on top of v2.51 rather than v2.42.

Images:

* Available for: Venus GX & Octo GX (filename beaglebone), RaspberryPis (filename raspberrypi), and Maxi GX, EasySolar-II GX & MultiPlus-II GX (filename nanopi), and Cerbo GX (filename Einstein).
* https://www.dropbox.com/sh/ox8zo65w94hekwe/AAAhQzF0RB53mHyCkypvlk3Ca?dl=0
* Not available for CCGX & CANvu500 (not enough diskspace)

Once installed, navigate to http://[ipaddres]:/1880/ to see the files.

Used branches are meta-victronenergy/matthijs/node-red2 & openembedded-code/matthijs/node-red2, and for the rest all the b2.50 branches (ie. start with venus/b2.50 and reconf to redo this on your own).

warwickchapman, 2020-04-02:
Trivial but I cannot edit it myself and some might find it confusing: in the issue text, there is a bullet with suggested deleting indicated:

- Users will not be able to install their own nodes; also not via the Node Palette. ~That Node Palette.~

sparkmark, 2020-04-07:
Hey guys ... Ran into an issue. I resized my FS using /opt/victronenergy/swupdate-scripts/resize2fs.sh which was fine.

I copied across the latest install for node-red and installed that using /opt/victronenergy/swupdate-scripts/check-updates.sh -swu /home/root/venus-swu-beaglebone-20200307064055-v2.51-node-red-2.swu

All worked fine, and I was able to access node-red.

However after a few minutes, my venus rebooted itself and reverted back to its previous version (being 2.53). Disk space also reverted back to standard.

mpvader, 2020-04-07:
Hey @sparkmark: make sure to disable automatic updates. You can do that already before you install the node-red install.

sparkmark, 2020-04-07:
Of course ! ... thanks @mpvader

mpvader, 2020-04-07:
No problem; welcome!

@warwickchapman : I’ve seen your msg - I’ll sort it. Thanks!

MARKUS-001, 2020-04-25:
Hi Matthijs, I am using node red quite a bit now since you introduced it to me on the 14th March. It’s actually quite amazing what can be achieved with node red. I have installed quite a few new nodes and created customized dash boards. Grid set point, max inverter power variations, weather forecasts, relay 2 switches on algorithms after checking loads, SOC, combined MPPT energy with no developer skills at all. At first I really struggled to get familiar with the function node, most others are fairly simple. So I googled node red examples then cut and pasted until something started working. With the lockdown in place and many hours later i’m able to produce some meaning full results with the function node and some formulas. 
I did come a cross one problem though, relay 1, I switched it to manual as per the popup message in node red when trying to use it, but it seems it’s dedicated to to the Venus OS and can’t be used in node red. Is this temporary until further development is done on the node red side of things or will relay 1 always be unavailable for use in node red. Do you think the aux relays on the Multi’s and MPPT’s will eventually be programable in node red or will those remain dedicated to those units through assistants. Is there a relay board available that can connect to the ve.bus or ve.can
Thanks Markus.

mpvader, 2020-04-25:
Hi @MARKUS-001 thats great to hear, thanks for the update.

mpvader, 2020-04-25:
That first venus gx relay should simply work. Something for @sbender9 to look into.

Other relays (multis, mppts) no not anytime soon.

Relay boxes, we’re not working on anything. My best bet would be to buy something online, ethernet or usb connected; & control that from within node red.

RobDuthie, 2020-04-25:
Hi
Her is open source PCB design for a output unit for the raspi units, it could be adapted to suit.
The PCB design just needs a couple more ground plains and is done ready for production.
Rob D
NZ
[Victron Raspi Hat.pdf](https://github.com/victronenergy/venus/files/4534125/Victron.Raspi.Hat.pdf)

mpvader, 2020-04-29:
Hello!

An update; I've just now built, and (very briefly) tested v2.60~16-large-image1. Which includes both signalk and node-red. And, those two features together in one image, is also what is now the plan: there will be the normal images as we've been building for years now. With all normal & supported functionality.

All credits for the progress go to @sbender9 by the way - he is doing the actual work here; I've merely packed it together & kicked off the build.

And for optional install there will be the "large images". Which will have signalk & node-red. Switching between them will be easy: a simple switch in the gui. Updating will be easy to, the online updates will work, just like they do now for the normal images.

### Details of the new available v2.60~16 work

This node-red+signalk build is based on v2.60~16, a beta version. To learn about whats in there, [read here](https://community.victronenergy.com/questions/47872/venus-os-v26016-available-for-testing.html).

Details:
- node-red was updated from v0.19.5 to v1.0.5.  [Change log here](https://github.com/node-red/node-red/blob/master/CHANGELOG.md#105-maintenance-release).
- node-red-contrib-victron was updated to the latest version (v1.3.1). Significant changes:
  - a node for reading meteo data (irradiance sensor) was added
  - issue wrt MppOperationMode input and output not available was fixed
- signalk-server updated from v1.18 to v1.28. [Change log here](https://github.com/SignalK/signalk-server-node/releases)
- switches have been added in the gui, to be able to, individually, enable & disable signalk & node-red. Note that they are disabled by default
- also available for Cerbo GX now

Branches:

- https://github.com/victronenergy/openembedded-core/commits/matthijs/large-image
- https://github.com/victronenergy/meta-victronenergy/tree/matthijs/large-image
- https://github.com/victronenergy/meta-victronenergy-private/tree/matthijs/large-image
- rocko_venus as per today for all other branches - which mind you (accidentally) includes an update to Qt4. Most likely harmless.

### Downloads

- For Venus GX, Cerbo GX as well as MultiPlus-II GX & EasySolar-II GX available [here](https://www.dropbox.com/sh/z3zq6zwce4ukjai/AABgLSOL6N11Gc3DF7vb-B2ka?dl=0)

### Pipeline:

https://github.com/victronenergy/meta-victronenergy/tree/matthijs/large-image

### Screenshots:

The new settings:

[image]

The new, and for now useless, switch in the Online firmware update menu that makes it check for the "large" image:

[image]

MARKUS-001, 2020-05-02:
Evening All,
I can’t seem install v2.60_16-large-image1.swu
Check for firmware on SD/USB : no firmware found. I reformatted usb Drive as well as SD card to FAT, FAT32 exFAT still no firmware found. I copied v2.60_11 and v2.51-node-red-2.swu back onto the same USB drive, Venus detects the firmware on the USB as soon as it’s inserted.

Markus

mpvader, 2020-05-03:
Hello Markus and others,

Sorry, there were two issues with the previous files. Now both fixed:

Venus 2.60~16-large-image2:

- files are renamed so that they are recognised properly by the offline install mechanism
- fix bug that stopped the services from starting.

Downloads:

https://www.dropbox.com/sh/8fhjmcpjhbr1zhx/AADXQCJjD0L0We82zMk9ylVFa?dl=0

Rafl53, 2020-05-04:
Hello, 
I have successfully installed the 2.60-16 on my BBB. Node red seems to work perfect. The only thing I am missing is: I don’t habe solar power in W in my Smart Solar MPPT chargers. I can chose all other values but there in not the power in Watts. 

Is it a mistake or is it misting on purpose. 

Many thanks for your help in advance. 

Regards

Rainer

mpvader, 2020-05-04:
Hi @Rafl53 , thanks. I'm not sure. Its available on Venus OS internally, on the path `/Yield/Power`.  @sbender9 can you have a look?

mpvader, 2020-05-04:
ps: note that above is solar power. So thats power going to the battery plus solar power going to the load output.

Rafl53, 2020-05-05:
[image]
[image]
Can’t find it.

houser42, 2020-05-11:
Hello,
Wanted to test this out. Used the online updater and it seemed to install fine. 
I assume  we then need to manually install npm and node.js also? 
Could you pass on the correct links for the relevant versions please? Many thanks.

sbender9, 2020-05-11:
@houser42 you need to update using the .swu linked above. This is not in the release candidate yet.

houser42, 2020-05-11:
I see. Version numbers were the same so missed that. 
Up and running. Interesting!
This installed npm -v 6.4.1 and node -v v8.15.0
Ok to update to most recent: Node.js Version v12.16.3 and Npm Version v6.14.5 ?
Ok to additionally try and install a lightweight NodeJS server via npm and try out?

sbender9, 2020-05-11:
I’m not sure if updating nodejs will even work, but you can give it try. 

You should be fine installing other nodejs packages. Though you will lose them when updating the firmware.

houser42, 2020-05-11:
Ok, thanks. Will try.
If I had the old partition system on my Venus GX, has that been updated when I ran the installer? I seem to be out of space before I started ;)
I would regardless probably need to allow a little more room for this to work.

root@beaglebone:~# df
Filesystem           1K-blocks      Used Available Use% Mounted on
/dev/root               645300    631184         0 100% /
devtmpfs                247188         4    247184   0% /dev
tmpfs                   255892       216    255676   0% /run
tmpfs                   255892       416    255476   0% /var/volatile
/dev/mmcblk1p5          499656     28188    434772   6% /data
/dev/sda1            120146240    133408 120012832   0% /run/media/sda1
root@beaglebone:~#

houser42, 2020-05-11:
So not too much luck here for now. Did a factory reset via the pushbutton.
System seems to work ok, but can not access  the remote console. Also no ssh access and it seems no ports are open on the Venus GX. Any other way of doing a reset when you can not see the remote console or terminal? Many thanks.

sbender9, 2020-05-11:
I think you probably need more space on the rootfs, you need to run resize2fs.sh.

Explained here: https://www.victronenergy.com/live/ccgx:root_access

sbender9, 2020-05-11:
You'll need to load another image of the os first to do that.

houser42, 2020-05-11:
Thanks Scott,

I can not get into the terminal or remote console now though.
Any hint there on how to get back online? Thanks..

mpvader, 2020-05-11:
Oops, that little adventure halted quickly :o). I can think of two ways to get out:

1. Connect with a serial console cable, as explained here: https://github.com/victronenergy/venus/wiki/bbb-user-instructions#connecting-to-the-console-developer-feature. You do need to have such cable in order to do that. And then I'm expecting it will work well enough for you to simply be presented a prompt. Use that to make space on the rootfs, either by running resize2fs.sh or by deleting what you've installed. Also make sure that there is space on the data partition.

2. By running the installer as documented here: https://github.com/SignalK/signalk-server-node/issues/517, chapter IV. And afterwards you'll then have to recreate the first three of the there listed files yourself. Simply put in a password in the wpa-psk one, put in the serial number in the serial number one, etc.

houser42, 2020-05-11:
Well the adventure is still happening.. Just resting a bit ;)

I am in the wilderness with no IT gear but a decent laptop and an SSD that ethers says
is too big.
But I have internet and it seems the remote IP for the Venus answers on SSH.
Is that a password that Victron has and can share by any chance?
The system appears to be working fine, so no biggie right now regardless.
But would be good to try and fix this typo before I leave..
Many thanks for your help!

mpvader, 2020-05-11:
Hi, no I cant share that. There is no password; its public/private keys.

Is the system on Internet& vrm? And is remote support enabled? Maybe that works. I’d then need the vrm url.

Otherwise you could perhaps make a port forward on your router and expose the venus gx ssh port on a public ip: I can then login (most likely still only if you’ve enabled remote support).

You didnt set a root password or installed a public key of yourself?

houser42, 2020-05-11:
Yes, the system is on internet and vrm. can I send something in pm or email?
Remote support was enabled, yes. Port forward is ready on router.
I think i may have made a typo in the terminal. or something went wrong with the node-red npm
But again, it seems to answer on the remote IP in the vrm device list.
Let me know if we can do something? I can put my email in here for a while and then delete?

mpvader, 2020-05-11:
You can email me at mvader@victronenergy.com; I’ll delete this commment later. Note that its nighttime here (NL), I wont have time until somewhere tomorrow.

houser42, 2020-05-11:
Apologies, I thought I was talking to the US...
I'll send you an email. Hope to be in tough tomorrow. Many thanks!

mpvader, 2020-05-11:
Just to make sure: no email arrived yet.

houser42, 2020-05-11:
I did not want to disturb you more in this hour so planned on sending email in the morning.
Sorry if that was unclear. Get some sleep now ;) I will try too...

Rafl53, 2020-05-12:
Sorry to ask again. I didn’t find jet a solution to get the power in W of the solar charger into node red.
Maybe somebody has a solution and can help me.

[271 more replies]

closed 2022-06-25 by mpvader