victronenergy/venus #379: Add support for Carlo Gavazzi Ethernet energy meter

izak, 2018-12-04:
Probably add modbus-tcp support to dbus-cgwacs. To be decided.

RobDuthie, 2018-12-04:
Is this for the ET112 and ET340 range which Victron sells? or other models, as you know that you can run 
more than one, on mod-bus in parallel and just change the mod-bus address for each one, so saving the use of multiple USB cables to feed each one to the CCGX etc. which is limited to only two USB ports.
Just by using Carlo Gavazzi configuration software to configure each one, I have fully tested this feature and all works OK with multiple ET112 energy meters on one USB port.
Just a thought. As they say in the manual you have to use one USB port for each energy meter.

RobDuthie, 2018-12-04:
I see that that Victron Web site has changed with new diagrams etc.
As the Mulitplus 2 has issues with the energy wiring is not correct, reads wrong etc. as i have being testing the different configurations. Would you include this Ethernet for the Raspi?

mpvader, 2018-12-04:
> As the Mulitplus 2 has issues with the energy wiring is not correct, reads wrong etc. as i have being testing the different configurations

Yes indeed, visualisation of MultiPlus-II power flow on the Venus OS is not correct. Its on our list to sort out & fix whats possible and/or document. Note that its only the visualisation. And, as also discussed elsewhere, please dont use that optional MultiPlus-II specific Current Sensor. Without it, all is already complicated enough.

> Would you include this Ethernet for the Raspi?

Yes, it will work for all platforms; as all features.

As the previous one; I'll be hiding/deleting these comments again. Not at all to annoy anyone; but for R&D I need to keep this issue clean. thanks!

mpvader, 2019-04-26:
- there might more than one meter (see the role)
- code needs to cope with ip address changes (use a serial number or mac address or something)
- hopefully the meters support some discovery mechanism

mpvader, 2019-06-04:
It seems that there is no discovery protocol in the meter. We asked CG to confirm if thats the case, and they replied this:

——
Please find below how this model works : 

1) By default the DHCP is not active ... 
2) Power - ON the instrument 
3) Enable the DHCP function  (this means that as soon as the meter is connected in the LAN, it will be detected by the server and the server  will assign an available IP address to the EM24) 
4) Exit from the programming mode 
5) By the INFO pages the user will be able to see which is the assigned IP address 
6) By the UCS software ( http://www.productselection.net/Download/UK/ucs.zip )  it is possible to configure all the parameters in the instrument ... the configuration can be done also manually via the front joystick 
7) Now in the setting the DHCP can be disabled and in the instrument you can confirm the network parameters (assigned by the network administrator) 
8) Via the new IP set in the meter now it is possible to manage the standard communication 

Note that the DHCP function has to be used for the first step in the commissioning only ... after that a FIX IP address has to be managed

mpvader, 2019-06-04:
Hey @mansr, instructing installers to put it on a fixed IP is not a good idea. That will work well only in very controlled and managed environments that might work. But not for the typical house hold.

So, I wasn’t too happy with above (half expected) answer. So I think the best option is along what we discussed earlier::

In the gui, the option to do a scan (but no auto configuring). Found devices available on some dbus path. And when a unit is configured (= added to localsettings) by the installer; then store it by serial number & ip. Whenever it can’t be found on that IP; scan & search for it.

As well as make option to manually enter the ip to add a meter.

And option to delete it.

And then most customers will leave/set the meter itself it to DHCP; and the people that want to can use a fixed IP.

For the scan, you’ll need to make sure to not list other devices (PV Inverters for example). @izak also has some experience in distinguishing CG meters from other devices.

Any thoughts? Also @izak ?

mpvader, 2019-07-02:
[EM24_DS_ENG.pdf](https://github.com/victronenergy/venus/files/3349501/EM24_DS_ENG.pdf)
[EM24 Ethernet Protocol Rev 1.2.pdf](https://github.com/victronenergy/venus/files/3349502/EM24.Ethernet.Protocol.Rev.1.2.pdf)
[EM24_E1_IM_USE.pdf](https://github.com/victronenergy/venus/files/3349503/EM24_E1_IM_USE.pdf)

closed 2020-01-24 by jhofstee

donnib, 2021-02-12:
@mpvader I write to ask how you interpret the usage of this new meter, in my household i want to use static ip addresses making sure nothing changes during time so instead of fiddling with the meter itself i let it use DHCP and then configure the dhcp server to always lease the same address based on it's mac address. That should work fine and it had for some time but then the meter started dropping out, i can however reconnect the Ethernet plug then it's alive again. Do you think CG made some funky stuff (i haven't seen that on other equipment in many years of supporting networks) that forces you to set the meter **itself** for static address ?

mpvader, 2021-02-12:
Hi @donnib, you are also on Community, right? This thread: https://community.victronenergy.com/questions/66501/em24-ethernet-support-nightmare.html

right? Lets keep it there - thanks!

donnib, 2021-02-12:
> Hi @donnib, you are also on Community, right? This thread: https://community.victronenergy.com/questions/66501/em24-ethernet-support-nightmare.html
> 
> 
> 
> right? Lets keep it there - thanks!

True. Yes of course. Thx