victronenergy/venus #383: Restore the option to put a Multi in pass-through / inverter mode

mpvader, 2018-12-08:
Using VE.Bus firmware 414 or earlier, in combination with a Venus device, it was possible to set the AC input current limit to 0A. A trick, to which the Multi responds as follows:

- PowerAssist enabled -> Multi switches to "inverter only"
- PowerAssist disabled -> Multi switches to "pass thru"

When using the old Digital Multi Control panel, exactly the same happens.

This feature was removed as part of the 415 improvements in on/off & input-current-limit settings. More about that [here](https://github.com/victronenergy/venus-private/issues/56#issuecomment-393889179)

There are various options to restore it; to be discussed.

mpvader, 2019-11-04:
@vejpasop can you do this? Or does this require changes in multi firmware? Or should we do something else?

Or asked differently: how do you suggest we solve the whish from customers to force a unit into inverter mode?

mpvader, 2019-11-04:
Tip from JEroen: one solution is dropping the "readonly" attribute on the Inverter only option in the pulldown.

vejpasop, 2019-11-04:
Exactly that would solve it. Writing value = 2 to the ```Mode``` path forces the Multi in inverter mode

closed 2019-12-02 by jhofstee