victronenergy/venus #386: Extend product_id mapping for the Skylla-IP44 chargers

rvanzundert, 2018-12-10:
Please extend the product id mapping table so the firmware update will also support the Skylla-IP44 chargers.

```
        "product_id_mapping": {
            "A391": "A390",
            "A392": "A390",
            "A393": "A390",
            "A394": "A390",
            "A001": "A000",
            "A002": "A000",
            "A003": "A000",
            "A004": "A000",
            "A005": "A000",
            "A006": "A000",
            "A007": "A000",
            "A008": "A000",
        }
```

Currently these chargers show up as "Product not in library".
[image]

mauritsvdvijgh, 2018-12-10:
Done.

closed 2018-12-10 by mauritsvdvijgh

mpvader, 2018-12-10:
Testes by Roy and works