victronenergy/venus #388: systemcalc: add /Batteries path

mpvader, 2018-12-13:
- To be used by html5 app; that needs to show all batteries to the user
- Will contain all battery data; for example coming from:
  - BMV -both main measurement; and starter battery voltage measurement)
  - Skylla-i - up to three single voltages (no current)

If there is a Multi and a BMV: use the BMV data; if there is only a Multi; use that; and-so-forth. Hopefully that logic can be re-used from systemcalc.

Contents will be a dict.

to be continued.

mpvader, 2018-12-13:
more info also here: https://github.com/victronenergy/venus-html5-app/issues/55

mpvader, 2018-12-21:
For the navetta; it will be a list off:
- main battery (comes from the bmv-700) voltage, soc, everything
- stern thruster battery (Skylla-i channel 1) voltage and charge current only
- bow thruster battery (Skylla-i channel 2) voltage and charge current only

And skylla-i channel 3 is to be ignored.

izak, 2018-12-21:
# Implementation details.

- DBus is watched for battery, vebus, and charger services
  - If a battery service shows up, it is also checked for a second channel, typically a starter battery, and this is added as a separate battery
  - If a charger shows up, up to three channels are turned into battery objects
  - If there is a vebus object and it is selected as the active battery service on the CCGX, it also shows up as a battery. Otherwise it is hidden.
  - The active battery service, as selected in Settings -> System Setup, will have `active_battery_service = true`.

# Published Dictionary

Presently this is published as a json strong, but ideally it should be a dbus dictionary

```
[
    {
        "active_battery_service": false,
        "current": 1.29999995232,
        "id": "com.victronenergy.charger.socketcan_can0_di10_uc15639:0",
        "name": "Skylla-i 24/100 (3) #0",
        "voltage": 26.4200000763
    },
    {
        "active_battery_service": false,
        "current": 0.0,
        "id": "com.victronenergy.charger.socketcan_can0_di10_uc15639:1",
        "name": "Skylla-i 24/100 (3) #1",
        "voltage": 26.4300003052
    },
    {
        "active_battery_service": false,
        "current": 1.10000002384,
        "id": "com.victronenergy.charger.socketcan_can0_di10_uc15639:2",
        "name": "Skylla-i 24/100 (3) #2",
        "voltage": 26.4300003052
    },
    {
        "active_battery_service": true,
        "current": 22,
        "id": "com.victronenergy.battery.ttyO1",
        "instance": 0,
        "name": "BMV-700",
        "power": 23,
        "soc": 20,
        "state": 2,
        "timetogo": 20,
        "voltage": 21
    },
    {
        "active_battery_service": false,
        "id": "com.victronenergy.battery.ttyO1:1",
        "name": "BMV-700 #1",
        "voltage": 23
    },
    {
        "active_battery_service": true,
        "current": 12,
        "id": "com.victronenergy.vebus.ttyO1",
        "instance": 0,
        "name": "Multi 12/3000",
        "soc": 20,
        "voltage": 23
    }
]
```

mpvader, 2019-02-06:
https://github.com/victronenergy/dbus-systemcalc-py/blob/master/delegates/batterydata.py

izak, 2019-02-19:
Screenshots from GUI

[image]

[image]

closed 2019-03-15 by jhofstee