victronenergy/venus #389: mqtt-rpc: add support for xupd

WrdVictron, 2018-12-13:
xupd is the same as dup: a commandline tool to update firmware; but then for devices, connected on a VE.Direct port, that have a xup bootloader.

- [ ] Add xupd commandline tool to Venus - already listed on the todo
- [ ] Add xupd to [mqtt-rpc](https://github.com/victronenergy/mqtt-rpc)
  - not for scanning; only for updating. Selecting between xupd and dup commandline tools is to be done on filename extension (`xup` vs `dup`).
- [ ] Add support for .xup files to the firmware library (Ytec with input from Wiebren)
- [ ] Add error codes to [internal mqtt-rpc documentation](https://github.com/victronenergy/venus-private/wiki/mqtt-rpc-and-vrm) (Wiebren)
- [ ] Check & amend [customer documentation](https://www.victronenergy.com/live/vrm_portal:remote-firmware-update) where necessary (Wiebren)
- [ ] VRM: modify popup text: change `.dup` to `.dup or .xup`; see image below (El Nino)

[image]

mpvader, 2019-01-09:
@mauritsvdvijgh, the firmware library part is done; and four files are in the library. I want to make sure that the version formatter on your end works well: how would it format the version thats now in the library for productid 41905?

thlassche, 2019-01-10:
@mpvader This issue states @izak do the `mqtt-rpc` part, but in the Slack thread you said: `Or Izak; but if you can do it; then please do. Lets do that indeed: you do it; and @izak reviews afterwards.`

Should we start with it? (it's a benefit for us if the `mqtt-rpc` is done before we start with the `vrm-mqtt-rpc` part)

mpvader, 2019-01-10:
Ah I forgot; sorry. Yes please you do it. If you update a venus device to v2.30~21; which is in the testing feed, you’ll find xupd on it.

mauritsvdvijgh, 2019-01-18:
Updated a CCGX to v2.30~23 and it does not seem to contain xupd, was it removed or am I looking in the wrong place? It is not on the PATH and not in /usr/bin. 

Btw is there a changelog for the candidate releases for Venus OS?

mpvader, 2019-01-19:
once you're done with mqtt-rpc @mauritsvdvijgh and @thlassche, ask @izak for review; I also added that to the todo list above.

mauritsvdvijgh, 2019-01-19:
> @mauritsvdvijgh, the firmware library part is done; and four files are in the library. I want to make sure that the version formatter on your end works well: how would it format the version thats now in the library for productid 41905?
> 
> Update: how to format it is explained here
> https://github.com/victronenergy/velib/blob/master/data/protocols/vregs.xml#L2986

without any changes:
```
>>> version_parser(131583)
'2.01'
```

Seems correct to me.

mpvader, 2019-01-19:
> 2.01

yes thats the right version. So the library works. Now we'll just wait for you for the (vrm-)mqtt-rpc work 😉 .

mauritsvdvijgh, 2019-01-19:
Is there a way to test xupd with some device we have or with a virtual/demo type setup?

mpvader, 2019-01-19:
no I don't think so. @martinbosma pls arrange this for maurits (venus-device etc). thanks.

mauritsvdvijgh, 2019-01-20:
mqtt-rpc changes: https://github.com/victronenergy/mqtt-rpc/pull/5
vrm-mqtt-rpc changes: https://github.com/victronenergy/vrm-mqtt-rpc/pull/3

All backwards compatible and everything but the xup functionality is tested and working.

Awaiting review by @izak for mqtt-rpc and a test setup.

vrm-mqtt-rpc and the vrm frontend need some changes to make the functionality available in vrm still (frontend needs to know if a device is xup or dup and accept only files with the right file extension).

izak, 2019-01-21:
@mauritsvdvijgh Reviewed, not rocket science and pretty much exactly what I was going to do. Once tested we can merge it without much ado.

mauritsvdvijgh, 2019-01-22:
Great, definitely not rocket science.

vrm frontend changes are ready to be merged: https://gitlab.trimm.nl/fronteers/victron-gonzales-frontend/merge_requests/29

Currently VRM  does not differentiate between xup and dup files, both are allowed for duplist/VE.Direct devices. The only way readily available to me to inform the frontend of the update type (dup or xup) is by having vrm-mqtt-rpc download the firmware for every (VE.Direct) device while scanning and inspecting the file. This seemed excessive which is why I have gone for the simpler option of allowing both filetypes for VE.Direct.

mauritsvdvijgh, 2019-01-23:
Got a test installation.

Initial testing pointed out that:
- xupd needs to be called with a slightly different command than dup (no -i parameter). 
- xupd xml is different from the xml that vup and dup produce so vrm-mqtt-rpc will need to be adapted for this.

Btw I can't access:
- https://git.victronenergy.com/victron/vedirect_lib/blob/master/inc/exit_codes.h (gitlab saying "The page could not be found or you don't have permission to view it." after login)
- https://git.victronenergy.com/ve-apps/xup/blob/master/xupd/README.md#exit-codes (same as above)
- https://wiki.victronenergy.com/wiki/rend/xup/start (Permission Denied Sorry, you don't have enough rights to continue. after login)

mauritsvdvijgh, 2019-01-24:
Got access to the resources, thanks.

Turns out the output format was not different but the output was not flushed and did not arrive in vrm-mqtt-rpc (was not even sent by mqtt-rpc). Tested this with a build of xupd with line buffering for the xml output made by @martinbosma.

Made changes to mqtt-rpc after testing it:
- removed a flag that is not supported in xupd
- added --force to support downgrade and "upgrade" to the same version
- added logic to take care of dup updates for which the firmware file is already on the venus device (put there by the current version of vrm-mqtt-rpc).

https://github.com/victronenergy/mqtt-rpc/pull/5/commits/d1bbb7b70491ca43315e77f47cfe84ec9aaf119b

mpvader, 2019-01-24:
> added logic to take care of dup updates for which the firmware file is already on the venus device (put there by the current version of vrm-mqtt-rpc).

Can you make that comment in the code a bit clearer so that we also understand it a year from now? It looks like you made a change in the vrm side of things: adding extension?

And perhaps its better/cleaner/safer to just reject (& delete) files in the cache that don’t have an extension in their name; so they are redownloaded. It all feels a bit sensitive to errors now; especially since I know that we have recently already changed the  way the naming of files in the cache works.

Or othwise just introduce xupupdate as a command.

mpvader, 2019-01-24:
Ps: purpuse of cache is if someone wants to update 5 of the same devices in a row. No need at all to make cached files survive a venus firmware update.

mpvader, 2019-01-25:
No its not wiped; what I mean is that if deleting is simpler; do that. The simpler the code and the logic is; the better.

mpvader, 2019-01-25:
And if xupupdate is even more simple; then do that.

mauritsvdvijgh, 2019-01-25:
Overhauled the approach to adding xup support:

It is implemented now using a seperate command (xupd-update), which is called by vrm-mqtt-rpc when it detects that a VE.Direct firmware is of the xup format.

mqtt-rpc: https://github.com/victronenergy/mqtt-rpc/pull/5
vrm-mqtt-rpc: https://github.com/victronenergy/vrm-mqtt-rpc/pull/3

tested and working.

mauritsvdvijgh, 2019-03-07:
xup support is live for vrm-mqtt-rpc and vrm itself (frontend).

closed 2019-08-27 by WrdVictron