victronenergy/venus #396: Add end-point to store html5-app errors

heeboA, 2019-01-08:
Create an endpoint for ui errors and maybe some basic usage data for Venus GX. Write in the device log file.

mpvader, 2019-01-10:
Jeroen made a solution; it will be in Venus soon. Here is some example html/js code that uses it:

```
<!DOCTYPE html>
<html lang="en">
<style>
pre {
    background: #EEE;
    border: 1px solid #CCC;
    padding: 10px;
}
#page-wrapper {
    border-top: 5px solid #69c773;
    margin: 1em auto;
    width: 950px;
}
</style>
<script>
window.onload = function() {
    function $(selector) {
        return document.querySelector(selector);
    }
    var socket = new WebSocket('ws://canvu500:7890');
    socket.onopen = function(event) {
        $('#messages').innerHTML = 'Connected<br>';
    };
    socket.onmessage = function(event) {
        $('#messages').innerHTML += 'Received:<br>' + event.data + '<br>';
    };
    socket.onclose = function(event) {
        $('#messages').innerHTML = 'Disconnected ' + event.reason;
    };
    $('#submit').onclick = function(e) {
        socket.send($('input').value + "\n");
        $('#messages').innerHTML += 'Sent:<br>' + $('input').value + '<br>';
        $('input').value = '';
    };
};
</script>

<div id="page-wrapper">
    <pre id="messages">Connecting...</pre>
    <input id="message" required>
    <button id="submit">Send Message</button>
</div>
```

mpvader, 2019-01-15:
https://github.com/victronenergy/meta-victronenergy/commit/de387dff10d2d65f4b0df5fd32345c170e23b383

jhofstee, 2019-02-28:
included in v2.30~23

closed 2019-02-28 by jhofstee