victronenergy/venus #401: vedirect-interface: add exceptions VREGs that need them

mpvader, 2019-01-10:
In the low level vedirect driver in VictronConnect, there are a few exceptions implemented. Now that we're going to change settings via MQTT and Venus; those same exceptions need to be taken care of in the vedirect-interface driver on Venus.

- [ ] (@jepefe) Make a list of exceptions
- [ ] (@ThiemoVanEngelen) Implement them

ThiemoVanEngelen, 2019-01-15:
The exceptions in the C++ stack are:
- There is some knowledge on VE_REG_APP_VER and VE_REG_UDF_VER. This is not requested via vregs, but deduced from the ping respons and/or text protocol
- VE_REG_PRODUCT_ID returns 2 bytes for the MPPT. This data is requested during the detection phase using a :4 command.
- VE_REG_SERIAL_NUMBER - BMV - The returned value contains 2 serial numbers, separated by a comma. The second serial number is of the assembled product, which is returned by the stack
- VE_REG_[ALARM/RELAY]_[LOW/HIGH]_VOLTAGE_[2_][SET/CLEAR] - BMV - BMV uses 0.1V and the VREG spec is 0.01V so this is converted in set and get
- VE_REG_[ALARM/RELAY]_MID_VOLTAGE_[SET/CLEAR] - BMV, <= v3.07 - Bmv returns only lower byte of uint16_t so the result is unusable. This vreg is discarded by the stack
- VE_REG_DC_CHANNEL1_CURRENT - MPPT - This vreg is unknown by the MPPT. It is converted to VE_REG_CHR_CURRENT and the received result is transformed back. Also VE_REG_DC_OUTPUT_CURRENT (= load current) is substracted.
- VE_REG_DC_CHANNEL1_VOLTAGE - MPPT - This vreg is unknown by the MPPT. It is converted to VE_REG_CHR_VOLTAGE and the received result is transformed back.

ThiemoVanEngelen, 2019-02-01:
Fix is included in vedirect-interface 3.28, which is on the todo to be included in Venus v2.30~28.

closed 2019-02-04 by jhofstee