victronenergy/venus #402: Make vrmlogger deal with full disk

wiebeytec, 2019-01-15:
I ran into an installation that kept sending the same bulk of data, filling up the server queue.

The installation logged disk full errors:

```
2019-01-15 08:48:49.189790500 MainThread-http_endpoint: Exception occured when writing to buffer (database or disk is full). Switching to internal storage (if not already) and trying again.
2019-01-15 08:48:52.360048500 Traceback (most recent call last):
2019-01-15 08:48:52.365459500   File "/opt/victronenergy/vrmlogger/vrmlogger.py", line 591, in <module>
2019-01-15 08:48:52.371564500     main()
2019-01-15 08:48:52.371574500   File "/opt/victronenergy/vrmlogger/vrmlogger.py", line 574, in main
2019-01-15 08:48:52.371578500     start_logging()
2019-01-15 08:48:52.371580500   File "/opt/victronenergy/vrmlogger/vrmlogger.py", line 360, in start_logging
2019-01-15 08:48:52.371584500     endpoint.send(VrmCommandType.ANNOUNCE, get_announce())
2019-01-15 08:48:52.371587500   File "/opt/victronenergy/vrmlogger/http_endpoint.py", line 476, in send
2019-01-15 08:48:52.371590500     self._put_in_buffer_with_fallback_on_internal(data, int(time.time()))
2019-01-15 08:48:52.377136500   File "/opt/victronenergy/vrmlogger/http_endpoint.py", line 442, in _put_in_buffer_with_fallback_on_internal
2019-01-15 08:48:52.377147500     self._active_buffer.put(data, timestamp)
2019-01-15 08:48:52.377150500   File "/opt/victronenergy/vrmlogger/nonvolatilequeue.py", line 112, in put
2019-01-15 08:48:52.377154500     cursor.execute("DELETE FROM queue WHERE timestamp < (strftime('%s', 'now') - ?)", (self._max_age,))
2019-01-15 08:48:52.380802500 sqlite3.OperationalError: database or disk is full
```

Vrmlogger stops storing to sqlite when the disk space reaches 1 MB. That may need to be increased, because it leaves too small a margin for other files, and a 100% full scenario is too easily reached.

However, it also needs a more graceful handling of disk full errors.

mpvader, 2019-01-15:
Pls close and add the info to #399 instead; since thats same / similar; right?

wiebeytec, 2019-01-15:
Alright, moved the info to the other ticket.

closed 2019-01-15 by wiebeytec