victronenergy/venus #409: dbus-canbusbms: add product id for Freedom Lite

mpvader, 2019-02-06:
[image]

mpvader, 2019-02-06:
This fixes VRM portal telling the user that its a LG Chem resu. And prepares for showing a Freedom Lite icon/logo on the overview.

jhofstee, 2019-03-15:
on the site I checked there is no manufacturer info available on the CAN-bus

mpvader, 2019-03-16:
I’ve asked Freedomwon for information.

mpvader, 2019-03-30:
@jhofstee, There is a site available now. They’ve done an update in the bms to send out the string.

https://vrm.victronenergy.com/installation/33717/overview

[image]

closed 2019-05-29 by jhofstee