victronenergy/venus #410: gui: show Freedom Won logo on battery icon

mpvader, 2019-02-06:
Depends on #409.

mpvader, 2019-12-11:
[Logo FreedomWon (White).pdf](https://github.com/victronenergy/venus/files/3952474/Logo.FreedomWon.White.pdf)

@jepefe can you make a logo out of that for on the gui overview? If its not the right format, let me know.

jepefe, 2019-12-11:
> [Logo FreedomWon (White).pdf](https://github.com/victronenergy/venus/files/3952474/Logo.FreedomWon.White.pdf)
> 
> @jepefe can you make a logo out of that for on the gui overview? If its not the right format, let me know.

Yes, the logo in the pdf is a vector image so it is valid.

mpvader, 2020-06-28:
@jepefe pls arrange this (though only after all urgent Vc things for current release in progress are done). Thank you

jepefe, 2020-07-06:
@mpvader Do you only need the logo or also want me to place it in the overview?

jepefe, 2020-07-07:
@mpvader The logo is too long to place it on the battery. I see they use a different logo on their Linked in profile:

mpvader, 2020-07-14:
yes

jepefe, 2020-07-21:
Done, added to the v2.60 todo.

The short icon version is used,  the larger one doesn't fit in a way that looks good and is readable on a CCGX screen.

[image]

closed 2020-07-21 by jhofstee