victronenergy/venus #411: node-RED: admin UI authentication

tkurki, 2019-02-06:
node-RED allows one to add code to be run server side. There are reports about node-RED installations that are open to the Internet (intentionally or by mistake) attract malware.

The current plan is that node-RED will be disabled by default and the user needs to enable it in the GUI. 

Furthermore the admin UI and API should be password protected by default.

For headless setups Remote Console needs to work out of the box, so we probably can not link  node-RED authentication to RC authentication being on. From user's perspective having one username and password would make sense and I doubt there would be need for fine grained access control (=separate accounts for RC and node-RED).

node-RED provides out of the box HTTP Basic Auth support.

https://nodered.org/docs/configuration
- adminAuth
https://nodered.org/docs/security

Another option would be to bind node-RED to localhost and proxy all traffic through the main http server, delegating all security handling there. This way the auth handling would be in one place, not scattered around the system.

I am not familiar enough with how the remote access to http services to work, but having everything available under the default port might enable also that?

See also #378

tkurki, 2019-02-28:
- the existing  Remote Console password configuration is at Settings > (4th item) Remote Console 
- Node-RED activation is planned to be at Settings > (bottom most item) Services

As the two settings are pretty far apart in the menu connecting the two authentications (Remote Console VNC, Node-RED admin) feels cumbersome. I fear end users will not be able to see them as one. If we change the default authentication setup for Remote Console there is going to be confusion and turning on VNC authentication risks locking people out of the their GX's. VNC authentication is just a password, for node-RED admin it is going to be username + password.

I think the most straightforward course of action would be to generate a "nice" (letters & numbers, maybe characters) separate password when you enable node-RED and show username `node-red and the password` in the GUI. 

Once generated the password is stored in hashed format, so it is not recoverable. The easiest solution to password reset would be to show a prompt to disable & re-enable node-RED, or we can add a `Reset Password` option.

As node-RED is experimental we should take the easy, but safe way out here: set a password and reset by cycling enabled status.

mpvader, 2019-03-07:
Decided: while experimental, we'll have no password on the node-red service. Just an on/off switch; and people need to take care what they are doing (which they need to anyway; an energy system is no toy).

Once for real; we promote the VNC access password to a more generic password; and use it for both features. And at the same time; we make that password to be blank by default on the CCGX; just like it is on all the other devices. One less exception; hurray.

mpvader, 2022-08-17:
Hey @dirkjanfaber , can you read all above and let me know your take on this?

mpvader, 2024-05-21:
Hi, somewhere since above discussion we solved this. Here is the entry from the Venus OS Large user manual:

> Q3: How can I password protect Node-RED
> Password protection of Node-Red is linked to the remote console. If a password is set for the remote console, this password is also used for securing Node-Red. The username is admin, the password is the remote console password.

so closing this issue.