victronenergy/venus #412: vrmlogger: report last error

mpvader, 2019-02-06:
To help customers self-diagnosing their vrm connectivity issues; it would be nice to see the last error in the GUI.

Some examples of things I've seen going wrong are:
- dhcp server didn't set a gateway: No route to host exception in the _session.post() call.
- simcard not properly configured or activated, yet GX GSM menu says `Online`. I don't know what error VRM logger actually had
- some place holder page served, so text != `vrm: OK`).

So, what I'd like to do is have a list of error codes:

- 50: Unexpected HTTP contents   (answer.text != `vrm: OK`)
- 51: HTTP response code not 200
- 52: Unknown error: [contents of ex here].

I'd put it all in a string, no need to have the code as an integer; and show it in the gui in a multiline block. And then [in the manual](https://www.victronenergy.com/live/vrm_portal:troubleshooting_ccgx_vrm_connectivity) we'll make a list of common causes and how to troubleshoot.

Here is the code, copied from the [source](https://git.victronenergy.com/ccgx/dbus_vrm/blob/master/vrmhttp.py)
```
try:
	answer = self._session.post(self._logurl, data=data, timeout=5, verify=CA_BUNDLE, headers=headers)
except Exception, ex:
	logger.debug(ex)
	self._session = requests.Session()
	pass
else:
	result = (answer.status_code == 200 and answer.text == 'vrm: OK')
	if result:
		self._succesful_network_connection_seen = True
	else:
		logger.warning("http response rejected. Status_code: %s, Text == 'vrm: OK': %s" %
			(answer.status_code, answer.text == 'vrm: OK'))
```

izak, 2019-03-04:
Implementation for review: https://git.victronenergy.com/ccgx/dbus_vrm/commits/show_last_network_error

If all well, I can merge and delete this comment.

wiebeytec, 2019-03-25:
My feedback (without running it, because I can't test broken connections very well remotely):

`http_endpoint.py->_send_jobs_to_vrm()` calls `_update_last_network_result_f()` (note '**result**'), but `vrmlogger.py` assigns this value to 'last **error**'. This results in the last error being overwritten with a next successful post, effectively losing the 'last error'.

I would only set 'last error' on an actual error, and include a time stamp, otherwise you can't see if it was yesterday or last year.

And as long as we're changing this, this line has always bugged me:

    logger.warning("Even though the HTTP response was %s, the response is rejected. Text response: %s" % (answer.status_code, answer.text))

Because an example result of this is: '_Even though the HTTP response was 403, the response is rejected_', which makes no sense. A better line would be "_The HTTP code and/or body was invalid, so the response is rejected. Response code was '{0}' with body: {1}._".

I'll review by testing later, but I wanted to let you know this feedback already because doing so, to avoid having to do it twice.

mpvader, 2019-03-25:
Thanks for the feedback @wiebeytec. Wrt to

> This results in the last error being overwritten with a next successful post, effectively losing the 'last error

I don’t want to show any error if all is working well. Ie if the last result was success; then don’t show a last error to the user. A timestamp: could be nice; so you can see how many seconds ago the last attempt was; in case of errors.

wiebeytec, 2019-03-26:
Perhaps I misunderstood the change, but what is the difference between the normal 'state' and 'last error' then?

If it's a matter of not showing 'last error' in the GUI when the current state is OK, it may still be necessary to not update the dbus paths `/Vrm/LastNetworkError` and `/Vrm/LastNetworkErrorMessage` on OK, otherwise the lastLogData entries will only ever contain OK, because you can't log when there is no network.

izak, 2019-03-27:
As I understand it we're not going to log these to VRM. We're only going to show it in the GUI so the user has some idea what is wrong.

mpvader, 2019-03-27:
Indeed @izak

wiebeytec, 2019-03-27:
I don't know if my confusion is representative of the user base, but I would call the dbus path `/Vrm/NetworkResult` (with the GUI elements appropriately similar), because:

- 'Last' reads like 'previous', which threw me off apparently.
- It really does store the _actual_ result. 

As for the following from the issue:

> I'd put it all in a string, no need to have the code as an integer; and show it in the gui in a multiline block. And then in the manual we'll make a list of common causes and how to troubleshoot.

Currently, there _is_ a separate dbus path for the code. A code would be good for translating, but I assume/suggest not translating the errors (try programming .NET on a Dutch PC...). Just mention the English errors in the manual.

However, this is the error when you have no DNS now:

> "HTTPConnectionPool(host='ccgxlogging.victronenergy.com', port=80): Max retries exceeded with url: /log/logbulk.php?crc32=2101602381&id=9059zf6z4zz1 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0xb56b7ed0>: Failed to establish a new connection: [Errno -2] Name or service not known',))"

That is hard to fit on the screen. 

There are no consistent exceptions to check (this one was `NewConnectionError` ), but perhaps a fuzzy translate function can be used. Search for some strings in the error and return something more useful, and only return the full string when there is no fuzzy match.

To conclude, I would change `UnexpectedContents` and `UnexpectedResponse` to `UnexpectedHttpContents` and `UnexpectedHttpResponseCode` respectively. It's not clear enough that one talks about the code or not, and that it's HTTP.

That's all my feedback for now.

izak, 2019-03-27:
Hi @wiebeytec ,

I made a release with this earlier today. I changed the names of the constants as suggested already. The idea was that /LastNetworkError gives you a general idea of what is wrong (so we put that in the manual with possible reasons), while the /LastNetworkErrorMessage gives you the literal exception text and would not be of any use to anyone not at least a little bit IT-capable.

I envisioned that /LastNetworkErrorMessage will only be used for unknown errors (159, those not corresponding to a speciffic exception) or for debugging, using a kind of [Guru meditation][1] message in the GUI. Obviously not exactly like that, just in principle.

For the known exceptions (150 thru 153), we can have a translated block in the GUI explaining the problem. There are only 4 of them.

What do you think?

[1]: https://en.wikipedia.org/wiki/Guru_Meditation

wiebeytec, 2019-03-27:
(when I did the second part of my review, I first checked if the original branch was still there, and it was, unmodified. I missed the changes to master, sorry).

First about the dbus path names: to me 'last' really reads to me like '_it works now, but an hour ago xyz was the problem_'. Hence my suggested change. But, perhaps it's just a rose by any other name.

Then, when I read the original ticket, I goal seems to document in the manual what certain technical exceptions mean? But they probably won't fit on the screen now, you can see how long they are in my previous comment. 

Most errors now will be `ConnectionError - 153`, with some `UnknownError - 159`. These can include 'no route to host', 'socket read timeout', 'no DNS' and SSL errors. This seems too limited to me.

If we are going to rely on making the actual error visible so users can look it up in the manual, they are best shortened. Looking at the urllib3 source, it doesn't set any special properties with the OS error, just that long `ex.message`. So, the fuzzy translator is the best I could come up with.

mpvader, 2019-03-29:
@izak, pls try to get text on the gui. Perhaps by removing the first part of it, everything up to and included `caused by`; some a bit more intelligent.

mpvader, 2019-03-29:
ps: I don't want to write a manual that becomes a generic 'how to solve networking errors' manual. For that, people that don't recognize the word route or DNS, will have to find someone else who does.

My intention is to give people that do know this, the actual error code; so they known where to start looking.

izak, 2019-05-24:
## Concept text

- [ ] Needs to be added [here](https://www.victronenergy.com/live/vrm_portal:troubleshooting_ccgx_vrm_connectivity#step_3_-_check_connectivity_between_ccgx_and_vrm_portal).

 Navigate to Settings → VRM online portal.

[image]

Check the 'Connection error' value. If a Connection error is shown, the CCGX is not able to contact the VRM database.

The connection error will show an error value that indicates the nature of the connectivity problem. In addition an Error message is included that will aid a suitably qualified network administrator.

In cases of connectivity problems it is advised to test your connection from another device, such as a laptop computer, by attempting to access https://vrm.victronenergy.com/.

Possible problems include:

### DNS failure

The CCGX could not resolve the host name of the VRM server. Make sure that a valid DNS server is configured or that your DHCP server assigns one.

[image]

### Connection timeout

Making a connection to VRM timed out. This could indicate a poor quality internet connection or a restrictive firewall.

[image]

### Connection error
A connection to VRM could not be established. This could indicate a routing problem.

[image]

### Unknown error
This is a catch-all error for errors that cannot be directly categorised. In such cases the Error message will provide information about the problem.

[image]

## Poor and intermittent connections

'Buffered items' will indicate how many records are not yet logged to VRM. 'Last contact' will indicate how long it as been since contact was established with VRM. All data is sent using the First in First out principle: the VRM Portal will only show the most up to date information after all old data has been sent.

## If all else fails

In case all above checks have been done and the internet connection was tested from another device, with no success, try rolling the firmware back to a previous version. It is also possible to log into the CCGX using ssh and to debug the problem directly on the CCGX, but that requires in-depth knowledge of Linux and working on a text console.

mpvader, 2019-05-26:
I've added it: https://www.victronenergy.com/live/ccgx:start#trouble_shooting_data_logging

mpvader, 2019-05-26:
@izak, some errors were missing from above text. Please complete the manual.

Code:
https://git.victronenergy.com/ccgx/dbus_vrm/blob/master/vrmhttp.py#L34

Manual:
https://www.victronenergy.com/live/ccgx:start#trouble_shooting_data_logging

madsci1016, 2020-11-24:
Might be helpful is 503 Service Temporarily Unavailable is added to the error feedback to bubble up to the end user. Right now it just shows as "Unexpected HTTP Response", which I figured might have been my problem. But if I had seen "503 Service Temporarily Unavailable" I would have figured that VRM was down and just waited. 

Also VRM is down. Probably not the right place to report that though...

[image]

mpvader, 2020-11-25:
Good suggestion, thanks!

closed 2021-03-29 by jhofstee