victronenergy/venus #417: ccgx v2.23~3: doesn't boot after the update

vejpasop, 2019-02-11:
- Was running v2.30~28
- Installed v2.23~3
- there was a SD card inserted with (only) a v2.12 swu file on it
- Stuck while booting v2.23~3; showing the logo. Not `Starting`. Waited long (minutes, in the mean time I updated another color control) and nothing happened
- Also unable to login over ssh: no route to host
- Resetted several times by pressing the top left and right buttons; same result
- Connected a VE.Direct -> USB cable to see the console 
- Resetted a few more times; still the same
- Removed the SD card, resetted again, and then it booted
- Now it also boots with SD card

Questions:
- Is it normal that there is no kernel boot log?
- serial console used to not require a password; but now it does? What if there is no gui or network; how can you login when not being able to set a root password?

Console log:
```
U-Boot SPL 2013.01.01 (Jan 10 2017 - 23:02:47)

U-Boot 2013.01.01 (Jan 10 2017 - 23:02:47)

AM35XX-GP ES2.0, CPU-OPP2, L3-165MHz, Max CPU Clock 600 Mhz
BPP3 + LPDDR/NAND
I2C:   ready
DRAM:  256 MiB
NAND:  512 MiB
MMC:   OMAP SD/MMC: 0
In:    serial
Out:   serial
Err:   serial
Die ID #240e000100000000015d82741001201d
Net:   DaVinci-EMAC

NAND read: device 0 offset 0x340000, size 0x7f800
 522240 bytes read: OK
gpio: pin 26 (gpio 26) value is 1

gpio: pin 29 (gpio 29) value is 1

NAND read: device 0 offset 0xa00000, size 0x600000
 6291456 bytes read: OK
## Booting kernel from Legacy Image at 80300000 ...
   Image Name:   Victron Energy/3.7.1-ccgx26/ccgx
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    2823408 Bytes = 2.7 MiB
   Load Address: 80008000
   Entry Point:  80008000
   Verifying Checksum ... OK
   Loading Kernel Image ... OK
OK

Starting kernel ...

U-Boot SPL 2013.01.01 (Jan 10 2017 - 23:02:47)

U-Boot 2013.01.01 (Jan 10 2017 - 23:02:47)

AM35XX-GP ES2.0, CPU-OPP2, L3-165MHz, Max CPU Clock 600 Mhz
BPP3 + LPDDR/NAND
I2C:   ready
DRAM:  256 MiB
NAND:  512 MiB
MMC:   OMAP SD/MMC: 0
In:    serial
Out:   serial
Err:   serial
Die ID #240e000100000000015d82741001201d
Net:   DaVinci-EMAC

NAND read: device 0 offset 0x340000, size 0x7f800
 522240 bytes read: OK
gpio: pin 26 (gpio 26) value is 1

gpio: pin 29 (gpio 29) value is 1

NAND read: device 0 offset 0xa00000, size 0x600000
 6291456 bytes read: OK
## Booting kernel from Legacy Image at 80300000 ...
   Image Name:   Victron Energy/3.7.1-ccgx26/ccgx
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    2823408 Bytes = 2.7 MiB
   Load Address: 80008000
   Entry Point:  80008000
   Verifying Checksum ... OK
   Loading Kernel Image ... OK
OK

Starting kernel ...

U-Boot SPL 2013.01.01 (Jan 10 2017 - 23:02:47)

[and so forth]
```

swupdate log
```
2019-02-11 07:07:55.490551500 installed: 20190204221311 v2.30~28
2019-02-11 07:07:55.491241500 available: 20190211001405 v2.23~3
2019-02-11 07:07:55.576796500 Starting swupdate to install version 20190211001405 v2.23~3 ...
2019-02-11 07:07:55.774015500 software set: stable mode: copy2
...
2019-02-11 07:09:13.266598500 Main loop Daemon
2019-02-11 07:09:13.283372500 do_swupdate completed OK. Rebooting
2019-02-11 07:31:30.266784500 *** CCGX booted (1) ***
2019-02-11 07:35:05.310760500 *** CCGX booted (1) ***
```

Nothing written to messages.
```
Feb 11 07:09:17 ccgx syslog.info syslogd exiting
Feb 11 07:31:27 ccgx syslog.info syslogd started: BusyBox v1.24.1
Feb 11 07:31:27 ccgx user.notice kernel: klogd started: BusyBox v1.24.1 (2019-02-11 01:16:06 CET)
Feb 11 07:31:27 ccgx user.info kernel: [    0.000000] Booting Linux on physical CPU 0
Feb 11 07:31:27 ccgx user.notice kernel: [    0.000000] Linux version 3.7.1 (oe-user@oe-host) (gcc version 7.3.0 (GCC) ) #1 Mon Feb 11 01:16:57 CET 2019
```

Demo mode setting was added on first successful boot:
```
2019-02-11 07:09:14.762188500 2019-02-11 07:09:14,750 WARNING handlerSignals received: 15
2019-02-11 07:09:14.797884500 Traceback (most recent call last):
2019-02-11 07:09:14.798254500   File "/opt/victronenergy/localsettings/localsettings.py", line 909, in <module>
2019-02-11 07:09:14.872552500     main(sys.argv[1:])
2019-02-11 07:09:14.873235500   File "/opt/victronenergy/localsettings/localsettings.py", line 907, in main
2019-02-11 07:09:14.957264500     run()
2019-02-11 07:09:14.957988500   File "/opt/victronenergy/localsettings/localsettings.py", line 854, in run
2019-02-11 07:09:14.959773500     MainLoop().run()
2019-02-11 07:09:14.992731500 KeyboardInterrupt
2019-02-11 07:31:30.045135500 *** CCGX booted (1) ***
2019-02-11 07:31:43.147644500 localsettings v1.0b starting up 
2019-02-11 07:31:43.748077500 2019-02-11 07:31:43,746 INFO Localsettings version is: 0x010b
2019-02-11 07:31:43.755065500 2019-02-11 07:31:43,753 INFO Localsettings PID is: 1536
2019-02-11 07:31:44.061065500 2019-02-11 07:31:44,059 INFO Settings file /data/conf/settings.xml validated
2019-02-11 07:31:46.981658500 2019-02-11 07:31:46,947 INFO Added new setting /Settings/Gui/DemoMode. default:0, type:i, min:0, max: 1, silent: False
2019-02-11 07:31:46.989685500 2019-02-11 07:31:46,988 INFO GetDefault /Settings/Gui/DemoMode
2019-02-11 07:32:02.535547500 2019-02-11 07:32:02,531 INFO Setting /Settings/System/RemoteSupportPort changed. Old: 37230, New: 42335
2019-02-11 07:35:05.085388500 *** CCGX booted (1) ***
2019-02-11 07:35:23.003082500 localsettings v1.0b starting up 
2019-02-11 07:35:23.564544500 2019-02-11 07:35:23,550 INFO Localsettings version is: 0x010b
2019-02-11 07:35:23.585021500 2019-02-11 07:35:23,583 INFO Localsettings PID is: 1287
2019-02-11 07:35:23.906646500 2019-02-11 07:35:23,896 INFO Settings file /data/conf/settings.xml validated
2019-02-11 07:35:31.705688500 2019-02-11 07:35:31,704 INFO Setting /Settings/System/RemoteSupportPort changed. Old: 42335, New: 0
2019-02-11 07:35:42.328531500 2019-02-11 07:35:42,326 INFO Setting /Settings/System/RemoteSupportPort changed. Old: 0, New: 56479
2019-02-11 07:36:41.061523500 *** CCGX booted (1) ***
2019-02-11 07:36:58.184783500 localsettings v1.0b starting up 
2019-02-11 07:36:58.636108500 2019-02-11 07:36:58,634 INFO Localsettings version is: 0x010b
2019-02-11 07:36:58.643066500 2019-02-11 07:36:58,641 INFO Localsettings PID is: 1341
2019-02-11 07:36:58.779541500 2019-02-11 07:36:58,765 INFO Settings file /data/conf/settings.xml validated
2019-02-11 07:37:06.622344500 2019-02-11 07:37:06,620 INFO Setting /Settings/System/RemoteSupportPort changed. Old: 56479, New: 0
2019-02-11 07:37:17.518955500 2019-02-11 07:37:17,516 INFO Setting /Settings/System/RemoteSupportPort changed. Old: 0, New: 41574
2019-02-11 07:41:35.069091500 *** CCGX booted (1) ***
2019-02-11 07:41:52.291870500 localsettings v1.0b starting up 
2019-02-11 07:41:52.734344500 2019-02-11 07:41:52,732 INFO Localsettings version is: 0x010b
2019-02-11 07:41:52.779907500 2019-02-11 07:41:52,778 INFO Localsettings PID is: 1337
2019-02-11 07:41:52.927185500 2019-02-11 07:41:52,925 INFO Settings file /data/conf/settings.xml validated
2019-02-11 07:42:00.183990500 2019-02-11 07:42:00,182 INFO Setting /Settings/System/RemoteSupportPort changed. Old: 41574, New: 0
2019-02-11 07:42:10.293922500 2019-02-11 07:42:10,291 INFO Setting /Settings/System/RemoteSupportPort changed. Old: 0, New: 36029
root@ccgx:~# cat /opt/
```

mpvader, 2019-02-11:
Answers:
- Is it normal that there is no kernel boot log? ANSWER: yes, its an exception specific for the CCGX: since the port doubles as a VE.Direct port; we don't want to write all that texts into a BMV, MPPT or other VE.Direct device.

- serial console used to not require a password; but now it does? What if there is no gui or network; how can you login when not being able to set a root password? ANSWER: indeed; there is a fix for this in v2.30; to drop the password requirement.

- what about the early watchdog (started in u-boot); it doesn't work; or something did start to reset the watchdog. ANSWER: apparently yes, something started resetting the watchdog.

mpvader, 2019-02-11:
Closing this issue; it has proven unreproducible; and has been only one incident; all other systems worked fine.

And; the incident could be related to first running new kernel (4.19); and then booting into 3.7. Wouldn't be the first time that that causes an issue I think; and such step won't happen on the units in the field on official release.

closed 2019-02-11 by mpvader