victronenergy/venus #418: watchdog: verify new kernel watchdog option

mpvader, 2019-02-11:
Newer kernels have a new watchdog feature, telling the kernel what to do when the watchdog is already running when the kernel loads.

The default of that setting is not good for us: it keeps the watchdog happy until the userland watchdog daemon takes over. Which leaves the system with a vulnerable time in case something goes wrong in userland.

Docs: https://github.com/victronenergy/venus/wiki/watchdog

mansr, 2019-03-05:
canvu500 fixed in v4.14.71-canvu500-5, pending release.

mpvader, 2019-04-26:
For the beaglebone, kernel needs to be upgraded for this issue. Which will be done once there is a good moment (not now; v2.30 needs releasing)

mansr, 2021-05-19:
All machines run a new kernel with correct configuration since v2.70~8.

closed 2021-05-19 by mansr