victronenergy/venus #420: dbus-systemcalc-py: Make multiple BMSes in the same system work well

izak, 2019-02-13:
DVCC will use the first available battery service with BMS paths to control the solar chargers and the Multi. In most systems this is fine, as there is only one BMS.

In general it would be better to always use the active battery monitor for this, but we also have customers who have multiple battery monitors (for example, some use a BMV with a BYD battery), so one cannot necessarily rely on the BMS in the same place as the selected battery monitor.

Suggested solution: If the active battery monitor has BMS paths, use that one. Otherwise use the first one you find.

mpvader, 2019-02-14:
Hi Izak,

this is technically nice; but I don't see a reason to work on this from software. My point of view:

- Using a BMV with a BYD: if that is an issue; we officially state that its not supported and issue closed.
- Supporting multiple BMS-es: I don't want systemcalc to have to add up the allowed charge currents of different BMSes; and-so-forth.

So closing; welcome to discuss or re-open if I overlooked something.

mpvader, 2019-02-15:
Hi Izak, you wrote that DVCC will now use the first available battery service with BMS paths.

What sorting is used? Or is it sorted by time of discovery?

In the code I see this:
https://github.com/victronenergy/dbus-systemcalc-py/blob/master/delegates/dvcc.py#L404

mpvader, 2019-02-15:
I've updated the initial tekst to the new spec. Izak will work on that soon. (within days)

closed 2019-02-18 by jhofstee

mpvader, 2019-04-13:
Added in v2.30~29; with a followup bugfix in v2.30~30