victronenergy/venus #422: cleanup vebus alarms

mpvader, 2019-02-16:
- [x] Remove double low battery, temperature, overload and dc ripple alarms - [Done](https://github.com/victronenergy/gui/commit/fdf93e2772dfe20c7112f2f05818639a1c9aba4a).
- [x] On VRM no longer generate alarm events from the generic paths (drastic; but I think its best; welcome to counter me on it) - Done in VRM db on 2019-02-16
- [x] On VRM no longer generator alarms for VE.Bus Error 1 (drastic; but I think its best; welcome to counter me on it) - Done on VRM db on 2019-02-16
- [ ] Further improve.

----

Right now its a mess in the gui and on VRM. Two issues I saw:
1. Low battery, temperature, overload and dc ripple alarms are available on two D-Bus paths. So trigger two events.
2. VE.Bus Error 1 is also used to create alarm notifications. But its always accompanied by something else (low battery for example).

Below screenshots shows both problems.

The simplest for issue 1 would be to remove those system wide alarm paths from mk2-dbus. But; we did that already and then put them back for a reason: it breaks existing ModbusTCP installations.

One solution we could do for VRM is see if those phase specific paths exists; and if they do; then supress the system wide ones to go into the database. There is a hook for such exceptions already.

[image]

[image]

mpvader, 2019-02-16:
Added VE.Bus Error 3 into the mix. From [the docs](https://www.victronenergy.com/live/ve.bus:ve.bus_error_codes#error_3_-_not_all_or_more_than_the_expected_devices_were_found_in_the_system):

-----
Possible causes and solutions:

1. This error often follows VE.Bus Error 1. Solution: solve the cause for VE.Bus Error 1. Note that when using an older CCGX (version before v1.40), it can be that the first error is not reported on the Alarm log on VRM. So even when it only lists VE.Bus Error 3, it can very well be that that error was preceeded by VE.Bus Error 1.
2. The system is not properly configured: all VE.Bus devices connected to the VE.Bus network must be configured as one parallel, split- and/or three-phase system. Do not connect two separate VE.Bus systems together.
3.. Communication cable error: check the communication cables.

------

That first cause should never make it to a user interface. So suppress in mk2-dbus? How feasible is that @vejpasop? Or even in the mk3 code or VE.Bus firmware? But then its on the long term list rather than fast improvement..

vejpasop, 2019-02-19:
I am hesitant to implement an alarm filter.  Such filter could hide important alarms which complicates debugging of faulty systems. 

For example:
VE.Bus alarm 1 (Device is switched off because  one of the other phases in the system has switched off)  can be triggered for multiple reasons.  If a low battery or a temperature alarm causes the error 1 we will see an a corresponding VE.Bus Alarm at that time also. If the cause of the switch off, of one units is due to faulty hardware we would like to see error 1.

For this I also do not like to filter out error 1 on VRM.

Wouldn't it be better to stack the Alarm log on VRM and in the user interface. The alarms that are registered within for instance 5 minutes apart are stacked and only the last alarm is shown as top of the stack.  The user  can than unfold the stack if it wants to see a detailed order of events. 

Side note:
The Total_kissosso site has configuration/design issues. To me the alarm log seems valid. This many alarms should not be triggered as part of "normal" operation.

mpvader, 2019-02-20:
> The Total_kissosso site has configuration/design issues. To me the alarm log seems valid. This many alarms should not be triggered as part of "normal" operation

In a normal operation there are no alarms at all. Indeed there is an issue. But because of all the alarms, especially VE.Bus Error 1 and 3; it ends up at R&D rather than a straightforward ‘hey a low battery, lets look into that’ case

mpvader, 2019-02-20:
Stacking the alarm log will be a lot of Ui work, in multiple UIs (VictronConnect, gui, vrm). I prefer to solve the real issue.

How about filtering out vebus error 1 and 3 in case there is an overload, temperature, low battery or dc ripple alarm?

vejpasop, 2019-03-18:
There is a big bug in the phase specific alarm path. They do not work in Multi phase systems when there is  the same alarm in more than one phase.  For instance "Low Battery" will never show  on these path. It is only indicated on the generic path.

Broken since: Venus v2.00 (Mk2-dbus v2.38) 2016-12-12

Consequences.
Venus devices in multi phase systems do  not show the "Low Battery" Alarm at all. With a bit of luck it only show the VE.Bus alarm 1 "Device switched off".  Apparently the GUI does not use the generic path (anymore?) 

Unfortunately by fixing the bug the alarm reporting will get even worse since we will get the "Low Battery" alarm on all three phases.

[image]

Before releasing this fix. We need to review the GUI and VRM logger code. 

Should they use the system wide "LowBattery" alarm path?  What about "Ripple" ?
For the "HighTemperature" and "OverLoad" path  it makes more sense to indicate them per phase.

@mpvader  Should I create a separate issue for this bug?

mpvader, 2019-04-02:
As discussed, the plan:

- LowBatteryVoltage & High DC Ripple: notifications will be based on the generic path, in both VRM and the GUI. And information per phase is visible in both VRM and the GUI.
- Temperature & Overload: notifications will be based on the phase specific path. And therefore we won't need a generic path anymore.

mpvader, 2019-04-03:
Ripple alarms done in VRM now. By the way this means that there are no notifications sent to people with version < 2.30. Thats just what it is; I can't think of any cleaner solution; and Ripple is not a often occuring issue.

[image]

vejpasop, 2019-04-04:
- mk2dbus: suppress VE.Bus Error when overload, temperature and low battery is active. -> Done (VE.Bus error 1 is now also suppressed when ripple alarm is active)
- update the dbus page with paths -> Done 
- mk2dbus: look into suppressing VE.Bus Error 3 -> Did not find a common cause for error 3 on VRM. We leave it for now.

mpvader, 2019-04-04:
Last pieces were done in v2.30~41 so closing

closed 2019-04-04 by mpvader