victronenergy/venus #426: gui: wifi password is not applied

martinbosma, 2019-02-22:
Applying the Wi-Fi password in Venus v2.30~29 does not seem to work. The state remains `Disconnected`.
Since there has been some rework on the Mb items on the GUI, I suspect this is a GUI issue.

I have tested this on a VenusGX:
- Press space on the `Password` item in the Wi-Fi menu
- Type the password in the edit box
- Hit Enter or click the accept button

When I use `connmanctl` to connect to the same Wi-Fi network I was able connect.

mpvader, 2019-02-22:
Martin, besides not applied, the wifi password was also lost after doing an update; right?

https://victrondevelopment.slack.com/archives/CDBK9A71S/p1550765240019600?thread_ts=1550585541.012900&channel=CDBK9A71S&message_ts=1550765240.019600

Heikki his venus gx was working fine before you updated his venus gx

martinbosma, 2019-02-23:
I don't know if the password was lost, but it was disconnected from the wifi after the update to v2.30~29.

jhofstee, 2019-02-25:
It won't loose the password, it can't be set anymore

closed 2019-02-25 by jhofstee