victronenergy/venus #427: mqtt-rpc: add support for xupc

WrdVictron, 2019-03-01:
xupc is the same as vup: a commandline tool to update firmware; but then for devices, connected to VE.Can, that have a xup bootloader.

- [x] Add xupc commandline tool to Venus - done in v2.30~27 
- [ ] (@elnino-ict ) Add xupc to [mqtt-rpc](https://github.com/victronenergy/mqtt-rpc)
  - not for scanning; only for updating. Scanning doesn't need any changes; `vup` will also find xup-devices. For updating, mqtt-rpc needs to select between xupd and dup commandline tools by filename extension (`.xup` vs `.vup`).
  - once done; ask Izak to review and release to Venus.
- [x] Add support for .xup files to the firmware library: [issue](https://github.com/victronenergy/venus-private/issues/84)
- [ ] (Wiebren) Add .xup files to Dropbox
- [x] (Wiebren) Add error codes to [internal mqtt-rpc documentation](https://github.com/victronenergy/venus-private/wiki/mqtt-rpc-and-vrm)
- [x] (Wiebren) Check & amend [customer documentation](https://www.victronenergy.com/live/vrm_portal:remote-firmware-update) where necessary
- [ ] (@mauritsvdvijgh) vrm-mqtt-rpc: add support for xupc: https://github.com/victronenergy/vrm-mqtt-rpc/blob/master/vrm_mqtt_rpc/client.py#L430
- [ ] (@AlexImb) VRM: modify popup text: change `.vup` to `.vup or .xup`; see image below (dup example)

Background info:
- Similar issue for adding xupd support (VE.Direct xup updater): #389
- For more details see: #389 

[image]

mauritsvdvijgh, 2019-03-07:
I need access to a venus device running v2.30~27 or higher with a device attached to it that has a xupc bootloader to test mqtt-rpc.py changes.

Somehow the Venus GX we have updated to v2.23 instead of v2.30~xx though "update to" is set to "Latest Release candidate". Even if I had the latest release candidate I still do not have a device with a xupc bootloader attached to it for testing however.

WrdVictron, 2019-03-07:
The firmware for the xupc product is not yet available. It is expected March 15th. 
Once released, I will make it available on Dropbox.
Then I will also give you access to a venus running v2.30~27 or higher, with the corresponding xupc product connected.

WrdVictron, 2019-03-14:
@mauritsvdvijgh: Release files are available on Dropbox (SmartSolar MPPT VE.Can)

thlassche, 2019-04-24:
I deployed `vrm-mqtt-rpc` and `vrm` with .xup support. Thiago tested it using a MPPT: It worked. So all that's left is the end-to-end tests and the .xup files in the Dropbox I think

jhofstee, 2019-08-19:
@WrdVictron can you update or close this one..

closed 2019-08-27 by WrdVictron