victronenergy/venus #431: 16  out mkx_rst not working on victronenergy,easysolar

jhofstee, 2019-03-07:
The mkx_rst pin doesn't work on the easysolar. One issue seems that it is not muxed properly. Trying to force that:

```
+&pio {
+    mk2reset: mk2reset {
+        pins = "PA16";
+        function = "gpio_out";
+    };
+};
+
 &reg_usb2_vbus {
     gpio = <&r_pio 0 11 GPIO_ACTIVE_HIGH>; // GPIOL11 / VUSB_ENABLE
     status = "okay";
@@ -76,7 +83,7 @@
 
 &uart1 {
     pinctrl-names = "default";
-    pinctrl-0 = <&uart1_pins>;
+    pinctrl-0 = <&uart1_pins &mk2reset>;
     status = "okay";
 };
 
```

doesn't seem to help either.

mansr, 2019-03-08:
There is an error on the PCB. Resistors R325 and R328 have been swapped.

mansr, 2019-03-11:
Compare schematic and PCB layout:
[image]
[image]

mkrenzke, 2019-03-11:
@mansr Yes, you seem to be right. Felix will take care on Wednesday.

jhofstee, 2019-03-20:
thanks, fixed

closed 2019-03-20 by jhofstee