victronenergy/venus #434: Incorrect time zone names in GUI

mansr, 2019-03-13:
A user reports that the time zone names for Australia are incorrect:

> ERROR 1.  There is no time zone label 'E. Australia.....' etc.  The correct naming for 'Brisbane' and QLD is Australian Eastern Standard Time (AEST). QLD does not have DST (Day light saving) so it always remains on AEST. NSW and Victoria have DST and therefore their time adjusts by one hour for DST.  
> 
> Therefore at this time of the year (East coast) we have AEDST (Eastern daylight saving time for Melbourne, Canberra, Sydney).  We have EST for QLD (Brisbane).
> 
> ERROR 2. There is no 'Tasmanian Standard Time' that I am aware of. See zone naming conventions link below.

The selections in the GUI are based on a list from the Unicode Consortium: https://unicode.org/repos/cldr/trunk/common/supplemental/windowsZones.xml
Evidently, this list disagrees with the names actual Australians use.

The grouping in the menus has caused past problems when countries have changed time zones (https://github.com/victronenergy/venus/issues/392). Perhaps using that list wasn't such a good idea.

jhofstee, 2019-08-19:
Perhaps not, I have never looked into this. Do you have better ideas?

izak, 2019-10-29:
Additional time zone issue received via level3 support: There is no timezone for Christmas Island.

mpvader, 2023-03-24:
Update: started to discuss this with Qinetic, on Slack, https://victrondevelopment.slack.com/archives/C020QG3JVEW/p1679668146726119

idea being to have Qinetic fix this in gui-v2, and leave current gui as it is.

askpatrickw, 2023-03-25:
If you have not already, have a look at the IANA's data
https://www.iana.org/time-zones

chriadam, 2023-04-13:
> Update: started to discuss this with Qinetic, on Slack, https://victrondevelopment.slack.com/archives/C020QG3JVEW/p1679668146726119
> 
> idea being to have Qinetic fix this in gui-v2, and leave current gui as it is.

After some thought bubbling, I think it might be worthwhile putting timezone enumeration into venus-platform, see https://github.com/victronenergy/gui-v2/issues/299#issuecomment-1506483721 for some context.

mpvader, 2023-04-14:
Understood.

Setting the time zone should for sure be done via venus-platform indeed.

But for getting the list I'm not so sure if doing that over venus-platform is a good idea. It would mean transferring all available timezones over dbus or mqtt. Quite an overkill for a large and completely static list. It can just as well be done compile time.

Needed info in the gui is:
- region (Antarctica, and so forth)
- city
- UTC offset
- time zone abbreviation

Venus OS seems to get it here: https://github.com/victronenergy/openembedded-core/tree/dunfell_venus/meta/recipes-extended/timezone.

@mansr any thoughts on this, ie you also agree that somehow we should be doing this compile time? Either one time building a list out of that tz database and adding that to gui code; or doing that at compile time of the gui.

Ps for my self a few screenshots on how the current menu looks like:

[image]

[image]