victronenergy/venus #438: Shared temperature from Multi does not work when Multi is switched off

vejpasop, 2019-03-27:
Venus V2.30~38
When using the Multi's battery temperature reading as shared temperature source the distributed temperature becomes -25 degrees Celsius when the Multi is switched off.

See https://community.victronenergy.com/questions/11876/venus-os-v23035-available-for-testing.html?childToView=12875#comment-12875

When the Multi is switched off, the Multi mk2dbus "Dc/0/Temperature" value becomes invalid. SystemCalc's  "Dc/Battery/Temperature" value also becomes invalid. 

How should this work when the Multi is switched off? The Multi is unable to read and report the temperature in this state.

jhofstee, 2019-05-16:
v2.30~35 changed STS to also include the onboard temperature sensor when set battery. When the multi is turned of the temperature becomes invalid and the onboard will be checked. Since the default is battery this likely works if a temperature sensor is connected. In this case it is and it reports 4 degrees Celsius, while the battery is 19 degrees, so likely a refrigerator is used for temperature compensation. At very least the default should not be used.

jhofstee, 2019-05-18:
Closing this issue, since the external sensors does seem to report the correct temperature. Voltages are correct as well. What does seem off is the temperature reported by the multi, but that might be an installation issue. Furthermore a BMV could report a bogus temperature in mid point setup when selecting a temperature function temporarily. That has also been fixed.

closed 2019-05-18 by jhofstee