victronenergy/venus #439: Put a warning in the GUI/Manual when DVCC is enabled and there is more than one battery

izak, 2019-04-03:
If you have two batteries; each with an mppt; both mppts to the gx device. One BMV (or intelligent battery witb canbus bms) and then you enable dvcc.

* Both SVS and STS is enabled by default.
* The lowest-instance battery is auto-selected as battery monitor
* Both MPPTs immediately follow battery 1.

That might be a problem that should be warned against.

mpvader, 2019-10-14:
ok, put it in systemcalc (& gui & vrmlogger) just like that other warning you're making.

jhofstee, 2019-11-08:
I changed the milestone, almost all alarms have no number and I think there is some effort going on to at least make charger errors unique, so we like want to know what they are doing before adding more random numbers...

mpvader, 2019-11-28:
There are also systems that are designed to have multiple batteries:

[image]

So lets back this out.

closed 2019-12-02 by jhofstee

mpvader, 2021-03-08:
After removing it from the gui back in 2019, it was left enabled on VRM (as part of the automatic alarms monitor)

Which is shouldn't be: the final purpose of this parameter was decided to be diagnostics just for ourselves, to help troubleshooting a site that behaving strange.

So I removed the alarm flags on this attribute just now in the vrm database; and expect that any sites with that alarm raised will automatically have it closed as soon as a new data-transmission comes in.