victronenergy/venus #449: Please make Venus OS compatible with MK2-USB

mildred, 2019-04-27:
Hello, I'm running Venus OS locally, and I found that it's not compatible with MK2-USB (that I bought instead of MK3-USB because of old stocks). It's really easy to make it compatible with MK2-USB the following way.

Here is what I changed in the configuration to start `mk2-dbus` on serial ttys connected to the computer. The first step is a modification on `/data` and I'm fine with configuring it manually

> create a file `/data/conf/serial-starter.d/mkx-usb.conf` with:
> ```
> alias	default		gps:vedirect:mkx
> ```
> (included from `/etc/venus/serial-starter.conf`)

However, `mk2-dbus` cannot start because the product name (taken from udev) is `MK2USB_COM_Interface` and the script `/opt/victronenergy/vecan-mk2/start-mk2-dbus.sh` does not recognize this value. The script must be modified to allow MK2-USB to be recognized:

> Modify `/opt/victronenergy/vecan-mk2/start-mk2-dbus.sh` the following way:
> ```diff
> --- /opt/victronenergy/vecan-mk2/start-mk2-dbus.sh
>+++ /opt/victronenergy/vecan-mk2/start-mk2-dbus.sh
>@@ -10,6 +10,9 @@
>     MK3-USB_Interface)
>         type=mk3
>         ;;
>+    MK2USB_COM_Interface)
>+        type=mk2
>+        ;;
>     builtin-mkx)
>         type=$($base/onboard-mk)
>         args="$args -i"
>
> ```

`mk2-dbus` itself is compatible, it makes sense for the startup script to be compatible too.

mpvader, 2019-04-27:
Hi @mildred; thanks for your input.

I'm afraid its not that easy: there are other issues when using the MK2 as well; I half remember they relate to firmware updates. Perhaps also on/off.

What I remember with 100% certainty is that this was discussed and that there were some good reasons not to go here. Sorry. There is a good reason for this to be the way it is; even if its not ideal from feature point of view.

mildred, 2019-04-27:
> I'm afraid its not that easy: there are other issues when using the MK2 as well; I half remember they relate to firmware updates. Perhaps also on/off.

I found of that too, but as I understand, unless you drop a configuration file in `/data/conf/serial-starter.d/`, the device will not be activated. If the device is activated (because you had shell access on Venus OS), it makes sense to accept MK2 if it's being plugged-in, no ?

If the MK2-USB firmware is too old, I suspect that mk2-dbus will try to update it, and fail, which will be no worse than having the startup script fail before even starting mk2-dbus, am I right ?

What I remember from reading was that MK2-USB and MK3-USB were identical, except that for firmware upgrades, MK2-USB was not able to be powered off programmatically (you had to pull the USB plug) and thus was not suited for an automated environment.

I took care to update my MK2-USB device using VE.Configure before doing that, and it now works well.

mildred, 2019-04-27:
If MK2-USB must be disabled by default at mk2-dbus level (even though it will not be started by serial-started by default), something like this could be imagined:

```bash
case $product in
    MK3-USB_Interface)
        type=mk3
        ;;
    MK2USB_COM_Interface)
        if ! [ -e /data/conf/mk2-dbus-force-allow-mk2 ]; then
            echo "Sorry, MK2-USB is disabled by default because firmware can't be updated"
            echo "If you enable it, this will void your warranty and kill your children"
            exit 1
        fi
        type=mk2
        ;;
    builtin-mkx)
        type=$($base/onboard-mk)
        args="$args -i"
        ;;
    *)
        log "Unknown device type"
        exit 1
        ;;
esac
```

mpvader, 2019-04-27:
Hi; yes it could be done. But still I don’t want it. sorry. Too small an advantage to start looking into. Its all complicated enough as it is.

I’m sure you’re aware, but still: you know that some day, in an auto update of Venus OS, your system will probably stop working; right? Easy enough to fix by then probably; until ofcoruse the day comes that in the mk3 software we start taking advantage of the extra code space compred to mk2 etc.

closed 2019-04-27 by mpvader