victronenergy/venus #451: Make the push button a network reset button

mpvader, 2019-05-03:
What it needs to do is reset various things to factory defaults.
- enable local access point
- reset ethernet, wifi, and any other network settings to dhcp; remove gateway etc.
- remove Remote Console password

The alternative would be to make it reset all settings to factory defaults. For example by deleting all files except whats in /data/venus, for example.

This will fix people no longer being able to access their Venus GX.

Contents of a random data partition:
```
./home/root/.bash_history
./home/root/2625428.set
./home/root/.python-history
./home/root/2046821656.rvsc
./home/root/temp.log
./home/root/.ssh
./home/root/.ssh/known_hosts
./home/root/snapshot.png
./home/root/Settings
./home/root/Settings/Trolltech.conf
./home/root/2625428.ram
./home/root/t.test
./home/vnctunnel
./home/vnctunnel/.ssh
./home/vnctunnel/.ssh/authorized_keys
./home/vnctunnel/.ssh/id_rsa.pub
./home/vnctunnel/.ssh/id_rsa
./var/lib/connman
./var/lib/connman/ethernet_028191859555_cable
./var/lib/connman/ethernet_028191859555_cable/data
./var/lib/connman/ethernet_028191859555_cable/settings
./var/lib/connman/settings
./var/lib/serial-starter/ttyUSB0
./var/lib/serial-starter/ttyUSB3
./var/lib/serial-starter/ttyUSB5
./var/lib/serial-starter/ttyUSB14
./var/lib/serial-starter/ttyUSB10
./var/lib/serial-starter/ttyUSB9
./var/lib/serial-starter/ttyUSB6
./var/lib/serial-starter/ttyUSB4
./var/lib/serial-starter/ttyUSB7
./var/lib/serial-starter/ttyUSB1
./var/lib/serial-starter/ttyUSB12
./var/lib/serial-starter/ttyUSB8
./var/lib/serial-starter/ttyS2
./var/lib/serial-starter/ttyS3
./var/lib/serial-starter/ttyUSB2
./var/lib/serial-starter/ttyUSB13
./var/lib/serial-starter/ttyS1
./var/lib/serial-starter/ttyUSB11
./var/lib/serial-starter/ttyUSB15
./var/lib/random-seed
./var/lib/mk2-dbus
./var/lib/mk2-dbus/mkxport-ttyS3.settings
./var/lib/mk2-dbus/mkxport.settings
./vrmfilescache
./log/*
./keys/ssh_host_rsa_key
./keys/ssh_host_ecdsa_key
./keys/ssh_host_ecdsa_key.pub
./keys/ssh_host_dsa_key.pub
./keys/ssh_host_rsa_key.pub
./keys/ssh_host_dsa_key
./etc/timestamp
./lost+found
./themes/overlay
./venus/installer-version
./venus/wpa-psk
./venus/unique-id
./venus/serial-number
./db/vrmlogger-backlog.sqlite3
./conf/mosquitto.d
./conf/mosquitto.d/vrm_bridge.conf
./conf/settings.xml
./conf/vrm_auth_token.txt
./conf/vncpassword.txt
```

mpvader, 2019-05-03:
https://victrondevelopment.slack.com/archives/C9QE9QRPW/p1555421247005000

jhofstee, 2019-10-21:
included in v2.40~36

closed 2019-10-21 by jhofstee