victronenergy/venus #453: dvcc: fixup warning text

mpvader, 2019-05-06:
1. some people think that it only requires enabling if you have a solar charger.
2. text is wrong, now that we have the new VE.Can MPPTs coming into stock soon.

[image]

izak, 2019-06-04:
This text also shows up on its own (without the DVCC switch below it) when the access is set to "User", which is confusing to users. The text should follow the same `show` logic as the switch below it.

izak, 2019-09-17:
Suggested by Matthijs, and I agree, implement checking for old firmware in systemcalc. The GUI then only need to look in one place.

closed 2019-09-30 by jhofstee