victronenergy/venus #458: vrmlogger: PV inverters disappear from VRM when set to Multiphase.

izak, 2019-05-14:
The problem appears to be that overall power is not logged to VRM, only the per-phase values. When a PV-inverter is set as Multi-phase, it does not report per-phase values, only an overall power value. This means nothing is logged to VRM unless the state or the power level changes, which means for long periods nothing is logged for a PV-inverter, causing VRM to think it's been removed.

Solution: We probably need to log overall power, and handle it VRM side.

izak, 2019-06-14:
A simple implementation that makes a single-phase inverter pretend to be split-phase if configured as Multiphase.

https://github.com/victronenergy/dbus-fronius/tree/american-cross-phase-connection

@mpvader I would appreciate your thoughts. It's a fairly simple change, it simply makes the assumption that when a single-phase inverter is configured as Multi-phase, then it implies that it sits across L1 and L2.

We can complete the functionality by removing the lock-out in the GUI.

izak, 2019-09-08:
A change in the gui is still required, reopening.

izak, 2019-09-11:
Selecting MultiPhase is now an option for single-phase inverters.

[image]

mpvader, 2019-09-12:
Allowing three phase sounds weird?

closed 2019-09-30 by jhofstee

skumancer, 2024-05-24:
Hey guys, I'm having an issue where if I chose Split-phase (L1+L2), I'm losing the Fronius PV Inverter from the VRM display. Any clues?